OmenMon/OmenMon #1: Is It Working for You or Not? Please Report Here

GeographicCone, 2023-11-04:
Is **OmenMon** working for you or not?
Please report here, using the following template (click the button on the right-hand side to copy it):

````markdown
* OmenMon Version: **0.50**
* Model & Version: **HP xxxx nn.nn** 
* Working:
  * [Please list what you tried that worked]
* Not Working:
  * [Please list what you tried that did not work]
````

###  Where to Get Model & Version

* If you are able to start the GUI, these are the **first three** items shown in the [main window](https://omenhub.github.io/gui#system):

[image]

* If you can't even get to the [main window](https://omenhub.github.io/gui#main):
  * Press <kbd>Win</kbd>-<kbd>X</kbd> and then <kbd>A</kbd> to start the _Administrative Command Prompt_
  * Type: `wmic baseboard get manufacturer,product,version` and press <kbd>Enter</kbd>

Feel free to provide more information but at the very least, please include this.

GeographicCone, 2023-11-04:
* OmenMon Version: **0.50**
* Model & Version: **HP 8A14 32.25** 
* Working:
  * Everything as per the docs at [omenmon.github.io](https://omenmon.github.io/)

Note: `8A14` aka `Ralph 22C1` is the _Omen 16 (2022)_ laptop. Tested on Windows 10 `21H2` (`10.0.19044`).

breadeding, 2023-11-05:
Can't get to the main window.

Manufacturer  Product  Version
HP            8BAB     76.39

[image]

GeographicCone, 2023-11-05:
@breadeding Thank you for the report!

Can you post what's under 详细信息? It'll say which BIOS call failed specifically. An example of what it should look like:

````
Exception: BIOS call failed: Command not available
OmenMon: Void Check(Int32)
   at OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   at OmenMon.Hardware.Bios.BiosCtl.GetBacklight()
   at OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   at OmenMon.AppCli.CliOp.BiosGet[TResult](String message, Func`1 biosMethod, String extra)
   at OmenMon.AppCli.CliOp.Loop(String[] args)
   at OmenMon.App.Main(String[] args)
````

The GUI tries to do many things at the same time. The same things can be done from CLI one by one. To isolate the problem, could you try the CLI mode? `OmenMon -Bios` reads and shows all the BIOS information. Output should look like [here](https://omenmon.github.io/cli#bios).

Regarding the temperature, can you run the following: `OmenMon -Ec CPUT GPTM RTMP TMP1 TNT2 TNT3 TNT4 TNT5 -Bios Temp`

These are the sensors being checked. Which one has the wrong temperature? An example output from my system:

````
OmenMon Hardware Monitoring & Control Utility Version 0.50-Preview
Embedded Controller (-Ec)
- Register 0x57 Byte: 0x1e = 0b00011110 = 30 [CPUT]
- Register 0xb7 Byte: 0x00 = 0b00000000 = 0 [GPTM]
- Register 0x58 Byte: 0x1e = 0b00011110 = 30 [RTMP]
- Register 0x59 Byte: 0x1b = 0b00011011 = 27 [TMP1]
- Register 0x47 Byte: 0x00 = 0b00000000 = 0 [TNT2]
- Register 0x48 Byte: 0x17 = 0b00010111 = 23 [TNT3]
- Register 0x49 Byte: 0x18 = 0b00011000 = 24 [TNT4]
- Register 0x4b Byte: 0x00 = 0b00000000 = 0 [TNT5]
BIOS (-Bios)
- Temperature: 0x17 = 0b00010111 = 23 [°C]
````

breadeding, 2023-11-05:
@GeographicCone 
Here are the details：
```
For more information about invoking just-in-time (JIT) debugging instead of this dialog box,
See the end of this message.

**************** Exception text ****************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 6
    at OmenMon.Hardware.Bios.Bios.Check(Int32 code)
    in OmenMon.Hardware.Bios.BiosCtl.GetThrottling()
    at OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
    in OmenMon.AppGui.GuiFormMain.UpdateSys()
    in OmenMon.AppGui.GuiFormMain.UpdateAll()
    in OmenMon.AppGui.GuiFormMain..ctor()
    in OmenMon.AppGui.GuiTray.ShowFormMain()
    at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
    at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
    at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
    at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
    at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
    at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
    at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
    In System.Windows.Forms.Control.WndProc(Message& m)
    In System.Windows.Forms.ToolStrip.WndProc(Message& m)
    in System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
    at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

**************** Loaded assembly ****************
mscorlib
     Assembly version: 4.0.0.0
     Win32 version:4.8.9181.0 built by: NET481REL1LAST_C
     Basic code: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
     Assembly version: 0.50.0.0
     Win32 version:0.50.0.0
     Basic code: file:///D:/Tools/OmenMon/OmenMon.exe
----------------------------------------
System.Drawing
     Assembly version: 4.0.0.0
     Win32 version:4.8.9032.0 built by: NET481REL1
     Basic code: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
     Assembly version: 4.0.0.0
     Win32 version:4.8.9172.0 built by: NET481REL1LAST_C
     Basic code: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
     Assembly version: 4.0.0.0
     Win32 version:4.8.9181.0 built by: NET481REL1LAST_C
     Basic code: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
     Assembly version: 4.0.0.0
     Win32 version:4.8.9032.0 built by: NET481REL1
     Basic code: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
     Assembly version: 4.0.0.0
     Win32 version:4.8.9032.0 built by: NET481REL1
     Basic code: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
     Assembly version: 4.0.0.0
     Win32 version:4.8.9181.0 built by: NET481REL1LAST_C
     Basic code: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Management.Infrastructure
     Assembly version: 1.0.0.0
[... 119 lines cut ...]
# 28: Cpu 4800 rpm Gpu 5100 rpm @ 46 °C
# 29: Cpu 4900 rpm Gpu 5200 rpm @ 47 °C
# 30: Cpu 5000 rpm Gpu 5300 rpm @ 47 °C
# 31: Cpu 5100 rpm Gpu 5400 rpm @ 47 °C
# 32: Cpu 5200 rpm Gpu 5500 rpm @ 47 °C
# 33: Cpu 5300 rpm Gpu 5600 rpm @ 48 °C
# 34: Cpu 5400 rpm Gpu 5700 rpm @ 48 °C
# 35: Cpu 5500 rpm Gpu 5800 rpm @ 49 °C
# 36: Cpu 5600 rpm Gpu 5900 rpm @ 50 °C
# 37: Cpu 5700 rpm Gpu 6000 rpm @ 50 °C
# 38: Cpu 5800 rpm Gpu 6100 rpm @ 51 °C
# 39: Cpu 5900 rpm Gpu 6200 rpm @ 51 °C
# 40: Cpu 6000 rpm Gpu 6300 rpm @ 52 °C
- Temperature: 0x2a = 0b00101010 = 42 [°C]
Exception: BIOS call failed: Unknown response from BIOS: 6

OmenMon: Void Check(Int32)
   在 OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   在 OmenMon.Hardware.Bios.BiosCtl.GetThrottling()
   在 OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   在 OmenMon.AppCli.CliOp.BiosGet[TResult](String message, Func`1 biosMethod, String extra)
   在 OmenMon.AppCli.CliOp.Loop(String[] args)
   在 OmenMon.AppCli.CliOp.Loop(String[] args)
   在 OmenMon.App.Main(String[] args)
```

GeographicCone, 2023-11-05:
@breadeding Excellent! I was just updating my last comment to mention what I thought was the culprit, and in the meantime this confirms it:

````
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 6
    at OmenMon.Hardware.Bios.Bios.Check(Int32 code)
    in OmenMon.Hardware.Bios.BiosCtl.GetThrottling()
````

wangzhengbin, 2023-11-05:
Hello author, I found on my omen laptop that after I set the fan curve and set the software to follow the system startup. If the GUI page is not opened after restarting the computer, the software will not be able to detect the correct temperature in minimize mode, and if the computer temperature increases at this time, the fan speed will not change according to the temperature.
[image]
I must open the GUI page every time I restart my computer to solve this problem. So is there a solution to this problem?

GeographicCone, 2023-11-05:
@wangzhengbin Fixed in the [latest build 0.52](https://github.com/OmenMon/OmenMon/releases/latest). Thank you for reporting this, it was a big omission!

mertcdev, 2023-11-05:
* OmenMon Version: **0.52 Preview**
* Model & Version: **HP            8BD1     75.75** 
* Working:
  * [Maximum mode Fan control in taskbar menu]
  * [Temperature in taskbar menu with Dynamic Icon but not ]
  * [Backlight enable/disable and  OmenMon Cool/OEM Default in taskbar menu]
* Not Working:
  * [Default/Performance/Cool/Off mode Fan control in taskbar menu]
  * [Graphics not showing submenu and shows this error]
  * [Can't get to the main window and Show Monitor got same error]
[image]

<details>

<summary> Details full text </summary>

```

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 4
   at OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   at OmenMon.Hardware.Bios.BiosCtl.GetGpuMode()
   at OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   at OmenMon.Hardware.Platform.Settings.GetGpuMode(Boolean forceUpdate)
   at OmenMon.AppGui.GuiFormMain.UpdateSys()
   at OmenMon.AppGui.GuiFormMain.UpdateAll()
   at OmenMon.AppGui.GuiFormMain..ctor()
   at OmenMon.AppGui.GuiTray.ShowFormMain()
   at OmenMon.AppGui.GuiTray.EventIconMouseClick(Object sender, MouseEventArgs e)
   at System.Windows.Forms.NotifyIcon.OnMouseClick(MouseEventArgs mea)
   at System.Windows.Forms.NotifyIcon.WmMouseUp(Message& m, MouseButtons button)
   at System.Windows.Forms.NotifyIcon.WndProc(Message& msg)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
    Assembly Version: 0.52.0.0
    Win32 Version: 0.52.0.0
    CodeBase: file:///C:/Users/MERT/Downloads/Programs/OmenMon.exe
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9172.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9200.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Management.Infrastructure
    Assembly Version: 1.0.0.0
    Win32 Version: 10.0.19041.1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Microsoft.Management.Infrastructure/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.dll
----------------------------------------
Microsoft.Management.Infrastructure.Native
    Assembly Version: 1.0.0.0
    Win32 Version: 10.0.19041.1 (WinBuild.160101.0800)
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
Microsoft.Management.Infrastructure.UserFilteredExceptionHandling
    Assembly Version: 0.0.0.0
    Win32 Version: 10.0.19041.1 (WinBuild.160101.0800)
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

```

</details>

  * [App can't continue working after press Omen Key]
[image]

Note: 8BD1 is the OMEN 15 2021 AMD LAPTOP (en1000). Tested on Windows 10 22H2. 
This model have not Mux support and ECO/Balanced/Performance mode with manuel fan control in OGH.

bbaallance, 2023-11-06:
version:0.53
resolution:2560*1440
scaling setting:150%
Meet a scaling issue, GUI can not show all the information.
[image]

azxcvbnmazx, 2023-11-06:
FAN control not work
[image]

wangzhengbin, 2023-11-06:
I discovered a bug while using the software. When the system temperature increases, the software will adjust the fan speed to dissipate heat, but sometimes the speed will suddenly decrease. It can be observed from the software that although the system temperature is very high, the fan is still at low speed, and the fan speed will wait for more than ten seconds before it can recover. An error event about ACPI can be found in the system's event viewer.
[image]
[image]

GeographicCone, 2023-11-06:
@mertcdev

> Graphics not showing submenu and shows this error
> Can't get to the main window and Show Monitor got same error
> App can't continue working after press Omen Key

All these issues boil down to the GPU mode BIOS query (the _Omen_ key too, since it's trying to show the main window).

I added the following workaround:
* The GPU mode query will now fail silently and return `Hybrid`
* The part of the menu for switching GPU mode will not be shown at all unless byte #7 of BIOS system data (`-Bios System`) has the 0x08 (for `Supported`) bit set

Hopefully this does the job as a fix and you can enjoy the rest of the app. Please try [build 0.54](https://github.com/OmenMon/OmenMon/releases/latest) and if you can still see the menu, let me know what the value of [System Data](https://omenmon.github.io/cli#system) byte #7 is.

> Auto mode no effect on fan RPM

**Auto** mode _is_ just the default. And what these "performance" modes actually do is a bit of a mystery except you definitely want _Performance_ for maximum performance. So it might as well be working.

If you want to check if fan control works, select **Const** mode and move the trackbars, then press the button. Or try `-Bios FanLevel=#,#` [from the command line](https://omenmon.github.io/cli#fanlevel).

> 60hz works but 165hz no effect, my panel 144hz

This is not autodetected. Set `<PresetRefreshRateHigh>144</PresetRefreshRateHigh>` in `OmenMon.xml`.

> Prog combobox is empty

You have to define a program in `OmenMon.xml`. See the [example](https://omenmon.github.io/config#fan-programs).

@bbaallance
and @wangzhebin - from your screenshot I see there's the same issue

> resolution:2560*1440
> scaling setting:150%
> Meet a scaling issue, GUI can not show all the information.

_System Status_: It isn't actually scaling (I spent a lot of time debugging that, and also tested 100-225% at 25 percentage-point steps, although it's a quagmire, so there might still be issues) but that your regional defaults ("settings for non-Unicode programs") override the default _MS Shell Dlg_ font and also its line height. Short of (1) forcing a custom font on everyone, which I wouldn't want, also because if someone later wants to translate the UI to, say, Chinese, there would be missing glyphs or other issues if I did that, or (2) making the text smaller for everyone (since line height can't be controlled directly), the best solution I can think of right now is as follows: in [build 0.54](https://github.com/OmenMon/OmenMon/releases/latest) I added the configuration option [GuiSysInfoFontSize](https://omenmon.github.io/config#guisysinfofontsize), from my tests the value of `14` should work well if you set the regional settings to _Chinese_. Hope this works around the problem.

As for the other marked part, unless I'm missing something, _the unit_ is what it's supposed to be: [′⁻¹] = [min⁻¹] = [1/min] = [rpm].

@azxcvbnmazx

> [Post since deleted but it showed a screenshot with the window size way too large for what it should be.]

Like I said I did a lot of testing with different DPI settings (more than I care to admit), but there are these edge cases. I suspect the culprit might be one workaround I added to resize the window if the DPI changes while the application is running (or perhaps when you move the window to another monitor). I added a setting for this as [GuiDpiChangeResize](https://omenmon.github.io/config#guidpichangeresize) and it's set to **false** by default, so it's disabled for everyone unless re-enabled.  Hopefully this fixes the wrong window size from the start for anyone that had it. But if someone needs to change the DPI settings while using the app, you can re-enable the resizing by setting this value to **true**, which will restore the pre-0.54 behavior.

@wangzhebin

> An error event about ACPI can be found in the system's event viewer.

I don't really know what happened there. This looks like an issue accessing the Embedded Controller. Possibly a conflict with another app also trying to use it. OmenMon tries to be a good citizen and wait for the `Global\Access_EC` mutex before doing anything. Not sure what more can be done other than reducing EC usage (but the EC has all the interesting information).

Separately, EC readings can sometimes be unreliable. If we're just displaying the information, then maybe it's benign but as far as a fan program is involved, there are a couple of ways this could be handled. Broadly, I agree the fan program logic could be more intelligent. This is something to be improved. Meanwhile you can reduce [UpdateProgramInterval](https://omenmon.github.io/config#updateprograminterval) if you want the fan program to be more responsive and increase [UpdateMonitorInterval](https://omenmon.github.io/config#updatemonitorinterval) or hide the main window when not in use to minimize EC usage (and switch to the static icon but I remember you had that already).

@Kubagf

I've now changed the GPU mode BIOS call to silently fail instead of causing an exception (see my reply to @mertcdev above), so you can try again running `-Bios` if you want and it will get past it. That being said, based on the system data being mostly empty, I think it might be difficult to get other stuff working as well. On a general note, seems some of these _Omen_ models don't really have that much in common, so it is a bit of a challenge.

**Thank you all for your reports!**

Kubagf, 2023-11-06:
OmenMon Version: 0.54
Model & Version: HP OMEN 15 15-en0000
Working:
Nothing
Not Working:
Everything

After update -bios command shows a little bit more but still not working

OmenMon Hardware Monitoring & Control Utility Version 0.54-Preview
BIOS (-Bios)
- Manufacturing (Born-On) Date: 20200213 [YYYYMMDD]
- System Design Data: c8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [↓]
- Status Flags: 0x00c8 = 0b0000000011001000 = 200 [BiosPerfPpab3, BiosPerf6, BiosPerf7]
- Unknown Byte: 0x00 = 0b00000000 = 0 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x00 = 0b00000000 = 0 [V0]
- Support Flags: 0x00 = 0b00000000 = 0 [0]
- Default CPU Power Limit 4: 0x00 = 0b00000000 = 0 [W]
- BIOS-Defined Overclocking: 0x00 = 0b00000000 = 0 [NotSupported]
- Graphics Mode Switching Support: 0x00 = 0b00000000 = 0 [0]
- Default CPU Concurrent Power Limit w/GPU: 0x00 = 0b00000000 = 0 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x00 = 0b00000000 = 0 [Off]
- GPU Processing Performance AI Boost (PPAB): 0x00 = 0b00000000 = 0 [Off]
- GPU Device Power State (DState): 0x00 = 0b00000000 = 0
- GPU Peak Temperature Sensor Threshold: 0x00 = 0b00000000 = 0 [°C]
- Graphics Mode (Legacy): 0x00 = 0b00000000 = 0 [Hybrid]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
Exception: BIOS call failed: Command not available

OmenMon: Void Check(Int32)
   w OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   w OmenMon.Hardware.Bios.BiosCtl.HasOverclock()
   w OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   w OmenMon.AppCli.CliOp.BiosGet[TResult](String message, Func`1 biosMethod, String extra)
   w OmenMon.AppCli.CliOp.Loop(String[] args)
   w OmenMon.AppCli.CliOp.Loop(String[] args)
   w OmenMon.App.Main(String[] args)

Wyjątek nieobsłużony: System.ArgumentOutOfRangeException: Wartość musi być większa lub równa zero i mniejsza od rozmiaru buforu konsoli w tym wymiarze.
Nazwa parametru: top
Rzeczywista wartość to 51.
   w System.Console.SetCursorPosition(Int32 left, Int32 top)
   w OmenMon.AppCli.Cli.Close()
   w OmenMon.App.OnExit(Object sender, EventArgs e)

mertcdev, 2023-11-06:
@GeographicCone
> All these issues boil down to the GPU mode BIOS query (the _Omen_ key too, since it's trying to show the main window).
>
> I added the following workaround:
>
> * The GPU mode query will now fail silently [...]

Thanks for suggestions. Graphics submenu works fine [build 0.54] Main window and Fan submenu still get crashes because of Bios call still failed. 

[image]

Dragofagnir, 2023-11-07:
* OmenMon Version: **0.54**
* Model & Version: **HP Omen 16 b0000 and Omen 17 Ck1000** 
* Not Working:
  * .NET Framework error :

Consultez la fin de ce message pour plus de détails sur l'appel du débogage
juste-à-temps (JIT) à la place de cette boîte de dialogue.

************** Texte de l'exception **************
System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à OmenMon.AppGui.GuiFormMain.SetupTmp()
   à OmenMon.AppGui.GuiFormMain..ctor()
   à OmenMon.AppGui.GuiTray.ShowFormMain()
   à OmenMon.AppGui.GuiTray.EventIconMouseClick(Object sender, MouseEventArgs e)
   à System.Windows.Forms.NotifyIcon.OnMouseClick(MouseEventArgs mea)
   à System.Windows.Forms.NotifyIcon.WmMouseUp(Message& m, MouseButtons button)
   à System.Windows.Forms.NotifyIcon.WndProc(Message& msg)
   à System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Assemblys chargés **************
mscorlib
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase : file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
    Version de l'assembly : 0.54.0.0
    Version Win32 : 0.54.0.0
    CodeBase : file:///C:/Users/Phuc/Downloads/OmenMon.exe
----------------------------------------
System.Drawing
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9032.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9172.0 built by: NET481REL1LAST_C
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9032.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9032.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9200.0 built by: NET481REL1LAST_C
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Management.Infrastructure
    Version de l'assembly : 1.0.0.0
    Version Win32 : 10.0.22621.1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Microsoft.Management.Infrastructure/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.dll
----------------------------------------
Microsoft.Management.Infrastructure.Native
    Version de l'assembly : 1.0.0.0
    Version Win32 : 10.0.22621.1 (WinBuild.160101.0800)
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
Microsoft.Management.Infrastructure.UserFilteredExceptionHandling
    Version de l'assembly : 0.0.0.0
    Version Win32 : 10.0.22621.1 (WinBuild.160101.0800)
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
mscorlib.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9032.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
System.Windows.Forms.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9032.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Débogage JIT **************
Pour activer le débogage juste-à-temps (JIT), le fichier de configuration pour cette
application ou cet ordinateur (machine.config) doit avoir la valeur
jitDebugging définie dans la section system.windows.forms.
L'application doit également être compilée avec le débogage
activé.

Par exemple :

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Lorsque le débogage juste-à-temps est activé, les exceptions non gérées
seront envoyées au débogueur JIT inscrit sur l'ordinateur
plutôt que d'être gérées par cette boîte de dialogue.

wangzhengbin, 2023-11-07:
Version 0.54 cannot start, with the following warnings.
[image]

GeographicCone, 2023-11-08:
@Kubagf

> System.ArgumentOutOfRangeException
> System.Console.SetCursorPosition(Int32 left, Int32 top)

Improved the workaround, and will also catch that exception just in case from now on.

_Background:_ every Windows executable is supposed to be either a GUI or a CLI ("CUI") application but officially can't be both. **OmenMon** does it unofficially and there are trade-offs. With the "legacy" `cmd.exe` shell it works reasonably well but a _PowerShell_ console (which is now the default) introduces an additional issue when the cursor does not move to the end of the buffer when the application is done. This workaround moves the cursor manually to the bottom of the window (or buffer).

@mertcdev (and @Kubagf)

> Graphics submenu works fine
> Main window and Fan submenu still get crashes because of Bios call still failed.

From 0.55 you can set `<BiosErrorReporting>false</BiosErrorReporting>` in `OmenMon.xml` to ignore all BIOS errors.

@Dragofagnir and @Wangzhebin

> La référence d'objet n'est pas définie à une instance d'un objet.
> OmenMon.AppGui.GuiFormMain.SetupTmp()

> 未将对象引用设置到对象的实例
> OmenMon.AppGui.GuiTray.Update()

This was a really stupid omission on my part, forgetting to define the default case properly. Well, at least I now get to see all those error messages in different languages. Fixed in 0.55. Thank you for reporting!

_Background:_ this is part of the groundwork for per-model fan and temperature sensor settings. People with different models who, for example, find the temperatures are off, can now define different ways to read temperature sensors for their models:

* https://github.com/OmenMon/OmenMon/blob/master/Hardware/Platform.cs#L130

…while sharing the rest of the application. Same about controlling the fans:

* https://github.com/OmenMon/OmenMon/blob/master/Hardware/Platform.cs#L51

_That's it for the fixes so far. I will post a new release shortly, and share an update here in another comment._

GeographicCone, 2023-11-08:
**[Build 0.55](https://github.com/OmenMon/OmenMon/releases/latest)** is out.

The GUI crash introduced in 0.54 is **fixed**.

This version adds the [BiosErrorReporting](https://omenmon.github.io/config#bioserrorreporting) configuration option. Adding `<BiosErrorReporting>false</BiosErrorReporting>` to `OmenMon.xml` will silently ignore any BIOS errors, making it possible to use **OmenMon** on not fully compatible hardware.

This is the first release built with an automatic _GitHub_ workflow. It took some time setting up but speeds up the process for every future release.

Full source code has been posted in the repository. If you want to try building **OmenMon** yourself, check out the [Build Instructions](https://omenmon.github.io/build#instructions).

Thank you for your interest in the project!

ArmynC, 2023-11-09:
* OmenMon Version: **0.55**
* Model & Version: **HPOMEN 17-ck0011nq** 
* Not Working:
  * Keyboard backlight colors (for under a second the color changes in something wrong then is reverted)
  * Fan max value. That value is 4600 rpm for my laptop under HP Omen official software, but the slider has it up to 5000+. Not going up to that value though if I set it to Max.

[image]

GeographicCone, 2023-11-09:
@ArmynC

> Fan max value. That value is 4600 rpm for my laptop under HP Omen official software, but the slider has it up to 5000+. Not going up to that value though if I set it to Max.

There is no way to autodetect maximum values for fan levels. The OEM app just pulls model-specific JSON data with `{C,G}PU_Fan_Speed_{Upp,Low}er_Bound_List` in it. You can find a dump of these files here: https://github.com/GeographicCone/OmenHwCtl/tree/master/Reference/Downloaded

So, you have to manually set the maximum level yourself in the config file: https://omenmon.github.io/config#fanlevelmax (and you can also set the minimum level as well)

> Keyboard backlight colors (for under a second the color changes in something wrong then is reverted)

Do you have the 4-zone keyboard? Per-key RGB is not supported right now (see #3).

Assuming yes, are the colors reported correctly, either on the picture or from the command line (`OmenMon -Color`), and only changing them doesn't work? Or even the reported values are wrong? If the former, could it be that _Omen Hub_ is running in the background and interfering with the color change? If the latter, can you please provide a couple of example outputs of `OmenHwCtl -GetColorTable` using [OmenHwCtl](https://github.com/GeographicCone/OmenHwCtl/) (**OmenMon's** the predecessor script) with known color values?

Thank you for the report!

ArmynC, 2023-11-09:
I understand. From my testing, I see that the fan (CPU side) goes (forced) up to **5000** RPMs under the slider set at highest point, which I think that's about **400** more RPMs than the default Max value, give or take. Wouldn't this induce a risk for the fan-life? If it isn't just a sensor false calculation, there should be a warning.

[image]

Secondly, you are right. My bad takeaway was reporting this before uninstalling the HP Omen software first (which I even thought of). I have a 4-zone keyboard. Now there is no conflict at all without the Light Studio. Speaking of that, a nice feature would be adding more customization like speed of wave, light angle and so on.

I've encountered something strange. The app-tray options have 60/165 Hz refresh rate. Is it hardcoded or unable to detect the correct value (144Hz)? In the case of a hard-coded string, a better suited variant would be keeping 'standard' and 'high' refresh-rate text.

[image]

Here are some debug info (just as a future reference):

```
OmenMon Hardware Monitoring & Control Utility Version 0.55-Release
BIOS (-Bios)
- Manufacturing (Born-On) Date: 20230522 [YYYYMMDD]
- System Design Data: 4a 01 35 01 01 c4 00 06 00 00 00 00 00 00 00 00 []
- Status Flags: 0x014a = 0b0000000101001010 = 330 [Unknown1, BiosPerfPpab3, BiosPerf6, Ppab8]
- Unknown Byte: 0x35 = 0b00110101 = 53 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x01 = 0b00000001 = 1 [V1]
- Support Flags: 0x01 = 0b00000001 = 1 [SwFanCtl]
- Default CPU Power Limit 4: 0xc4 = 0b11000100 = 196 [W]
- BIOS-Defined Overclocking: 0x00 = 0b00000000 = 0 [NotSupported]
- Graphics Mode Switching Support: 0x06 = 0b00000110 = 6 [Unset1, Unknown2]
- Default CPU Concurrent Power Limit w/GPU: 0x00 = 0b00000000 = 0 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x01 = 0b00000001 = 1 [On]
- GPU Processing Performance AI Boost (PPAB): 0x01 = 0b00000001 = 1 [On]
- GPU Device Power State (DState): 0x01 = 0b00000001 = 1 [D1]
- GPU Peak Temperature Sensor Threshold: 0x57 = 0b01010111 = 87 [°C]
- Graphics Mode (Legacy): 0x01 = 0b00000001 = 1 [Discrete]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
- Overclocking Support: 0x01 = 0b00000001 = 1
- Memory Overclocking Support: 0x00 = 0b00000000 = 0
- BIOS Undervolt Support: 0x01 = 0b00000001 = 1
- Keyboard Type: 0x02 = 0b00000010 = 2 [TenKeyLess]
- Keyboard Backlight Support: Yes
- Keyboard Backlight: 0xe4 = 0b11100100 = 228 [On]
- Keyboard Backlight Color Table: 4 Zones - Right: ffffff, Middle: ffffff, Left: ffffff, Wasd: ffffff [All White]
- LED Animation Table:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
- Fan Count: 0x02 = 0b00000010 = 2
- Maximum Fan Speed: No
- Fan Type: 0x00 = 0b00000000 = 0 []
- Fan #1 Type: 0x00 = 0b00000000 = 0 [Unsupported]
- Fan #2 Type: 0x00 = 0b00000000 = 0 [Unsupported]
- Fan #1 Level: 0x14 = 0b00010100 = 20 [Cpu]
- Fan #2 Level: 0x00 = 0b00000000 = 0 [Gpu]
- Fan Speed Level Table: 2 Fans, 15 Levels
# 00: Cpu 0000 rpm Gpu 0000 rpm @ 16 °C
# 01: Cpu 1800 rpm Gpu 0000 rpm @ 21 °C
# 02: Cpu 2000 rpm Gpu 1800 rpm @ 25 °C
# 03: Cpu 2200 rpm Gpu 2000 rpm @ 28 °C
# 04: Cpu 2300 rpm Gpu 2100 rpm @ 30 °C
# 05: Cpu 2400 rpm Gpu 2200 rpm @ 31 °C
# 06: Cpu 2600 rpm Gpu 2400 rpm @ 33 °C
# 07: Cpu 2800 rpm Gpu 2600 rpm @ 36 °C
# 08: Cpu 2900 rpm Gpu 2700 rpm @ 38 °C
# 09: Cpu 3000 rpm Gpu 3200 rpm @ 40 °C
# 10: Cpu 3500 rpm Gpu 3700 rpm @ 42 °C
# 11: Cpu 3700 rpm Gpu 3900 rpm @ 43 °C
# 12: Cpu 4000 rpm Gpu 4200 rpm @ 45 °C
# 13: Cpu 4300 rpm Gpu 4500 rpm @ 48 °C
# 14: Cpu 4600 rpm Gpu 4800 rpm @ 50 °C
- Temperature: 0x23 = 0b00100011 = 35 [°C]
- Thermal Throttling Status: 0x04 = 0b00000100 = 4 [Default]

```

```
Embedded Controller (-Ec)
0x _0 _1 _2 _3 _4 _5 _6 _7 _8 _9 _A _B _C _D _E _F
0_ 00 00 00 00 37 81 00 00 00 00 00 00 00 00 00 00
1_ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2_ 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 29 00
3_ 00 00 0d 00 ff ff 44 00 00 00 37 6d c4 00 00 00
4_ 0b 00 04 14 04 00 00 00 23 2f 00 00 00 00 00 00
5_ 80 00 04 00 00 05 00 42 2c 29 00 00 00 00 00 00
6_ 00 00 26 22 00 00 00 00 08 74 00 0a 00 08 08 00
7_ c6 11 7f 10 28 3c 80 33 31 33 2d 31 37 2d 34 42
8_ 2d 41 00 00 00 7f 10 d9 42 01 00 00 01 e0 00 00
9_ d1 d1 7f 10 04 30 62 00 00 00 00 00 00 b2 0b 00
a_ 03 80 00 aa 01 00 00 20 00 fb 10 00 00 01 00 00
b_ d2 07 00 00 00 37 81 28 00 00 01 05 05 00 00 00
c_ 03 01 00 00 00 80 00 00 00 00 00 00 00 00 00 00
d_ 62 00 00 00 02 04 00 01 62 00 1d 03 00 00 d1 d1
e_ 00 00 02 00 00 00 00 00 00 00 0a 00 00 6d 00 00
f_ 48 00 20 00 00 00 00 00 00 00 38 36 34 33 30 31
```

GeographicCone, 2023-11-09:
**I updated the top post in this issue, summarizing all the information we've got so far. I encourage everyone to have a look at it.**

----

@wangzhengbin

I'm really sorry I repeatedly got your name wrong, and I've only just realized it. Should be all fixed now, except in the [commit message](https://github.com/OmenMon/OmenMon/commit/83b8269d61eaa4b27c5fdc9d2fcbaa8f592c964c), which can't be changed anymore. Apologies for that again.

----

@ArmynC

> Wouldn't this induce a risk for the fan-life?

I don't think so. It just won't run faster than the design speed.

> If it isn't just a sensor false calculation, there should be a warning.

There is no calculation, the data is pulled straight from the Embedded Controller: you can get the same with `OmenMon -Ec RPM1(2) RPM3(2)`.

Sometimes the EC readings happen to be hilariously off, it's possible to briefly see a fan speed of 15,000 rpm for example. I have yet to decide what to do about it, since it's benign (this number is not acted upon, just reported to the user), and there is more than one way of addressing it (discard the incorrect reading and keep showing the previous one, or immediately repeat the query).

5,000 rpm is well within the realms of possibility though. Not sure what kind of warning you have in mind. In general, I'd rather the app not get in the way of what the user is trying to do, just assume that it's for a reason.

> A nice feature would be adding more customization like speed of wave, light angle and so on.

It's possible to use **OmenMon**'s command-line mode in a script to do this already. It's just a question of coming up with an algorithm for an interesting effect. If someone writes a script like this as a proof-of-concept, we can have it integrated into **OmenMon**.

> The app-tray options have a 60/165 Hz refresh rate. Is it hardcoded (based on the description) or unable to detect the correct value (144 Hz)?

There is no way to reliably autodetect this across all models. Even if there were, someone might want a custom setting (e.g. for use with an external monitor). So that's something you have to set manually in the configuration file: https://omenmon.github.io/config#presetrefreshratehigh

Once you do that, the checkmark will show correctly next to it when it's set.

> [From the website:] The high refresh rate is usually 144 or 165 Hertz, and the latter value is the default

The meaning of "default" here is the hard-coded value when the setting can't be read from the configuration file. But if it can, the configuration file takes precedence and overrides it.

> My laptop model has iGPU - dedicated GPU switch capabilities (MUX on Intel-only models on my region) in Gaming Hub, but I see that the OmenMon is unable to do it. I'm wrong on understading its meaning?

You're not wrong. There's a high chance it will work with **OmenMon** too. Initially it was enabled on all devices. But if you read the prior discussion, querying GPU mode on unsupported systems was causing BIOS error 4, so I disabled this setting unless byte #7 of [System Design Data](https://omenmon.github.io/cli#system) is `0x08` = `0b1000` because that's what _Omen Hub_ seems to check. On my system (where I can change the GPU mode too) the value is `0x0C` = `0b1100`. I now see from the `-Bios` output you posted that on your system the value is `0x06` = `0b0110`. So that's why it's not enabled for you.

**Can you please confirm that you can [switch the GPU mode](https://omenmon.github.io/cli#gpumode) from the command line with `OmenMon -Bios GpuMode=Optimus` and `OmenMon -Bios GpuMode=Discrete`?** (Note: have to reboot each time for the change to take effect.) If yes, I'll enable it for these flags too (the only question is if it's `0x02` or `0x04` that determines it).

wangzhengbin, 2023-11-09:
No big deal, we talked earlier about the Windows system log reporting ACPI errors.I used aida 64 to change the system temperature and observe the parameters while testing the stability of omenmon. I suspect that the software conflicts with omenmon's access to ec. Later I will close aida 64 to re-test omenmon, if there are other issues will be timely feedback to you.

bbaallance, 2023-11-09:
Version:0.55
I have tried to translate omenmon's text into Chinese, but I have encountered some issues.

In XML, when I tried to change the value of "GuiSysInfoFontSize" from the default of ’0’ to another value, there was garbled code in the system info area. When set to the default value, it can be displayed normally.
[image]
At the same time, there seem to be some issues in the original "OmenMon. en_US. xml" file, but I have not yet completed the translation work, so I did not locate all the errors. A more serious error is that after loading the XML, "About omenmon" will not be displayed, even if I have not modified any content. Others are some minor formatting issues.
[image]
[image]

Detailed information:

有关调用实时(JIT)调试而不是此对话框的详细信息，
请参见此消息的结尾。

************** 异常文本 **************
System.ArgumentException: 文件格式无效。
   在 System.Windows.Forms.RichTextBox.StreamIn(Stream data, Int32 flags)
   在 System.Windows.Forms.RichTextBox.StreamIn(String str, Int32 flags)
   在 System.Windows.Forms.RichTextBox.set_Rtf(String value)
   在 OmenMon.AppGui.GuiFormAbout..ctor(String title, String text)
   在 OmenMon.AppGui.GuiFormMain.EventActionHelp(Object sender, EventArgs e)
   在 System.Windows.Forms.Form.OnHelpButtonClicked(CancelEventArgs e)
   在 System.Windows.Forms.Form.WmSysCommand(Message& m)
   在 OmenMon.AppGui.GuiFormMain.WndProc(Message& m)
   在 System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** 已加载的程序集 **************
mscorlib
    程序集版本:4.0.0.0
    Win32 版本:4.8.9181.0 built by: NET481REL1LAST_C
    基本代码:file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
    程序集版本:0.55.0.0
    Win32 版本:0.55.0.5000
    基本代码:file:///C:/Users/RYJ/Downloads/OmenMon-0.55.0-Release/OmenMon.exe
----------------------------------------
System.Drawing
    程序集版本:4.0.0.0
    Win32 版本:4.8.9032.0 built by: NET481REL1
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    程序集版本:4.0.0.0
    Win32 版本:4.8.9172.0 built by: NET481REL1LAST_C
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    程序集版本:4.0.0.0
    Win32 版本:4.8.9181.0 built by: NET481REL1LAST_C
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
    程序集版本:4.0.0.0
    Win32 版本:4.8.9032.0 built by: NET481REL1
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    程序集版本:4.0.0.0
    Win32 版本:4.8.9032.0 built by: NET481REL1
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    程序集版本:4.0.0.0
    Win32 版本:4.8.9181.0 built by: NET481REL1LAST_C
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Management.Infrastructure
    程序集版本:1.0.0.0
    Win32 版本:10.0.22621.1
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Microsoft.Management.Infrastructure/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.dll
----------------------------------------
Microsoft.Management.Infrastructure.Native
    程序集版本:1.0.0.0
    Win32 版本:10.0.22621.1 (WinBuild.160101.0800)
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
Microsoft.Management.Infrastructure.UserFilteredExceptionHandling
    程序集版本:0.0.0.0
    Win32 版本:10.0.22621.1 (WinBuild.160101.0800)
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
System.Windows.Forms.resources
    程序集版本:4.0.0.0
    Win32 版本:4.8.9032.0 built by: NET481REL1
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_zh-Hans_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------
mscorlib.resources
    程序集版本:4.0.0.0
    Win32 版本:4.8.9032.0 built by: NET481REL1
    基本代码:file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_zh-Hans_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------

************** JIT 调试 **************
要启用实时(JIT)调试，
该应用程序或计算机的 .config 文件(machine.config)的 system.windows.forms 节中必须设置
jitDebugging 值。
编译应用程序时还必须启用
调试。

例如:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

启用 JIT 调试后，任何未经处理的异常
都将被发送到在此计算机上注册的 JIT 调试程序，
而不是由此对话框处理。

ArmynC, 2023-11-09:
@GeographicCone 

> I don't think so. It just won't run faster than the design speed.

I'm not that sure. Better said, the specified maximum speed (in my case 5000RPM) can _probably_ be reached under some conditions (e.g. unit test). In _absolutely no case_ was I able to reach 5000 RPMs with stock Omen Hub. Yet, with OmenMon that was possible. Also, I saw that the output RPM is somewhat messy, no problem on that one, is easy to differentiate.

So, the default provided app case (manual and max case), provided readings - between 4400-4800 at most:

[image]
-
[image]

And OmenMon sliders got it good, between 4800-5100 (see [pic above](https://github.com/OmenMon/OmenMon/issues/1#issuecomment-1803706435)).

In my case, the Max fan control respected the original RPM which was ~4600. Only the sliders (maybe other controls too?) are able to 'overclock' the fans over the given standard RPMs. So, a warning before making them accessible (grayed-out) to the user (~Are you sure, it may reach higher level than intended ones during normal conditions? Y/N) would be desired.

> There is no way to reliably autodetect this across all models. Even if there were, someone might want a custom setting (e.g. for use with an external monitor). So that's something you have to set manually in the configuration file: https://omenmon.github.io/config#presetrefreshratehigh

Is this feature a must? Since, the OS is able to modify the refresh rate quite easy, I think it is more confusing to the end user in this state. Otherwise, hiding the value and displaying it as Low / High refresh rate instead is better in the end. On my laptop it is outputing 165 which isn't right (144 in reality).

> Can you please confirm that you can [switch the GPU mode](https://omenmon.github.io/cli#gpumode) from the command line with OmenMon -Bios GpuMode=Optimus and OmenMon -Bios GpuMode=Discrete? (Note: [...]

Ok, so I've tried. It wasn't too pretty 😞

**First step**, toggled the Optimus setting. Note: 'Optimus' flag is case-sensitive (won't accept it). Since anything but the arg isn't case-sensitive, it would be nice to have it too as such.

[image]

Omen Hub (reported):

[image]

... restarted the laptop. Black screen the entire time. Even the OMEN logo wasn't rendering, nor the UEFI wasn't accessible. Somehow I managed to brick my bios. As an fix (hint), after trying to reboot several good times and checking it, I saw the power button flickering. I removed the charger, held the power button down for a more seconds (about 20 seconds), put the charger back in, turned on the laptop and it displayed the error:

[image]

Here, I think it reverted all the default BIOS settings or some custom OMEN settings at least (undervolting gone). Went back in the OS. Checked the flags and by risking it again (lol) it switched flawless back to discrete.

[image]

GeographicCone, 2023-11-09:
@wangzhengbin

> I used aida 64 to change the system temperature and observe the parameters while testing the stability of omenmon. I suspect that the software conflicts with omenmon's access to ec.

Yeah, I also observed that. If the EC is under heavy load such as when running `OmenRun -EcMon` with [EcMonInterval](https://omenmon.github.io/config#ecmoninterval) set to `200` [ms] and _AIDA64_ or _HwInfo64_ at the same time, it will trigger an "instant sleep" mode eventually (for lack of a better description).

Not sure what can be done about it on **OmenMon**'s side though, other than limiting EC usage (so [GuiDynamicIcon](https://omenmon.github.io/config#guidynamicicon) disabled and main window hidden or with an increased [UpdateMonitorInterval](https://omenmon.github.io/config#updatemonitorinterval)).

@bbaallance

Happy to hear about the translation! Great news!

> There seem to be some issues in the original "OmenMon.en_US.xml" file [...] after loading the XML, "About omenmon" will not be displayed, even if I have not modified any content. Others are some minor formatting issues.

It's actually the same underlying issue: the backslash escape sequences needed in the original `.cs` file [Library\LocaleData.cs](https://github.com/OmenMon/OmenMon/blob/master/Library/LocaleData.cs) are unnecessary for the XML and should have been removed. So instead of `\` there are `\\` in RTF control sequences, which causes them to be incorrect. And if the RTF header is affected, it triggers an exception. Thanks for spotting that, fixed it in [OmenMon.en_US.xml](https://github.com/OmenMon/Localization/blob/master/OmenMon.en_US.xml) already, but you can just replace `\\` with `\` everywhere in your translated version.

> In XML, when I tried to change the value of "GuiSysInfoFontSize" from the default of ’0’ to another value, there was garbled code in the system info area. When set to the default value, it can be displayed normally.

It's the GDI charset. I'll have to think about what the best solution is here but meanwhile wanted to update you on the other issue.

GeographicCone, 2023-11-09:
@bbaallance

> > In XML, when I tried to change the value of "GuiSysInfoFontSize" from the default of ’0’ to another value, there was garbled code in the system info area. When set to the default value, it can be displayed normally.

> It's the GDI charset. I'll have to think about what the best solution is here but meanwhile wanted to update you on the other issue.

It was actually something else altogether: Unicode characters from outside the 7-bit ASCII set need escaping for rich-text fields. It actually shouldn't have worked at all, and it didn't, except if you happened to have the regional settings for non-Unicode programs set to the same language you wanted to use for the messages (which is often the case but not always). Anyway, fixed now, and it should work regardless of the regional settings. Minimum test case:

````xml
<?xml version="1.0" encoding="utf-8"?> 
<OmenMon>
    <Config>
        <GuiSysInfoFontSize>14</GuiSysInfoFontSize>
    </Config>
    <Messages>
        <String Key="GuiAboutText">{\rtf1\ansi 口吅品 }</String>
        <String Key="GuiMainSysAdapterMeetsRequirement">\cf3 口吅品</String>
        <String Key="GuiAboutTextErrorPrefix">{\rtf1\ansi\deff0{\colortbl;\red255\green0\blue0;}\cf1 口吅品 </String>
    </Messages>
</OmenMon>
````

----

@ArmynC

> On my laptop it is outputing 165 which isn't right (144 in reality).

What you need to do is open the file `OmenMon.xml` (located right next to `OmenMon.exe`) with a text editor such as _Notepad_, find the line where it says `<PresetRefreshRateHigh>165</PresetRefreshRateHigh>` and change that to `<PresetRefreshRateHigh>144</PresetRefreshRateHigh>`, and save the file (while **OmenMon** is not running). It will then be 144 [Hz] as it should.

> I removed the charger, held the power button down for a more seconds (about 20 seconds), put the charger back in, turned on the laptop and it displayed the error: [...]

If this ever happens in the future, you can hold `Win`-`V` during boot to reset the settings to their defaults. (There's also `Win`-`B` for BIOS recovery.)

<ins>Note</ins>: **OmenMon** is not supposed to run in parallel with _Omen Hub_, what can happen is difficult to predict, so be advised.

> Checked the flags and by risking it again (lol) it switched flawless back to discrete (verified in omen hub too).

Good. So basically switching GPU mode works. Re-enabled in 0.55.1 for this flag (`0x04`).

----

**Build [0.55.1-Preview](https://github.com/OmenMon/OmenMon/releases/tag/0.55.1) is out, fixing the Unicode in RTF issue and also re-enabling GPU mode switching for @ArmynC's laptop. Marked as a pre-release, so doesn't show as the latest build.**

ArmynC, 2023-11-10:
I misspoke, I meant in general usage where these wrong values may mislead the user. In my case I know how to modify it manualy to output the correct refresh rate.

`Win-V` - Good thing about the shortcuts. I wasn't aware of that. Thanks! 
About the OmenHub that was relatively assumed, I wanted to see its output.

Regarding the BIOS error, wouldn't it be a possibility that the my system only allows Hybrid and Discrete, thus bricking it on Optimus? Omen Hub has the first two options only. What are the differences?

ArmynC, 2023-11-10:
Used the preview Build. Tried to change from Discrete to iGPU. Bricked again the BIOS.

After CMOS reset it went on Hybrid mode.

[image]
-
[image]

GeographicCone, 2023-11-10:
@ArmynC

> Regarding the BIOS error, wouldn't it be a possibility that my system only allows Hybrid and Discrete, thus bricking it on Optimus?

> Used the preview Build. Tried to change from Discrete to iGPU (Optimus Soft Switch). Bricked again the BIOS.

Thank you for the testing and reporting. In **[build 0.55.2](https://github.com/OmenMon/OmenMon/releases/tag/0.55.2)** just released, it will switch between `Discrete` and `Hybrid` instead of `Optimus` for devices that don't have bit 3 (`0x08`) set but do have bit 2 (`0x04`) set in byte 7 of [system design data](https://omenmon.github.io/cli#system).

bbaallance, 2023-11-10:
Version:0.55.1
Model: HP omen 17-ck1000, 8A1A

After a period of testing, I finally figured out some things.

Due to the use of machine translation, the meaning may not be completely accurate. Please understand. : )

I noticed an extremely important switch in the ec register, which directly affects the total power consumption of the machine and the long-term power consumption of the CPU. Meanwhile, I believe this has also become the main reason why omenmon cannot replace OGH. For my machine, location: 0x95. Its default value is 0x30. When this position is the default value, the total power consumption of the machine is limited to around 215w, and the long-term power consumption of the CPU is limited to around 55w (not very certain). In the introduction on the omenmon webpage, I learned that you seem to already know that there is a timer in the ec register. But the function of this timer is not just to reset the fan system, it can also reset the switch located at 0x95 (and other seemingly insignificant things). When I used omenmon, I found that it reset the value of this switch while resetting the countdown, resulting in the activation of the power lock.

Next, add some specific information: After switching to performance mode in OGH, the value of 0x95 has also been modified to 0x31. After disabling OGH and restarting the machine, I manually change the value of 0x95 to 0x31, and the timer will be activated. However, if I manually change the value of the timer to 0x00 (position: 0x63) at this time, nothing will happen and nothing will be reset. Next, I will use omenhwctl to modify the GPU mode, using the following parameters:

`. omenhwctl. ps1- maxgpupower`

This seems to refresh the power consumption limit. At the same time, due to the timer being manually reset by me, the effect of unlocking the power limit is permanently retained.

But when I used omenmon, I found that even if I manually changed the value of 0x95 to 0x31, omenmon would soon reset it to 0x30, which obviously immediately led to a decrease in CPU power consumption.

ps:At present, I am aware that the ec registers of other models may be different from those of my machine.
[image]

At the same time, there is also a problem with omenmon. When omenmon is working in the background, the timer reset function seems to fail because my fan stopped running soon. But when I opened the GUI, the fan started working again.

GeographicCone, 2023-11-10:
@bbaallance FYI for _RwEverything_ you can use this template I made long time ago:
https://github.com/GeographicCone/OmenHwCtl/blob/master/Reference/EC_8A14.irw

EC Register `0x95` (`HPCM` in the DSDT for my model) is just the "fan mode" aka "performance mode": 

https://omenmon.github.io/cli#fanmode
https://github.com/OmenMon/OmenMon/blob/master/Hardware/BiosData.cs#L61

````cs
// Fan performance mode
// Source: HP.Omen.Core.Common.PowerControl.PerformanceMode
public enum FanMode : byte {
    LegacyDefault     =  0,  // 0x00 = 0b00000000
    LegacyPerformance =  1,  // 0x01 = 0b00000001
    LegacyCool        =  2,  // 0x02 = 0b00000010
    LegacyQuiet       =  3,  // 0x03 = 0b00000011
    LegacyExtreme     =  4,  // 0x04 = 0b00000100
    L8                =  4,  // 0x04 = 0b00000100
    L0                = 16,  // 0x10 = 0b00010000
    L5                = 17,  // 0x11 = 0b00010001
    L1                = 32,  // 0x20 = 0b00100000
    L6                = 33,  // 0x21 = 0b00100001
    Default           = 48,  // 0x30 = 0b00110000
    L2                = 48,  // 0x30 = 0b00110000
    Performance       = 49,  // 0x31 = 0b00110001 
    L7                = 49,  // 0x31 = 0b00110001 
    L3                = 64,  // 0x40 = 0b01000000
    Cool              = 80,  // 0x50 = 0b01010000
    L4                = 80   // 0x50 = 0b01010000
}
````

Yes, for maximum performance you want it set to `0x31`, which is the value for _Performance_ in the `enum` and throughout the UI. That's why the dynamic icon background will change to "warm" to let you know whenever this mode is active.

But **OmenMon** sets it via the BIOS WMI call already, which means you don't have to do it via the EC.

You are also right about how the timer works. You get 120 seconds (which the UI shows) and it'll be back to _Default_ if this runs out. As long as you're running a fan program or the main window is shown and _Const_ mode selected (not need to be applied), the timer will be rearmed when the countdown falls below some value, which can be configured: https://omenmon.github.io/config#fancountdownextendinterval

I know that you can write `0x00` to the timer register to make it permanent but I didn't want **OmenMon** to do that for a reason: it's a user-space process that can always be terminated for one reason or another (crash or get killed by the user), and then it'd be permanently left with constant fan levels no matter how much the temperature rises (since the countdown applies to this too). So I embraced the idea of having a timer to revert to hardware defaults, it's not an omission.

In short: I see what you're saying, and you are right, but what is new here? And where is the problem?

On every fan program update (30 seconds by default, the value of https://omenmon.github.io/config#updateprograminterval) **OmenMon** updates the fan mode if it detects a change:

https://github.com/OmenMon/OmenMon/blob/master/Hardware/FanProgram.cs#L311
````cs
// Updates the fan mode, optionally only if different than the desired target state
private void UpdateFanMode(
    bool forceUpdate = false,
    BiosData.FanMode? mode = null) {

    // Set the default mode if empty
    if(mode == null)
        mode = Config.FanProgram[this.Name].FanMode;

    // Skip if the settings are the same already, unless forced not to
    if(forceUpdate
        || this.Platform.Fans.GetMode() != mode)

       // Set the fan mode
       this.Platform.Fans.SetMode((BiosData.FanMode) mode);

}
````

Called from: 
https://github.com/OmenMon/OmenMon/blob/master/Hardware/FanProgram.cs#L170

````cs
// Updates the fan program, if any is running
public bool Update() {

    // Fail if no program active
    if(!this.IsEnabled)
        return false;

    // Find out the current maximum temperature,
    // the temperature level for the given temperature,
    // and the target fan levels for the given level
    byte temperature = Platform.GetMaxTemperature(true);
    byte level = GetTemperatureLevel(temperature);
    byte[] fans = GetFanLevel(level);

    // Note: the above could all be accomplished with
    // a single nested call, except we also want to report
    Status(Severity.Notice,
        Config.Locale.Get(Config.L_PROG + "T") + Config.Locale.Get(Config.L_PROG + "SubMax") + " " 
        + Conv.GetString(temperature, 2, 10) + Config.Locale.Get(Config.L_UNIT + "Temperature") + " "
        + Config.Locale.Get(Config.L_PROG + "Lvl") + " " + Conv.GetString(level, 2, 10) + " "
        + Config.Locale.Get(Config.L_PROG + "Fans") + " "
        + Conv.GetString(fans[0], 2, 10) + ", " + Conv.GetString(fans[1], 2, 10));

    // Set the fan level
    SetFanLevel(fans);

    // Perform other updates, only if necessary
    UpdateFanMode();
    UpdateGpuPower();
    UpdateCountdown();

    // Report success
    return true;

}
````

If the mode resets itself from `0x31` to `0x30` it could be a bug in the above code or elsewhere **OmenMon**. Or it could also be being reverted by the hardware for a number of reasons such as:
* Not on AC power (using battery)
* AC adapter doesn't meet the requirements (wrong "smart" AC adapter status, charging from USB-C, etc.: this is why the UI shows this status)
* Battery absent or charge level below some arbitrary threshold, which I don't know

Or the change could be getting reverted by _Omen Hub_.

What you describe should all work already. If it doesn't due to a bug, I'm definitely interested in fixing it. But it's not necessary to write `0x31` to EC register `0x95` because that's what the [fan mode](https://omenmon.github.io/cli#fanmode) BIOS call already does. (Like I wrote before, when there's a choice between a BIOS call and writing to the EC directly, I prefer to make a BIOS call since that's the "native interface").

GeographicCone, 2023-11-10:
@bbaallance

> FYI for _RwEverything_ you can use this template I made long time ago: https://github.com/GeographicCone/OmenHwCtl/blob/master/Reference/EC_8A14.irw

Not in the _RwEverything_ format but this file is a summary of my research into the EC on the model I have (but they're largely similar): https://github.com/GeographicCone/OmenHwCtl/blob/master/Reference/EC_8A14.txt

The part with identified registers:
````text
  Off   Len             Description [Unit]         RW  Value Range
  0x2C  (1)  -          CPU Fan Set Speed [%]      RW  0x00 ... 0x64 [0% ... 100%] / 0xFF [Auto]
  0x2D  (1)  -          GPU Fan Set Speed [%]      RW  0x00 ... 0x64 [0% ... 100%] / 0xFF [Auto]
. 0x2E  (1)  -          CPU Fan Get Speed [%]      R.  0x00 ... 0x64 [0% ... 100%]
. 0x2F  (1)  -          GPU Fan Get Speed [%]      R.  0x00 ... 0x64 [0% ... 100%]
  0x34  (1)  SRP1       CPU Fan Set Speed [krpm]   R.  0x00 [Off] / 0x15 ... 0x37 [2,100 ... 5,500 rpm] / 0xFF [Auto]
  0x35  (1)  SRP2       GPU Fan Set Speed [krpm]   R.  0x00 [Off] / 0x15 ... 0x39 [2,100 ... 5,700 rpm] / 0xFF [Auto]
  0x47  (1)  TNT2       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
. 0x48  (1)  TNT3       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
. 0x49  (1)  TNT4       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
  0x4A  (1)  IRSN       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
  0x4B  (1)  TNT5       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
. 0x57  (1)  CPUT       Temperature: CPU [°C]      R.  0x00 ... 0x64 [0°C ... 100°C]
. 0x58  (1)  RTMP       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
. 0x59  (1)  TMP1       Temperature [°C]           R.  0x00 ... 0x64 [0°C ... 100°C]
  0x5F  (1)  -          HID Disable Toggle         RW  0x00 [All On] / 0x08 [TouchPad Off] / 0x10 [Win Key Off] / 0x18 [Both Off]
  0x62  (1)  OMCC       Manual Fan Control         RW  0x00 [Off / Auto] / 0x06 [On / Manual] / DSDT: Bit #0: OMCC [#1234567 -]
  0x63  (1)  -          Manual Auto Countdown [s]  R.  0x78 ... 0x00 [120s ... 0s]
             -                                     .W  0x00 [Permanent Manual Mode]
  0x95  (1)  HPCM       Performance Mode           RW  0x30 [L2 / Default] / 0x31 [L7 / Performance] / 0x50 [L4 / Cool]
  0x96  (1)  -          Battery Charge Level [%]   R.  0x00 ... 0x64 [0% ... 100%]
  0xA0  (1)  QBHK       Last Hotkey                R.  0x?? [State Changes with Fn Key Combinations Pressed]
  0xA2  (1)  QBBB       HID-Related (?)            R.  0x40 .. 0x43 [State Related to 0x5F HID Disable Toggle]
. 0xB0  (2)  RPM1 RPM2  CPU Fan Get Speed [rpm]    R.  0x0000 ... 0x158A [0 ... ~ 5,500 rpm]
  0xB2  (2)  RPM3 RPM4  GPU Fan Get Speed [rpm]    R.  0x0000 ... 0x1648 [0 ... ~ 5,700 rpm]
. 0xB7  (1)  GPTM       Temperature: GPU [°C]      R.  0x00 ... 0x64 [0°C ... 100°C] [Probe: 0x00 - Not in Use, 0x28 - Minimum]
  0xBA  (1)  CLOW       Minimum Power State        RW  0x00 .. 0x05 [Highest TDP ... Lowest TDP]
  0xBB  (1)  CMAX       Maximum Power State        RW  0x00 .. 0x05 [Highest TDP ... Lowest TDP]
  0xEC  (1)  FFFF FFFS  Max Fan Speed Toggle       RW  0x00 [Off / Regular Speed] / 0x0C [On / Max Speed] [DSDT: Bit #1: FFFF, Bit #2: FFFS (#034567 -)]
  0xF4  (1)  SFAN       Fan Toggle                 RW  0x00 [Fans On] / 0x02 [Fans Off] [Probe: 0x00 - Usually, 0x90 - Briefly]
  0xF9  (1)  GFXM FTHM  Thermal Threshold Reached  RW  Bit #4: GFXM, Bit #7: FTHM [#012356 -]
````

bbaallance, 2023-11-10:
@GeographicCone 

I understand, probably due to the translation of the text, I misunderstood. I mistakenly thought that 'fan program' was only fan control and did not include performance control. But I think this may be a point that needs improvement. Usually, the daily usage mode of people in our discussion group is "performance mode" and requires custom fan mode. But for most of these users, using XML for configuration or using cli mode is not an easy task. Is separating fans and performance control in the GUI a better choice? Just like OGH, adding graphical fan curve drawing (although this may not be an easy task)

ArmynC, 2023-11-10:
> In [build 0.55.2](https://github.com/OmenMon/OmenMon/releases/tag/0.55.2) just released, it will switch between Discrete and Hybrid instead of Optimus for devices that don't have bit 3 (0x08) set but do have bit 2 (0x04) set in byte 7 of [system design data](https://omenmon.github.io/cli#system).

I've tested it. It works just fine. 😃

[image]
-
[image]

wangzhengbin, 2023-11-10:
I've found that using omen gaming hub (ogh) doesn't conflict with aida 64. I also saw in your software introduction that you referred to the note book fan control (nbfc). I've used the software before, and it also has problems with ec access conflicts (and worse). So I want to ask, does omenmon read ec in the same way as nbfc? Or it accesses the ec through the wmi-bios.

GeographicCone, 2023-11-10:
@bbaallance I'm open to all improvement ideas and suggestions, particularly if anyone wants to contribute an implementation. The way fan programs are defined now isn't very friendly for the casual user I guess: you have to come up with the numbers and put them in a text file. But on the other hand, how often do people change these anyway? And what is the workflow? Put the laptop at 100% load and find the lowest fan levels that keep it below the overheat threshold of 87°C. Then fill out the lower thresholds. Would it really make such a difference if one can arrive at the same result by dragging an apex instead? I guess I'm biased because personally I'd rather just type the numbers. I'm not against this as a feature but given limited resources I'd rather focus on improving the actual fan program logic rather than presentation details that don't enable us to do anything new.

I don't really know how to separate the performance and fan control since it's HP that decided to link these together and even apply the same timer to both of them for some reason I can't claim to understand. As far as I know, setting the mode to _Performance_ also enables a different built-in fan-level preset, at least on some devices, so that it's all together is a reflection of HP's (or their ODM's) design decisions, alongside other bright choices like putting the `Pause` key where `Delete` should be on the keyboard and having the Wi-Fi module screw perfectly covered by the heat sink so that the former can't be replaced without removing the latter.

But seriously, I can see what I came up with is based on a certain idea I had in mind and after spending too much time looking at it it's all become kind of "obvious" to me. I always appreciate a fresh look and a new perspective that comes with it. So feel free to let me know how you'd rather have it, perhaps in a new issue, and let's see what can be done.

> In the "fan monitor&control area", I can only use two Chinese characters to replace the original text. But for only two Chinese characters, it is difficult to express the practical use of these functions.

I can see the 自定义 you wanted to put there doesn't fit. 🤣 But I don't know how to make more space there short of rearranging the whole UI. Good news Chinese is probably the world's most succint language. Maybe a bit outside my area of expertise but how about 自设?

> Errors in these data displays

Yeah, I can see the fan speed values on the screenshot are ridiculous. I've just added a constraint for them: in the next build, anything above [FanLevelMax](https://omenmon.github.io/config#fanlevelmax) `× 110` will be rejected. This value works for my model: `FanLevelMax = 55`, real highest speed observed is `~ 5,700 rpm ≤ 55 × 110 = 6,050 [rpm]`. If anyone thinks that's too low, let me know.

----

@ArmynC

> I've tested it. It works just fine. 😃

Glad to hear something works as it should for a change!

----

@wangzhengbin

> I've found that using omen gaming hub (ogh) doesn't conflict with aida 64. I also saw in your software introduction that you referred to the note book fan control (nbfc). I've used the software [...]

**OmenMon**'s EC routines are based on [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor)'s implementation, which is basically a wrapper around the _WinRing0_ driver by [OpenLibSys](https://openlibsys.org/manual/WhatIsWinRing0.html). It's the same driver everybody uses, written by @hiyohiyo (the _CrystalDiskMark_ guy), and it's been around since 2007, so I'd say it's battle-tested.

[NotebookFanControl](https://github.com/hirschmann/nbfc) used [OpenHardwareMonitor](https://github.com/openhardwaremonitor), now discontinued, of which _LibreHardwareMonitor_ is a fork. When writing that part of the code, I compared the two implementations (most recent LHM and last OHM NBFC fork). There's a note about it in the comments: https://github.com/OmenMon/OmenMon/blob/master/Hardware/Ec.cs#L182-L187

They're practically the same except two changes: one is purely cosmetic and makes whoever came up with this one-liner feel very smart: `if(((byte) status & (isSet ? (byte) ~value : value)) == 0)` . But it's not very legible, even I wouldn't write something like that. The other is that they added `Thread.Sleep(1);` in the same `Wait()` loop, which doesn't strike me as a great idea either.

Basically, the EC code is almost the same across versions, it's been used in many places, and it's kind of too simple to have bugs. I modified the LHM driver wrapper quite a lot from the original but not in the part where it runs the I/O port operations since there's nothing to change there really, it just passes the stuff to the kernel driver.

I don't think _Omen Hub_ accesses the EC at all, which is why there are no issues. If EC usage is really a problem then the solution should probably be about finding other ways to query the temperature sensors. Otherwise, I'd be happy to be proven wrong but I don't think there's much that can be improved with **OmenMon**'s EC handling.

GeographicCone, 2023-11-10:
**[New release 0.56.0](https://github.com/OmenMon/OmenMon/releases/latest) is up** with all the latest fixes.

@bbaallance I tried to replicate the issue you mentioned with dropping out of _Performance_ mode but couldn't. For me the mode remains for as long as the fan program is active, even with no window and static icon only. (With no fan program active, to stay in _Performance_ mode you have to select _Const_ and keep it the window open: that's currently by design).

dd871, 2023-11-11:
[image]

I did a bit of testing and found a couple of things which might be bugs:

1) I know you are aware of the CPU fan values reported sometimes to be crazy high. I noticed that for me, this only happens for the CPU fan _percentage_ and it constantly reverts back and forth between (27% - correct current value) and 80ish percent. I did monitor the EC registers but could not actually find the same issue there: 0x2E was fixed at the correct value of 0x1B (27) the whole time. Not sure why the application thinks it goes up to 80ish...

2) Sometimes when OmenMon is running in the background, I get a popup saying my battery is low or something about the charger having an issue (sorry, it is not easily replicable, hence I cant recall the exact error message). In any case, the error is not correct, as I am always connected to the charger and it is at 100% - also it only happens when OmenMon is running and even then only very sporadically. I remember I had the same issue a while back when I was trying to create a working profile for Nbfc, so must be something related to setting the EC registers manually.

3) Whilst looking at the above, my screen suddenly flickered and almost all the EC registers got overwritten once to 0x90. This never happened before, but thought I mention it. Of course this might not have had anything to do with OmenMon and just some weird glitch. Some extract of the EC dump below. In iteration four, almost all registers got set to 0x90 for some reason:

...
0xE0: 00,00,00,90,00,00,00
0xE5: 00,00,00,90,00,00,00
0xEA: 0B,0B,0B,90,0B,0B,0B
0xEB: 00,00,00,90,00,00,00
0xEC: 00,00,00,90,00,00,00
0xF0: 19,19,19,90,19,19,19
0xF2: 20,20,20,90,20,20,20
0xF6: 00,00,00,90,00,00,00
0xF7: 00,00,00,90,00,00,00
0xF9: 00,00,00,90,00,00,00
0xFA: 37,37,37,90,37,37,37
0xFB: 38,38,38,90,38,38,38
0xFE: 30,30,30,90,30,30,30

P4R1H, 2023-11-11:
* OmenMon Version: **0.56-Release**
* Model & Version: **HP 8787 22.59** 

* Working:
  * RGB
  * Counters (temperature, fan speed etc)
  
* Not Working:
  * Changing the prog tab
  * Changing the Auto tab
  * Changing fan speed manually
  
  
  (unhandled exception bios call failed)

P4R1H, 2023-11-11:
* OmenMon Version: **0.56-Release**
* Model & Version: **HP  8787 22.59** (HP Omen 2020, en0037ax)
* Working:
  * RGB
  * Status information (rgb/fanspeed)
  * Switching between Max, const (when sliders are all the way down) and Off
* Not Working:
  * Switching performance modes (prog and auto tabs)
  * Applying const after moving sliders
  
  Error: Unhandled exception, BIOS call failed: Unknown response from BIOS: 46

GeographicCone, 2023-11-11:
@dd871 

To clarify first of all, **OmenMon** only _writes_ to the EC in three situations now:
* Extending countdown timer when a fan program is running
* Switching fans off and back on when set manually from the context menu or the main window
* In CLI mode when instructed by the user `OmenMon -Ec 0x##=...`

Other than that, it's only _reading_ from it. And it doesn't really alter what it reads, other than in the following three ways:
* Discard fan rates over `Config.MaxBelievablePercent` (= `100`)
* Discard temperature readings over `Config.MaxBelievableTemperature` (= `99`)
* Discard fan speeds over `Config.FanLevelMax * (100 + Config.MaxBelievableFanSpeedPercentOverMax)` (= `6050` by default)

So what you see is what it got. 

While there is some wiggle room for EC optimization, I think we'd eventually find the only lasting solution to problems with EC use is not to use the EC, since the root cause is that reading from the EC directly is not how it's "supposed to be done" in the first place. But then the EC has all the interesting raw data. So the way things are is an ugly compromise. I don't think it's feasible to make it 100% bullet-proof with no trade-offs.

There are three separate issues here:
1. **OmenMon**'s interactions with the EC can interfere with other hardware functionality, like for example the battery level briefly being reported as `0` that you mentioned or the system suddenly entering sleep mode or shutting down. This obviously should not be happening. I have been able to observe this but only when running **OmenMon** in both GUI and CLI mode, showing the main window and `-EcMon` updating at 200 ms update interval, and also _AIDA64_ and _HwInfo64_ sensors all at the same time. As long as I _don't_ include `-EcMon` in the routine, this doesn't seem to happen in any short timeframe but I agree the potential for things to go south is definitely there.
2. Temperature readings being wrong could cause the fan program to set fan levels unsuitable to the situation. Since the decision is based on the maximum reading across 9 sensors, the risk of fan levels being accidentally set too low is not very significant. However the converse is more likely, i.e. a one-off falsely high temperature reading causing the fans to spin up unnecessarily.
3. The presentation issue with the data shown in the main window being incorrect at times. This I think is relatively benign as long as the data isn't being further acted upon, i.e. does not lead to any wrong decisions (such as about the fan speeds). That being said, of course it's an issue that has to be acknowledged as well. I've also seen many (all?) register values being shown as `0x90` like you did – in my case this can be observed when resuming from hibernation or standby.

So, what can be done about it all? On a general note, my approach has been to use the EC as a last resort, substituting WMI BIOS calls where possible. However, I also like all the data that can only be pulled from the EC, no matter if sometimes they're off. It satisfies my inner craving for seeing all the constantly-updating indicators even if it doesn't matter much in practice: that's why I wanted to have them all in the main window. The idea is that whenever the user decides to open the main window, it's for a reason: he wants to see all the information, and should be able to do so.

To solve (3), the reading algorithm could be expanded, so that if a current reading differs significantly from the previously-read value, it would take another reading and, of the two candidate values, return the one closer to the previous reading. However, I've been relucant to implement it, since it further increases the burden on the EC, which would exacerbate (1), and like I said before, at the end of the day it's merely a presentation detail.

There is a way of retrieving fan levels via a WMI BIOS call, already implemented with `-Bios FanLevel`. This is how the _Omen Gaming Hub_ gets it. These are values like `22` or `35` for 2,200 or 3,500 rpm respectively, so the other values could be calculated from this instead of depending on the EC but it would mean losing some information with the fan speeds being rounded to 100s. It also means being dependent on the user to provide the correct value for `Config.FanLevelMax`, which can't be auto-detected, for calculating relative values correctly. Like I wrote before, the _Omen Gaming Hub_ pulls this from a downloaded model-specific JSON configuration preset. Considering these limitations, I think the trade-off here is too big to be acceptable: it would make **OmenMon** an objectively worse application.

And the above still doesn't solve the question of temperature readings, which can't be obtained via the WMI BIOS, except for the IR sensor. On some systems, the sensor data can be retrieved from the WMI via another _HP_-specific extension. But on my machine, this returns an empty result even though the classes are there: `wmic /namespace:\\root\HP\InstrumentedBIOS path HPBIOS_BIOSNumericSensor`. And as for the standard WMI way of reading temperature, there's only one thermal zone defined: `wmic /namespace:\\root\wmi PATH MSAcpi_ThermalZoneTemperature`. So, again, not very useful.

Are you running any other programs in the background that also interact with the EC? In such a scenario, I'd advise configuring **OmenMon** to limit EC use: switching off the _Dynamic Icon_, hiding the main window unless necessary, increasing [UpdateMonitorInterval](https:// [... cut ...]
There's of course the possibility the EC routines could be improved but I don't immediately see how. Basically as long as the EC is being used for anything at all, a race condition can manifest when two programs attempt to access it at the same time. **OmenMon** and some other applications wait for a global mutex before doing so but there's no single standard agreed upon for this lock, and the hardware vendor's implementation doesn't assume it has to share the EC with anyone, and that's the root of the problem. The best approach would be to move away from the EC in favor of data accessed by some documented APIs.

GeographicCone, 2023-11-11:
@P4R1H Thank you for the report!

Can you say the error 46 code from BIOS is for which call exactly? It's in the first couple of lines of the [stack trace](https://github.com/OmenMon/OmenMon/issues/1#user-content-exception) when you click on the _Details_.

Note that you can set [BiosErrorReporting](https://omenmon.github.io/config#bioserrorreporting) to **false** in the `OmenMon.xml` configuration file and use the application without getting any exceptions for BIOS-related errors.

_Duplicate post deleted. FYI You can click on the `…` (ellipsis) in the top-right corner to edit, instead of posting again._

GeographicCone, 2023-11-11:
@bbaallance

> [...] adding graphical fan curve drawing 

Just a comment on this: I think this task is really well-suited for an online tool:

[image]

Link to an editable curve for the _Power_ program from the example configuration file I just updated:
https://www.desmos.com/calculator/6vfpghtud0

ramius86, 2023-11-12:
* OmenMon Version: 0.56-Release
* Model & Version: OMEN 17,3" Gaming Laptop PC 17-ck1000 (509V8AV) 
* Working:
  * Keyboard colour ok, fans ok, most of temperature report ok
* Not Working:
  * TNT2 and TNT5 temperature says "interpretation unknown" and popup sometimes a temperature report and suddendly disappear
  
  I saw the development of a similar tool but for Asus and the developer was using a powershell script that was reporting endpoints, don't know if you was already doing something similar or you want to take a look https://github.com/seerge/g-helper/issues/38

GeographicCone, 2023-11-12:
@ramius86 Thank you for the report!

> TNT2 and TNT5 temperature says "interpretation unknown" and popup sometimes a temperature report and suddendly disappear

That's by design because not all sensors are active all the time (e.g. the `GPTM` sensor if the discrete GPU is not in use). As for the tooltips, it's because I'm not certain what they measure, which might also be model-dependent. But you can redefine them in the config file. Example with the defaults:
````xml
<?xml version="1.0" encoding="utf-8"?> 
<OmenMon>
    <Messages>
        <String Key="GuiTipTmpCPUT">CPU Temperature</String>
        <String Key="GuiTipTmpGPTM">GPU Temperature</String>
        <String Key="GuiTipTmpBIOS">Temperature reported by the BIOS</String>
        <String Key="GuiTipTmpIRSN">Infrared Sensor Temperature</String>
        <String Key="GuiTipTmpRTMP">Platform Controller Host Temperature</String>
        <String Key="GuiTipTmpTMP1">Memory Temperature</String>
        <String Key="GuiTipTmpTNT2">Interpretation Unknown</String>
        <String Key="GuiTipTmpTNT3">Storage</String>
        <String Key="GuiTipTmpTNT4">Storage</String>
        <String Key="GuiTipTmpTNT5">Interpretation Unknown</String>
    </Messages>
</OmenMon>
````
> I saw the development of a similar tool but for Asus and the developer was using a powershell script that was reporting endpoints, don't know if you was already doing something similar or you want to take a look [Endpoint discovery for different laptop models seerge/g-helper#38](https://github.com/seerge/g-helper/issues/38)

It's an interesting approach but that's _Asus_-specific. I don't know about _Asus_ but in our case, we have a working implementation that can be analyzed (with the _Omen Gaming Hub_), so doing that is much more efficient. I wrote a bit about it back here: https://github.com/GeographicCone/OmenHwCtl#knowing-what-to-send if anyone's interested.

wangzhengbin, 2023-11-12:
I have closed all software that may conflict with omenmon. Set the notification bar icon of omenmon to static mode, set UpdateMonitorInterval to 3 seconds, and run the software in minimized mode. Unfortunately, the system logs still report ACPI errors and even events such as incorrect power changes and critical battery level triggering. These can lead to sudden performance degradation and sleep issues in the computer. 
[image]
[image]
As you mentioned earlier, omenmon can cause these problems. After testing, I found that even if the time interval for reading EC is continuously increased (within a reasonable time range of course), these problems still cannot be avoided. You also mentioned that if WMI-BIOS is used, more interesting data cannot be obtained. In fact, most users may only be concerned about CPU, GPU temperature, and fan speed control. Perhaps switches can be added in omenmon to allow users to choose to read only a few important parameters or all parameters themselves. For the control of fan speed, you also mentioned that obtaining the speed through WMI-BIOS requires users to determine the maximum fan speed based on their computer model in advance, which can be easily obtained in the Omen Gaming Hub (OGH).  In short, thank you for your series of efforts to save OMEN users, and I also hope that omenmon can completely replace OGH in the near future.

GeographicCone, 2023-11-12:
@wangzhengbin

> As you mentioned earlier, omenmon can cause these problems.

No, I did not. In fact, I can say for sure that **OmenMon** _per se_ does not "cause" these problems since I've been running it for weeks now and do not have a single error like that in _my_ logs.

What _could_ be causing these is a race condition between different programs accessing the EC or trying to do so at the same time. Or, it _could_ just as well be an unrelated issue related to your setup or hardware.

> After testing, I found that even if the time interval for reading EC is continuously increased (within a reasonable time range of course), these problems still cannot be avoided.

"These problems" can probably be avoided by looking at what else is running on your system already.

Let's say you have an empty 0.75L bottle. You open a 0.33L can of _Pepsi_ and pour into the bottle. So far so good. So you open another 0.33L can of, say, _Dr Pepper_, and pour it there too. Things are still looking great. Then finally, you get yet another 0.33L can, this time with _Coke_ inside, and start pouring it in as well. But at some point the excess liquid begins to spill out of the bottle. Who's to blame for that? The _Coca-Cola_ company? After all, neither _Pepsi_ nor _Dr Pepper_ spilled out of the bottle. But beyond the superficial reason, maybe it's actually the arithmetics of 3 * 0.33 > 0.75?

> In fact, most users may only be concerned about CPU, GPU temperature, and fan speed control.

We have to take a step back here. I don't know about "most users" but neither do you. Let's just speak for ourselves. I'll start.

I wrote **OmenMon** for my use case, which is predicated on not having other programs accessing the EC in parallel. It solves the problem _I_ had (i.e. not wanting to have to run _Omen Gaming Hub_). _I_ like all the features _I_ put in there, and they serve _my_ needs.

I am not going to spend my time simply removing the functionality I've just added and that works for me perfectly well on the odd chance that it solves a poorly-defined problem related to some specific setup and interaction with other programs.

Let's put things in perspective. Even software that you have to pay for generally doesn't come with a promise it would solve _your_ specific problem and be rewritten for _your_ specific needs, or even with any expectation of support. All the more, this _free_ application I am sharing with you is not guaranteed to solve all the problems of all the people in the world trying to run it on all kinds of _HP_-branded hardware released throughout the years in conjunction with all kinds of other random software that might possibly interfere with it. At least not out of the box.

That being said, I wanted to share **OmenMon** with everyone, and I want to improve it for other users as well, and, as you can see from all the version and comment history, have spent significant time doing just this already. I welcome all feedback but I expect it to be _constructive_. _GitHub_ is supposed to be a place for somewhat technologically-savvy people: the more you demand, the more the onus is on _you_ to show me or at least tell me exactly how you'd do it better. Just because you have a chance to get something I made available that isn't exactly what you need or want, don't try to shame me into solving _your_ problem for _you_ by exaggerating the issue and claiming to speak for "most users."

Sorry I have to put it like that but certain things have to be said clearly so that we can have a productive discussion.

> Perhaps switches can be added in omenmon to allow users to choose to read only a few important parameters or all parameters themselves.

There are already lots of related configuration options, all [well-documented](https://omenmon.github.io/config): `EcFailLimit`, `EcMonInterval`, `EcMutexTimeout`, `EcRetryLimit`, `EcWaitLimit`, `GuiDynamicIcon`, `GuiDynamicIconHasBackground`, `UpdateIconInterval`, `UpdateMonitorInterval`, `UpdateProgramInterval`. Practically every aspect of EC interaction is user-configurable.

There's already the facility for temperature sensors to be model-specific (in [Hardware\Platform.cs:130](https://github.com/OmenMon/OmenMon/blob/master/Hardware/Platform.cs#L130), which I mentioned earlier). As long as all the people using **OmenMon** with a given model agree, the range of sensors can be reduced for their particular model. Having this configurable at runtime would incur extra work to implement and, most importantly, is not really guaranteed to solve the problem you described anyway. But I'm not saying no, although in this case I'd like to see a pull request.

Then, there's the fully-functional [CLI](https://omenmon.github.io/cli) mode as well, which can be used in cases where the GUI is doing too much for someone's needs. The CLI mode only does what you instruct it to and can be used in a script or called from another program. [Fan programs](https://omenmon.github.io/cli#prog) can be run from the CLI with `-Prog <Name>`. The EC is not queried for fan speeds in such a scenario.

All other options are on the table as well, which is why I created this issue in the first place. But if you want _me_ to do something, you'll have to make your case and convince _me_ how it actually improves things for "most users." And let's not forget I'm a user too.

[... cut ...]
It's a shame that people who pay real money for these laptops face the choice of either having to run an ad-ridden and telemetry-infested bloatware or not getting everything they should be able to out of the great hardware that's inside. At the very least, there is now a third option: **OmenMon**, even if it's with some hiccups and not for everyone.

@wangzhengbin, @Kubagf and everyone else who feels this way: I apologize **OmenMon** did not live up to your expectations. I appreciate that you gave it a chance, spent your time testing it, and shared your feedback.

GeographicCone, 2023-11-12:
Just for some further context, on my _Omen_ laptop where I was not just running **OmenMon** but developing and testing every build, which involves running code with genuine errors, and/or trying to make it fail in every possible way, I had, **since 2023-08-15**, when the OS was installed:

* **273** _ACPI_ warnings that "the EC controller returned data when none was requested."
* **203** _ACPI_ errors that "the EC controller didn't respond within the timeout."

This also includes all the other applications like _AIDA64_ or _HwInfo64_ since, believe it or not, they also generate these events, not even mentioning the possibility that some of these could equally be caused by the hardware, the drivers, or the OS itself, with no application to blame.

But let's assume they were <ins>all</ins> "caused" by **OmenMon** single-handedly. As far as I can tell, I started the developing it on 2023-09-04, working on it practically every day, which makes it **69 days today**. For the record, most of these events also happened in clusters when I was deliberately trying to overload the EC, but never mind that either. So, that's about **3 warnings** and **4 errors** per day. Is it a lot or not? You decide.

But let me first tell you that, over the same period, I've also had:
* **48** _Distributed COM_ errors about some service not being able to start
* **507** _Distributed COM_ warnings about some permissions not being granted

Arguably these are my fault, must be related to some _Windows_ component I removed. But it's not something I'd ever notice if I didn't look. For what it's worth, the OS runs faster without these missing components, even if it leads to an occasional error.

Then on top of that, I've had:

* **1,488** _Kernel-Processor-Power_ warnings that the CPU performance is being limited by firmware

Basically throttling at the power limit, something nobody can do anything about. But it's good to illustrate what counts as a "warning" and how important it can be.

And, finally, the icing on the cake:

* **94** _NDIS_ errors that the _Wi-Fi Direct Virtual Adapter_ "failed a power transition to operational power"
* **212** _WLAN-AutoConfig_ warnings that the _WLAN Extensibility Module_ `IntellIHVRouter12.dll` has stopped
* **1,741** `Netwtw12.sys` warnings for which the description cannot be found but that's again the most recent _Intel_ Wi-Fi driver for the newest iteration of their product (AX211).

So I guess it's high time to tell _Intel_ "I hope their Wi-Fi can completely replace Ethernet in the near future" but, right now, their drivers are just not up to my standards!

Now, if only someone can point me where I can speak to whoever's responsible at _Intel_ for their Wi-Fi driver so that we can have a similar discussion like here? Where is the "Does it work for you or not?" issue anyone can leave a comment on? Right, there isn't. It's take it or leave it and they don't care.

That's not how _I_ want it. I appreciate all the feedback. But I hope you get my point. Please be realistic about what you expect. The issue when the battery state changes on its own: like I said, it could be an EC conflict issue. Maybe I'm just lucky but it doesn't happen to me in normal usage at all. I was able to reproduce it exactly once by overloading the EC: the battery charge was briefly showing 0% for a second, then it returned to normal. Not the end of the world as far as I'm concerned. That's the trade-off with using the EC: there's only so much load it can sustain before things like this might start to emerge. But there is a certain number of operations per unit of time within which it does work reliably without hiccups. Based on my experience, I have reasons to think **OmenMon** steers clear of that limit. It is then for the user to decide if they want to stay within the safety margin, possibly limiting the use of other EC-accessing applications running concurrently, or run them all together even and possibly deal with the side effects.

I believe there's a real issue for at least one user here but there is also demonstrably no such issue for at least one other user (me). In view of these two facts together, it is difficult to see how **OmenMon** could be the sole cause. I am reluctant to remove functionality that I know to work well for me. Similarly, I cannot justify the significant additional workload of replacing the EC queries with other ways of obtaining temperature readings, although I would definitely consider any pull requests regarding this or other possible solutions. And of course, if anyone can point out how **OmenMon**'s EC implementation can be improved, I'm all ears.

I hope **OmenMon** works for everyone. However, if you have _Omen Gaming Hub_ installed, even if the UI is not being shown, some background processes are always active and you may run into issues. This might also be the culprit in this particular situation. I'm still interested in fixing any such issues but since that is not the expected use case, the burden is higher on anyone reporting these and expecting a fix. I hope you can agree that this is not unreasonable of me. Please be mindful of what the documentation has been saying all along: https://omenmon.github.io/more#considerations

> This application might not work well with other programs that access the Embedded Controller frequently. In particular, since it is intended as a replacement for the Omen Hub (Omen Control Center), it is not expected to run concurrently with it. For interactions with other applications, Embedded Controller footprint may be reduced by hiding the monitor panel window, and switching to a static notification icon.

I appreciate all the interest in **OmenMon**. I hope it serves you well. If it's not a good fit for your use case or hardware model, I'm sorry about it. If you have ideas how to change it and improve, I want to hear them but might not be able to act upon them, which I hope you can understand. **Thank you.**

n-elia, 2023-11-13:
Hi, I'm the Reddit guy asking support for a _Envy 15 15-ep0003nl_.

* OmenMon Version: **OmenMon-0.56.1-Preview**
* Model & Version: **HP 878D 18.33** 
* Working:
  * Keyboard-related features are correctly greyed out
  * GUI (with some exceptions listed below)
  * Fan Monitoring & Control: rpm readings, "Auto" modes (Performance, Default, Cool), "Maximum" mode," Off" mode
  * Graphic card power settings
* Not Working:
  * When applying a Fan Mode, the error [1] is raised

<details>
  <summary>Error log [1]</summary>

```
************** Testo dell'eccezione **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 46
   in OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   in OmenMon.Hardware.Bios.BiosCtl.SetFanLevel(Byte[] data)
   in OmenMon.Hardware.Platform.FanProgram.Update()
   in OmenMon.Hardware.Platform.FanProgram.Run(String name)
   in OmenMon.AppGui.GuiFormMain.EventActionFanSet(Object sender, EventArgs e)
   in System.Windows.Forms.Control.OnClick(EventArgs e)
   in System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   in System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   in System.Windows.Forms.Control.WndProc(Message& m)
   in System.Windows.Forms.ButtonBase.WndProc(Message& m)
   in System.Windows.Forms.Button.WndProc(Message& m)
   in System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```
</details>

GeographicCone, 2023-11-13:
@n-elia Great to see you here!

@P4R1H _[Re: Double post]_ No problem, just had to mention it for full disclosure.

----

@n-elia and @P4R1H Thank you both for the reports! Seems it's about the same issue in both situations: setting the fan levels, so I'll answer you jointly, hope you don't mind.

**There's another way to set the fan levels using the EC. Maybe it works for you. Can you please try the latest build?**
https://github.com/OmenMon/OmenMon/releases/tag/0.56.3

**What changed:** there will always be an error check now after trying to set fan levels using the BIOS, even if `<BiosErrorReporting>` is set to **false** (actually, that error should not have happened with `<BiosErrorReporting>` set to **false** previously, not really sure why it did but anyway, it'll be irrelevant following this change).

Then, if the BIOS call failed, an attempt is made to set each fan level individually by writing to the Embedded Controller registers `SRP1` and `SRP2` (`0x34` and `0x35`) respectively.

Before that happens, fans are also put into manual mode by writing `0x06` to `OMCC` (register `0x62`). On my laptop, this is not necessary: setting the fan levels works regardless. But I've seen it mentioned as a prerequisite on some other models, so it's being done as well, just to be sure.

Here's a summary of what I know about the Embedded Controller, if anyone's interested:
https://github.com/GeographicCone/OmenHwCtl/blob/master/Reference/EC_8A14.txt

**General note:** This build incorporates the changes I made in the previous one in order to reduce the I/O intensity when interacting with the Embedded Controller. If the EC is being used by other applications, it's now less likely to cause system-wide side effects (such as _Event Log_ messages) but it's more likely to result in an in-application error pop-up: "failed to acquire Embedded Controller exclusive lock." The update interval is also increased from 1 to 3 seconds.

I am still undecided whether to keep these changes or revert them, so if you have any opinion about it, please let me know. In any case, these are just the defaults, and they can be adjusted back to the previous ones using the configuration settings.

dd871, 2023-11-13:
> @dd871
>
> To clarify first of all, **OmenMon** only _writes_ to the EC in three situations now:
>
> * Extending countdown timer when a fan program is running
> * Switching fans off and back on when set [...]
<...>

@GeographicCone 
Thank you very much for the detailed and thoughtful reply. I believe I now understand the whole architecture a lot better. Based on this, I stopped monitoring the EC registers in a second window and found another one or two Omen/HP processes which I had earlier not yet stopped. In addition, I also doubled the Mutex timeout value (not sure though if that part mattered).

After the above, it now runs relatively stable, I have seen the low battery message only once since.

I do still have the glitch though that only happens for fan 0 (CPU fan). I thought I illustrate it with screenshots. As per the below, the fan level is (almost) completely constant, but the fan rate (%) displayed for the CPU fan rate jumps all over the place. This is also happening continuously, meaning if constantly flicks between correct rate and the wrong rate, every couple of seconds.

[image]

[image]

n-elia, 2023-11-13:
> @n-elia and @P4R1H Thank you both for the reports! Seems it's about the same issue in both situations: setting the fan levels, so I'll answer you jointly, hope you don't mind.

Uhm... I will try and let you know. However I might be unclear in my previous post: the fan control actually works despite the error being raised. I can feel the changes, and I can also manually control the fan speed, despite the error popup.
Should I still try this new build?

GeographicCone, 2023-11-13:
@dd871 Thank you!

> I also doubled the Mutex timeout value

Higher [EcMutexTimeout](https://omenmon.github.io/config#ecmutextimeout) values could lead to an unresponsive application UI, and cause system-wide side effects because of a longer period when the EC is being continuously accessed by applications, which increases the risk of interfering with a hardware operation that also wants to use the EC during this period.

On the other hand, lower values may lead to application-level error pop-ups: "failed to acquire Embedded Controller exclusive lock" when the EC is also being used by other processes.

So, pick your poison… ideally, you'd most likely want to lower the value until you start seeing application errors, and then increase it back just a bit. On the other hand, noticeable errors are annoying, maybe it's better to just have a message about it going into the _Event Log_.

> The fan level is (almost) completely constant, but the fan rate (%) displayed for the CPU fan rate jumps all over the place. This is also happening continuously, meaning if constantly flicks between correct rate and the wrong rate, every couple of seconds.

Broadly, there is the issue with the reliability of the data being retrieved from the EC. So far, like I wrote before, any values that are obviously outside the bounds of credibility are rejected.

The remaining issue is that even some readings that would normally be believable in isolation, are clearly wrong: like in your example, the fans can't go from 30% to 50% in 1 second if that's what's happening. Looking at the value, we can immediately tell it must be off by how much it's changed. But the question is how to make it into an algorithm without increasing the EC load while also handling all the edge cases.

For example, as we increase [UpdateMonitorInterval](https://omenmon.github.io/config#updatemonitorinterval), which I've now bumped up to 3 seconds by default, it becomes increasingly difficult to be sure what happened in between. 1 second is clearly not enough but could it get from 30% to 50% in 3 seconds? Maybe. And then there're the issues like: what if the wrong number is the _first_ reading, right after monitoring starts. Then the second, correct reading is rejected for being too off from the erroneous one. What then? Handling all this in an elegant way is more challenging than it sounds.

And on top of that, there's the philosophical question of what **OmenMon** should and should not do: one way of looking at it is that it's just a conduit for reporting EC data to the user, who can then decide what to do about it. Since these fluctuations don't happen all the time (at least not in my experience), it can also be an indication that something's going on when they do. Something the user might want to be aware of. By discarding it, we're actively making effort (using CPU cycles and possibly extra EC I/O) to withhold information from the user, which might mean something to him (e.g. all this happens when another application was launched).

By comparison, another approach would be that **OmenMon** should validate all the information before reporting it to the user, so when it briefly reports a 50% fan rate instead of the actual 30%, it's an error of inaccurate reporting, as opposed to properly and accurately reporting possibly unreliable information retrieved from EC. This approach in turn begs the question if the user should be notified that some of the data was rejected as invalid, and if yes, in what way should that happen.

I don't have a strong position on this issue, so beyond getting rid of the obviously erroneous readings like 15,000 rpm, I just left things as they were for the time being. Opinions welcome.

n-elia, 2023-11-13:
Uhm... sorry if I was unclear.

Let's first clarify that everything in the following message refers to:
* OmenMon Version: **OmenMon-0.56.1-Preview**
* Model & Version: **HP 878D 18.33** 

> @n-elia So that we're on the same page, are you saying that setting arbitrary fan speeds (the _Const_ mode) works for you despite the error?

Yes.

> Because if you mean that _Auto_, _Max_, _Off_ modes work, these are actually controlled by different BIOS settings. It's possible that setting the maximum speed or turning the fans off would work on some models but more specific fan-level setting would not.
> 
> Please try the new build anyway, and can you also try setting fan levels manually using the CLI mode with `OmenMon -Bios FanLevel=30,30` (just an example, try a couple of different values) to make sure it really works? Details here: https://omenmon.github.io/cli#fanlevel

The result of issuing `OmenMon -Bios FanLevel=30,30` is that the change actually happens and the log is:

```
OmenMon Hardware Monitoring & Control Utility Version 0.56-Preview
BIOS (-Bios)
+ Fan #1 Level: 0x1e = 0b00011110 = 30 [Cpu]
+ Fan #2 Level: 0x1e = 0b00011110 = 30 [Gpu]
Expected a pair of fan speed levels: <Fan1>,<Fan2> (<Fan#>: 0-255|0x00-0xFF|0b00000000-0b11111111)
```

Here is the step-by-step test procedure I did for testing, and the related results:

- I open the GUI.

- I click Max, then the Apply button, then the fans go max power. (No errors raised, and works)

- I click Off, then the Apply button, then the fans go off. (No errors raised, and works)

- I click Const, then the Apply button, then the fans go off (because the sliders are at 0 by default). (No errors raised, and works)
- I raise the CPU slider and click Apply, then the error [1] is raised. The CPU fan goes to the target speed. (Errors raised, but works)
- I raise more the CPU slider and click Apply, then the error [1] is raised. The CPU fan goes to the target speed. (Errors raised, but works)
- I lower the CPU slider to 0 and click Apply. Then the CPU fan remains spinning to the previous speed. (No errors raised, but doesn't work)

- I click Auto, I select  "Performance" and press the Apply button, then the Performance profile is selected (I can understand that it is not working though, because the fan speed is lower than expected), and the error [2] is raised. (Errors raised, doesn't work)

- (At this point, if I right-click the **tray icon**, it shows Fan "Default". If I select "Performance" mode (from the tray icon), it works without any error.)

- Coming back to the GUI. I select "Prog", choosing "Power" profile, and I click Apply. Then, the error [3] is raised. The CPU fan changes speed, but I can't tell if it is working. Taking a look to the temperatures of "Power" profile in the .xml config file, it seems that it is working. (Errors raised, but seems working)

- However, at this point, the error [4] is raised every (more or less) 30 seconds.

<details>
  <summary>Error log [1]</summary>
```
************** Testo dell'eccezione **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 46
   in OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   in OmenMon.Hardware.Bios.BiosCtl.SetFanLevel(Byte[] data)
   in OmenMon.AppGui.GuiFormMain.EventActionFanSet(Object sender, EventArgs e)
   in System.Windows.Forms.Control.OnClick(EventArgs e)
   in System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   in System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   in System.Windows.Forms.Control.WndProc(Message& m)
   in System.Windows.Forms.ButtonBase.WndProc(Message& m)
   in System.Windows.Forms.Button.WndProc(Message& m)
   in System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```
</details>

<details>
  <summary>Error log [2]</summary>
```
************** Testo dell'eccezione **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 46
   in OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   in OmenMon.Hardware.Bios.BiosCtl.SetFanLevel(Byte[] data)
[... 22 lines cut ...]
   in System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   in System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   in System.Windows.Forms.Control.WndProc(Message& m)
   in System.Windows.Forms.ButtonBase.WndProc(Message& m)
   in System.Windows.Forms.Button.WndProc(Message& m)
   in System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```
</details>

<details>
  <summary>Error log [4]</summary>
```
************** Testo dell'eccezione **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Unknown response from BIOS: 46
   in OmenMon.Hardware.Bios.Bios.Check(Int32 code)
   in OmenMon.Hardware.Bios.BiosCtl.SetFanLevel(Byte[] data)
   in OmenMon.Hardware.Platform.FanProgram.Update()
   in OmenMon.AppGui.GuiTray.Update()
   in System.Windows.Forms.Timer.OnTick(EventArgs e)
   in System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   in System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```
</details>

I will test the new version asap. Thank you!!!

Rupurudu, 2023-11-17:
* OmenMon Version: **[0.56.3](https://github.com/OmenMon/OmenMon/releases/tag/0.56.3)**
* Model & Version: **HP Victus 16-e0008nt (4H0S0EA)** 
* Working:
  * Fan Monitoring & Control (Auto/Max/Const/Off) 
  * GPU Power Mode
  * Intercept Omen key
* Not Working:
  * Temp Sensor Readings: Sensor readings are all over the place, sometimes some of the sensors work, sometimes they stop working and other sensors start working. Usually CPUT, GPTM, RTMP, TMP1, TNT4 works. Sometimes TNT2 and TNT5 works too. But usually when they do I lost CPUT reading.
  * AC Power usage does not work and always shows as "0W AC Power OK"
  * Thermal Throttling status does not work
  * Commandline fails unless "BiosErrorReporting" is set to false
 
GUI:
[image]

BIOS with BiosErrorReporting=true
```D:\Desktop\OmenMon-0.56.3-Preview>omenmon -bios
OmenMon Hardware Monitoring & Control Utility Version 0.56-Preview
BIOS (-Bios)
- Manufacturing (Born-On) Date: 20220803 [YYYYMMDD]
- System Design Data: c8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [↓]
- Status Flags: 0x00c8 = 0b0000000011001000 = 200 [BiosPerfPpab3, BiosPerf6, BiosPerf7]
- Unknown Byte: 0x00 = 0b00000000 = 0 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x00 = 0b00000000 = 0 [V0]
- Support Flags: 0x00 = 0b00000000 = 0 [0]
- Default CPU Power Limit 4: 0x00 = 0b00000000 = 0 [W]
- BIOS-Defined Overclocking: 0x00 = 0b00000000 = 0 [NotSupported]
- Graphics Mode Switching Support: 0x00 = 0b00000000 = 0 [0]
- Default CPU Concurrent Power Limit w/GPU: 0x00 = 0b00000000 = 0 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x01 = 0b00000001 = 1 [On]
- GPU Processing Performance AI Boost (PPAB): 0x01 = 0b00000001 = 1 [On]
- GPU Device Power State (DState): 0x01 = 0b00000001 = 1 [D1]
- GPU Peak Temperature Sensor Threshold: 0x00 = 0b00000000 = 0 [°C]
- Graphics Mode (Legacy): 0x00 = 0b00000000 = 0 [Hybrid]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
Exception: BIOS call failed: Command not available

OmenMon: Void Check(Int32, Boolean)
   konum: OmenMon.Hardware.Bios.Bios.Check(Int32 code, Boolean force)
   konum: OmenMon.Hardware.Bios.BiosCtl.HasOverclock()
   konum: OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   konum: OmenMon.AppCli.CliOp.BiosGet[TResult](String message, Func`1 biosMethod, String extra)
   konum: OmenMon.AppCli.CliOp.Loop(String[] args)
   konum: OmenMon.AppCli.CliOp.Loop(String[] args)
   konum: OmenMon.App.Main(String[] args)```

BIOS with BiosErrorReporting=false
```OmenMon Hardware Monitoring & Control Utility Version 0.56-Preview
BIOS (-Bios)
- Manufacturing (Born-On) Date: 20220803 [YYYYMMDD]
- System Design Data: c8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [↓]
- Status Flags: 0x00c8 = 0b0000000011001000 = 200 [BiosPerfPpab3, BiosPerf6, BiosPerf7]
- Unknown Byte: 0x00 = 0b00000000 = 0 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x00 = 0b00000000 = 0 [V0]
- Support Flags: 0x00 = 0b00000000 = 0 [0]
- Default CPU Power Limit 4: 0x00 = 0b00000000 = 0 [W]
- BIOS-Defined Overclocking: 0x00 = 0b00000000 = 0 [NotSupported]
- Graphics Mode Switching Support: 0x00 = 0b00000000 = 0 [0]
- Default CPU Concurrent Power Limit w/GPU: 0x00 = 0b00000000 = 0 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x01 = 0b00000001 = 1 [On]
- GPU Processing Performance AI Boost (PPAB): 0x01 = 0b00000001 = 1 [On]
- GPU Device Power State (DState): 0x01 = 0b00000001 = 1 [D1]
- GPU Peak Temperature Sensor Threshold: 0x00 = 0b00000000 = 0 [°C]
- Graphics Mode (Legacy): 0x00 = 0b00000000 = 0 [Hybrid]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
- Overclocking Support: 0x00 = 0b00000000 = 0
- Memory Overclocking Support: 0x00 = 0b00000000 = 0
- BIOS Undervolt Support: 0x00 = 0b00000000 = 0
- Keyboard Type: 0x00 = 0b00000000 = 0 [Standard]
- Keyboard Backlight Support: No
[... 32 lines cut ...]
# 13: Cpu 5800 rpm Gpu 5500 rpm @ 50 °C
- Temperature: 0x30 = 0b00110000 = 48 [°C]
- Thermal Throttling Status: 0x00 = 0b00000000 = 0 [Unknown]```

EC:
```D:\Desktop\OmenMon-0.56.3-Preview>omenmon -Ec
OmenMon Hardware Monitoring & Control Utility Version 0.56-Preview
Embedded Controller (-Ec)
0x _0 _1 _2 _3 _4 _5 _6 _7 _8 _9 _A _B _C _D _E _F
0_ 00 00 00 00 00 00 4f 4e 30 37 30 58 4c 2d 41 00
1_ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2_ 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 30 31
3_ 00 00 6a 00 00 ff 44 00 00 00 00 00 00 00 00 00
4_ 0b 01 04 14 04 10 00 00 00 2b 2b 00 00 00 00 00
5_ 90 00 00 0f 00 03 00 31 31 2a 00 00 00 00 00 08
6_ 00 00 06 16 00 00 00 00 00 74 00 0a 00 0c 00 00
7_ c6 11 23 0e 28 3c 80 33 31 00 2d 31 44 2d 34 42
8_ 2d 41 00 00 00 23 0e 81 42 00 00 00 01 e0 00 00
9_ d1 d1 23 0e 05 31 00 00 00 00 00 00 00 c6 0b 00
a_ 03 80 00 88 01 02 00 00 00 33 00 00 00 01 00 01
b_ 9a 08 4e 09 00 c0 5e 2d 00 00 00 07 00 00 00 00
c_ 03 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00
d_ 64 00 00 00 09 05 00 01 64 00 1d 03 00 00 00 00
e_ 00 00 00 00 00 00 00 00 00 00 00 00 00 68 00 00
f_ 08 00 20 00 00 00 00 00 69 00 38 30 37 32 30 31```

P4R1H, 2023-11-17:
> @n-elia Great to see you here!
>
> @P4R1H _[Re: Double post]_ No problem, just had to mention it for full disclosure.
>
> @n-elia and @P4R1H Thank you both for the reports! Seems it's about the same issue [...]

Great! all the features seem to work as expected now, along with the sliders

when i switch between legacy performance modes i can see the wattage changing too

GeographicCone, 2023-11-19:
**New build [0.57.0](https://github.com/OmenMon/OmenMon/releases/latest) posted with some relatively big changes.** Hope it works well for everyone.

----

@Rupurudu Thank you very much for trying **OmenMon**, and for your report!

In the [latest build 0.57.0](https://github.com/OmenMon/OmenMon/releases/latest) I disabled BIOS error check for [HasOverclock](https://omenmon.github.io/cli#hasoverclock) and some other queries, so maybe it works for you now with [BiosErrorReporting](https://omenmon.github.io/config#bioserrorreporting) set to **true**.

Temperature readings are just what the EC reports. That not all of them are active all the time is how things are. Broadly, there is the open question how to handle them better (see my [earlier post](https://github.com/OmenMon/OmenMon/issues/1#issuecomment-1808149010)), which also relates to the issue with [improving fan programs](https://github.com/OmenMon/OmenMon/issues/9). So it's on the roadmap.

----

@n-elia and @P4R1H In the [latest build 0.57.0](https://github.com/OmenMon/OmenMon/releases/latest) I changed the logic for fan level settings yet again, so you'll probably have to adjust your configuration. But on the plus side, they hopefully now cover all cases.

With [FanLevelNeedManual](https://omenmon.github.io/config#fanlevelneedmanual) set to **false**, and [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec) also **false** (the default), it just makes the BIOS WMI call. Errors are ignored to cover the situation when the settings still take effect, even though the BIOS returns an error. You can try using `OmenMon -Bios` [FanLevel](https://omenmon.github.io/cli#fanlevel) in the command-line mode, and an error will be reported if there is one (the message is not accurate, I forgot to update it this time).

If this doesn't work, you can set [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec) to **true** to bypass the WMI BIOS call entirely and use the EC.

Finally, in either mode you can set [FanLevelNeedManual](https://omenmon.github.io/config#fanlevelneedmanual) to **true** to enable the additional first step of writing `0x06` to the `OMCC` EC register before setting the fan levels, which is required on some models but not all (previously, this was always done).

----

@P4R1H

> i'd expect the silent mode to be well, silent, but it doesn't seem to reduce the fan speed

These example modes are not really that well-researched, you should experiment with your own, model-specific settings. With that in mind, the reason it felt like it didn't work might have been that the preset included setting both fans to 0, and this is not allowed by the hardware if the BIOS setting for _Fan Always On_ is enabled (the default).

It's still possible to set both fans off completely using a different command, which is used by the _Off_ and _Const_ modes, but I didn't want it done from within a fan program due to possible long-term implications.

So, basically you either have to disable _Fan Always On_ in the BIOS, or change the preset to minimum speed for at least one fan: I've done the latter now in the example configuration file, setting the CPU fan to 21 [krpm], as that seems to be the minimum on my system.

@ArmynC Forgot to comment on this previously:

> 'Optimus' flag is case-sensitive (won't accept it). Since anything but the arg aren't case-sensitive, it would be nice to have it as such, too.

Yeah, you're right, it'd just be surpringly cumbersome to allow for that, which is why I decided against it. But I think the documentation mentions these identifiers are all case-sensitive.

----

I think I'm all caught up with the feedback but if I forgot to answer someone, please remind me. 

Feel free to post model-specific reports and information as a separate issue (one per model please, so if one exists already, reuse it), adding the label _Model Info_ to them and using the following template:

````
[8xxx] HP Model Name xxxxx (YYYY)
````

n-elia, 2023-11-20:
> @n-elia and @P4R1H In the [latest build 0.57.0](https://github.com/OmenMon/OmenMon/releases/latest) I changed the logic for fan level settings yet again, so you'll probably have to adjust your configuration. But on the plus side, they hopefully now cover all cases.
> Feel free to post model-specific reports and information as a separate issue

Thank you for the new build! I tested it out, default configs are okay in my case. I'm posting a report to help new users. Bye!

P4R1H, 2023-11-21:
> **New build [0.57.0](https://github.com/OmenMon/OmenMon/releases/latest) posted with some relatively big changes.** Hope it works well for everyone.
>
> @Rupurudu Thank you very much for trying [...]

My fans always on has been disabled since day 1 though
Also thanks for the quick help and fixes

[22 more replies]