OmenMon/OmenMon #2: Source code is not available ?

ramius86, 2023-11-06:
Hi,

thanks for this, very appreciated from an ex-user of G-Helper with my previous Asus ROG laptop.

I was trying to figure out some things and noticed that the source code isn't public available, I can see only readme.md ( also if I download the releases source code.zip ), is this intended?

GeographicCone, 2023-11-06:
It'll be posted eventually, I just have to figure out some organizational stuff (basically, the license type, and whether it should be under my real name), and haven't had the time to think about this yet.

Meanwhile, there's the documentation at https://omenmon.github.io/ or, if you have a specific question, shoot.

closed 2023-11-07 by ramius86

GeographicCone, 2023-11-08:
Full source code is available now, just to let you know.