OmenMon/OmenMon #3: Support for per-key-RGB?

dd871, 2023-11-08:
Amazing work - I finally have some hope to be able to drop the awful OGH application!

Currently, the app supports the zoned keyboards. I was wondering if it would be possible at all to include support for per-key-RGB keyboards as well?

I understand if that would be too complex to do, especially if you dont actually have a laptop with it yourself, but how about disabling the RGB functionality (menu options, keyboard GUI, etc.) completely if it is not supported? As currently it might be a bit confusing that all the options are shown but have no effect.

Thanks!

GeographicCone, 2023-11-09:
> Amazing work

Thank you!

> I was wondering if it would be possible at all to include support for per-key-RGB keyboards as well?

It's not impossible but it'd take quite a bit of work. And it's not something I could realistically do myself, given I don't even have the hardware.

To the extent that I can tell, the per-key RGB keyboard must be controlled in a different way: the 4-zone color information data structure is only 128 bytes long: that's not enough space to fit all the per-key RGB (3-byte) color information for 80+ keys.

If there's an extra controller for it, meaning it's handled outside the Embedded Controller and/or the WMI BIOS calls, then it might already be documented elsewhere. But I'd have to start by looking at the hardware to confirm it, and possibly also at how it's being done in the OEM application. In any case, it's not a matter of just writing the code for a known interface.

Do you have a _Primax MCU_ device with PCI vendor ID `0461` and product ID `4E9B` or `4ECA`? This might be that extra controller. You can check with: `devcon find *VEN_0461` if you have `devcon.exe`, otherwise `wmic path win32_pnpentity where "deviceid like '%VEN_0461%'" get name,deviceid` (or `pnputil /enum-devices | find "VEN_0461"` but in _PowerShell_ you have to escape the quotes so ``pnputil /enum-devices | find `"VEN_0461`"``).

> How about disabling the RGB functionality (menu options, keyboard GUI, etc.) completely if it is not supported? As currently it might be a bit confusing that all the options are shown but have no effect.

That's definitely doable. Can you confirm on your system what the values are for:

* `OmenMon -Bios KbdType` (is it `0x03` == `PerKeyRgb`)
* `OmenMon -Bios HasBacklight`

And if backlight switching works for you? (Either from the GUI or with `OmenMon -Bios Backlight=On` or `Off`.)

If you could spare a moment on this, could you also run `OmenHwCtl -GetColorTable` ([OmenHwCtl](https://github.com/GeographicCone/OmenHwCtl) is **OmenMon**'s humble predecessor) and see if the table is anything other than zeroes? If so, does it change when you change the colors?

Ideally, I'd also hope to find someone with a single-color keyboard. I understand the color is either permanently red or white there, depending on the hardware version, and can't be changed. I'd like to know how to detect this variant so that both cases can be handled at the same time.

_If anyone else has any information regarding any of this, feel free to chip in._

bbaallance, 2023-11-09:
For omen17-ck1000, there are two ways to control per-key-RGB. The first way is to use OGH. The second way is to use omen light studio(OLS). Ogh only provides several modes. So I perfer using ols. 
Here is the screenshot:
[image]
Considering that you found many useful things in the file directory of OGH before. I think there may also be something useful  in OLS.
: )

dd871, 2023-11-10:
> Do you have a _Primax MCU_ device with PCI vendor ID `0461` and product ID `4E9B` or `4ECA`? This might be that extra controller. You can check with: `devcon find *VEN_0461` if you have [...]

Doesnt look like it:
--> wmic path win32_pnpentity where "deviceid like '%VEN_0461%'" get name,deviceid
No Instance(s) Available.
--> pnputil /enum-devices | find `"VEN_0461`"
(no value returned)

> That's definitely doable. Can you confirm on your system what the values are for:
> * `OmenMon -Bios KbdType` (is it `0x03` == `PerKeyRgb`)

It is indeed:
`Keyboard Type: 0x03 = 0b00000011 = 3 [PerKeyRgb]`

> * `OmenMon -Bios HasBacklight`

Returns:
`Keyboard Backlight Support: No`

> And if backlight switching works for you? (Either from the GUI or with `OmenMon -Bios Backlight=On` or `Off`.)

Does not work. It returns "Keyboard Baclight: Yes / No" respectively, but there is no change to the actual backlight status.

> If you could spare a moment on this, could you also run `OmenHwCtl -GetColorTable` ([OmenHwCtl](https://github.com/GeographicCone/OmenHwCtl) is **OmenMon**'s humble predecessor) and see if the table is anything other than zeroes? If so, does it change when you change the colors?

Initially shows this:
`Get Color Table
+ OK: 030000000000000000000000000000000000000000000000000080ff00ff0000ff00ffffff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000`

 Then, after changing color of every key in OGH, it returns exactly the same values, no changes in the above.

Thanks for looking at this, but sounds like it would be quite difficult to do. Maybe something for the future, once all the other features are mature.

GeographicCone, 2023-11-10:
@dd871

> Initially shows this:
> Get Color Table
> `030000...`
> Then, after changing color of every key in OGH, it returns exactly the same values, no changes in the above.

OK. So basically there is the whole interface for 4-zone color and backlight control, except it doesn't actually control anything (`03` would be the number of zones). The per-key RGB keyboard settings are handled in a completely different way.

> How about disabling the RGB functionality (menu options, keyboard GUI, etc.) completely if it is not supported? As currently it might be a bit confusing that all the options are shown but have no effect.

Added in 0.55.2.

maltyjava, 2024-01-23:
Hi, I have a 17-ck2001na with per-key RGB. Omenmon GUI still shows the unselectable keyboard options though, should it hide automatically? I couldn't find a config option to disable it. Thanks!

GeographicCone, 2024-01-23:
I don't know what you're seeing. It should show all keyboard-related controls grayed out and disabled, and this static picture: 

[image]

There are no configuration settings for this. It depends on automatic detection:

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Hardware/Settings.cs#L189-L201

If your keyboard type is reported as `PerKeyRgb` it's considered the equivalent of no backlight support:

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Hardware/BiosData.cs#L92-L98