OmenMon/OmenMon #4: Linux support

ArmynC, 2023-11-09:
Thinking about the fact that there probably aren't any compatible Omen Hubs for Linux, wouldn't it be possible in the future to make a build for this operating system? Maybe with a transition to the Qt framework. The BIOS calls should be the same.

GeographicCone, 2023-11-09:
Somebody [asked about this](https://old.reddit.com/r/HPOmen/comments/17oet58/omenmon_fan_gpu_keyboard_performance_monitoring/k804too/) in the [r/HPOmen](https://old.reddit.com/r/HPOmen/) subreddit. Thing is, these HP BIOS calls depend on _Windows Management Instrumentation_ (WMI). Not sure what an equivalent way of doing all that in Linux would be.

It should still be possible to do quite a lot using the Embedded Controller though. Much of the fan control currently being handled through the BIOS interface can alternatively be done via the EC (it's just a design preference to use the "native" interface). And the venerable [Notebook Fan Control](https://github.com/hirschmann/nbfc), which controlled fans using the EC, had a Linux version too.

ArmynC, 2023-11-09:
Oh, I see. That's a bit sad. Even if I'm not using this OS (at the moment), I'm pretty sure the Linux community would use it more so than the Windows community, since there's already an full-fledged app for Windows. Maybe a way of emulating WMI (or even trough Wine?) - sadly I'm _no expert_ on this one.

_Referencing (for future):_

https://stackoverflow.com/questions/921617/is-there-something-like-wmi-for-linux
https://download.lenovo.com/pccbbs/mobiles_pdf/lenovo_bios_setup_linux_wmi_sysfs_1.2.pdf
https://docs.kernel.org/hwmon/hp-wmi-sensors.html

GeographicCone, 2024-02-13:
I've noticed this link from the Arch Linux Wiki has appeared in the referral logs:
https://wiki.archlinux.org/title/HP_Omen_16-c0140AX

It has a good write-up on how to use the WMI-related features on Linux with these laptops.

realazizbek, 2024-07-06:
Hello, is there any changes for linux users? Is there any posibility thay you would make it to Linux?

lyrnya, 2024-07-14:
There's a script on GitHub that you can use to control the fan
https://github.com/alou-S/omen-fan