OmenMon/OmenMon #5: Undervolting feature

ArmynC, 2023-11-09:
If possible, a nice feature is integrating the undervolting component, as Omen Gaming Hub has it. A smaller counterpart of the _Throttlestop_ app.

[image]

GeographicCone, 2023-11-09:
The (unstated) mission of **OmenMon** is to focus on things that otherwise can't be done without _Omen Hub_. And, as you said, _ThrottleStop_ and _XTU_ can do this already. 

That being said, I'm open to all ideas. The source code is out, and the automated workflow is in place so that new builds can be done with zero extra effort, so everything is on the table.

ArmynC, 2023-11-09:
Yeah, but ThrottleStop is something local(?). I see that Undervolting values are injected in BIOS (I may be very wrong on this one). Also, a simpler UI is needed, ThrottleStop is very advanced.

Dragofagnir, 2023-11-12:
Well the problem is that on 12th and 13th gen of hx processors, undervolting cant be done without the omen gaming hub, fivr tab is locked on throttlestop and voltage control locked on xtu

GeographicCone, 2023-11-12:
@Dragofagnir Have you tried enabling it in the hidden UEFI settings? I have this in `CpuSetup`:

````
OneOf Prompt: "UnderVolt Protection", Help: "When UnderVolt Protection is enabled,
user will not be able to program under voltage in OS runtime. Recommended to keep
it enabled by default. Enabled: Allow BIOS undervolting, but enable UnderVolt
Protection in Runtime. Disabled: No UnderVolt Protection in Runtime.", 
QuestionFlags: 0x10, QuestionId: 0x382, VarStoreId: 0x2, VarOffset: 0x378,
Flags: 0x10, Size: 8, Min: 0x0, Max: 0x1, Step: 0x0
	OneOfOption Option: "Disabled" Value: 0
	OneOfOption Option: "Enabled" Value: 1, Default, MfgDefault
End 
````
There are a couple of other settings related to _Overclocking_ and _Cfg Lock_. Can't say for sure if enabling them is enough, since what I've got is an **i7-12700H**, so there really is no hope.

Dragofagnir, 2023-11-12:
Well 12700h can be undervolted by messing with LLC like msi laptops do, but the probkem is that hp bios is write protected and also flashing a modified bios will just brick the laptop

GeographicCone, 2023-11-12:
@Dragofagnir 
> hp bios is write protected and also flashing a modified bios will just brick the laptop

You don't have to modify the BIOS, just boot into EFI shell and edit the settings from there: https://github.com/datasone/setup_var.efi

Bedr0s, 2023-11-20:
Hi guys, i've tried to unlock undervolt by UEFI withouth success  using "setup_var CpuSetup "VarOffset value" 0x0"; then I checked and the value was changed to 0x0 but the undervolt functions are still disabled in both Throttlestop and XTU, could someone help me please?

GeographicCone, 2023-11-20:
@Bedr0s Which CPU model?

For example with my i7-12700H, it's locked at the CPU level with no way to unlock it, apparently.

Bedr0s, 2023-11-20:
My model is the same i7 12700h, I asked to unlock to intel and Asus but they said that is not possible, but I think that via uefi should be possible to Undervolt someway maybe

GeographicCone, 2023-11-20:
@Bedr0s Not that I know of. It's locked at an even lower level than UEFI. It's Intel, and they can do that.

I wanted to post a link to the ThrottleStop message board but I see you got the same answer there as well.

Bedr0s, 2023-11-20:
Yes on throttlestop community one of author said that form 12th gen onwards fivr option is  not available and not locked, so it seems impossible to Undervolt these new generations of CPU if you don't have an MSI motherboard.

GeographicCone, 2023-11-23:
I've also been interested in this topic, so to sum up what I know so far:

For the **i7-12700H**, you can lower the [AC loadline from the default 2.3](https://edc.intel.com/content/www/us/en/design/ipla/software-development-platforms/client/platforms/alder-lake-desktop/12th-generation-intel-core-processors-datasheet-volume-1-of-2/010/vcccore-dc-specifications/#GUID-72B6320F-EC78-47E8-802A-D4E16E520955) to 1.4 [mΩ] for a maybe 5% reduction in power under load. This can be done with my [UEFI Variable Tool](https://github.com/GeographicCone/UefiVarTool) from EFI shell: `uvt CpuSetup:0x0132(2)=140` or any other utility for editing EFI variables. There's a [long thread](https://old.reddit.com/r/XMG_gg/comments/11f8n0z/launch_undervolting_via_ac_loadline_in_xmg_and/) on Reddit about this setting.

On TechPowerUp, there's [another long thread](https://www.techpowerup.com/forums/threads/cpu-i7-12700h-ring-edp-other-red-dell-g15-5520-motherboard-05j7dy.312420/) what to do about ring EDP throttling (with [ThrottleStop](https://www.techpowerup.com/download/techpowerup-throttlestop/)): lowering down _FIVR_ → _Cache Ratio_ to 36 from the default 40 makes it go away at a performance cost. Other ideas include setting _Speed Shift_ to 4 minimum and 52 maximum, at a bigger performance cost, or increasing Icc<sub>MAX</sub> from the default 160 [A].

As for unlocking, these settings enable CPU configuration on the earlier firmware versions for my model (updates until September 2022, with ME version 16.**0**.x.x):

````
CpuSetup:0x0043=0  # Cfg Lock: No
CpuSetup:0x01D9=1  # Overclocking: Yes
CpuSetup:0x010E=0  # Overclocking Lock: No
````

Starting from October 2022, the firmware with a new ME version 16.**1**.x.x added the following two:

````
CpuSetup:0x0378=0  # Undervolting Lock: No
CpuSetup:0x0379=0  # (Missing description, should be "Undervolt Protection"): No
````

The second option is missing title and description even in the hidden settings but I'm fairly sure that it's supposed to be the _Dynamic Undervolt Protection_ newly-added by Intel. Unfortunately even changing that value to `0` from `1` (which it is by default) doesn't seem to disable it: [HwInfo](https://www.hwinfo.com/) still reports _Dynamic Overclocking Undervolt Protection: Enabled_.

For the **i7-12700H** it's largely a moot point since it can't be undervolted anyway but if someone has an -H**X** CPU (I think at least some _Omen 17_ models came with these), seems it might take a BIOS downgrade to fully unlock the undervolting settings.

Note: I _think_ the offsets within the `CpuSetup` UEFI variable are the same for all _Alder Lake_ CPUs, regardless of the model or even laptop manufacturer but it's never a bad idea to re-check yourself.

Bedr0s, 2023-11-24:
Thank you very much! I said that it had to be there a way to do something via Uefi shell, the AC loadline Undervolt works fine! I set a middle value to avoid loosing performance.

Bedr0s, 2023-11-25:
I cannot find a way to see my actual AC loadline value, hwinfo doesn't show this sensor, do you have something to suggest?

GeographicCone, 2023-11-25:
I've been looking for this too but I don't know of any way to check it. It would show in the BIOS if we had direct access to the menu. _HwInfo_ also does show it but only for some CPUs (all desktop SKUs I think). So what remains it to see how the voltage and other parameters change under load. That's all I know.

dd871, 2023-12-04:
I never used Throttlestop so not familiar with it. But by the sounds of it, it will also not work until I can figure out how to switch off VBS. E.g. as per this discussion https://www.techpowerup.com/forums/threads/locked-greyed-out-settings-while-undervolting-with-throttlestop-9-5.302664/

I will be able to do more tests tonight and can report back once I was able to find a way to fully disable VBS.

dd871, 2023-12-04:
Okay - it was actually really easy. Rather than following all the Windows guides, I should just have looked into the bios. I disabled virtualisation there and immediately got undervoltage showing as an option in XTU. I assume the same will be the case for Throttlestop.

E.g. for the P Cores the slider is now unlocked:
[image]

dd871, 2023-12-04:
It is really ridiculous how locked down the HP bios'es are. But from the 10 or so options that are remaining and that HP seems to trust its clients to change them, 'Virtualization' is indeed one of them. So I could just enter the bios normally and disable it.

Bedr0s, 2023-12-08:
> Okay - it was actually really easy. Rather than following all the Windows guides, I should just have looked into the bios. I disabled virtualisation there and immediately got undervoltage showing as an option in XTU. I assume the same will be the case for Throttlestop.
> 
> E.g. for the P Cores the slider is now unlocked: <img alt="image" width="407" src="https://private-user-images.githubusercontent.com/9387888/287575556-e45fdf4f-d36a-4e7d-8891-f415477ba08f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NjI2OTcsIm5iZiI6MTcwMTg2MjM5NywicGF0aCI6Ii85Mzg3ODg4LzI4NzU3NTU1Ni1lNDVmZGY0Zi1kMzZhLTRlN2QtODg5MS1mNDE1NDc3YmEwOGYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTEzMzE3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmJmMjQzNzkwZDY5MTdjOTFmMzNmYjA1MDkxMzRlMzEwNDgxYjM4NWNlMjJlYzg1MDMzZDFhMzJiNzRjYTI2OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.XkhATYPppKb8aJ23xFC2wq2XCVniD-63F5JzqpYaKAs">

I performed all the steps in uefi, I disabled virtualization but still the Undevolt option is blocked in xtu and not available in throttlestop, is it working for you?

Mentrauz, 2024-02-19:
> I've got a 13700HX and can confirm that changing the bios settings seem to work to disable the undervolt protection. For my bios, I did the following:
> 
> ```
> 
> CpuSetup:0x01D9                  OverClocking Performance Menu: OverClocking Feature [0x00 - Disabled, 0x01 - Enabled]
> - was already enabled
> 
> CpuSetup:0x0381                  OverClocking Performance Menu: UnderVolt Protection [0x00 - Disabled, 0x01 - Enabled]
> - set to disable
> 
> CpuSetup:0x0043                  View/Configure CPU Lock Options: CFG Lock [0x00 - Disabled, 0x01 - Enabled]
> - set to disable
> 
> CpuSetup:0x010E                  View/Configure CPU Lock Options: Overclocking Lock [0x00 - Disabled, 0x01 - Enabled]
> - was already disabled
> ```
> 
> This disabled it succesfully, I now am struggling with getting XTU to run again, as it says I have to disable virtualization if undervolt protection is off. Was following some Google results for this but so far none of them succeeded, but I likely find a solution soon.
> 
> One note, I have no idea why all guides say to also disable the 'CFG Lock' as I found no information anywhere how this is related to undervolting - only some information related to a requirement for Hackintosh...
> 
> This all makes me wonder how Omen Hub achieves the undervolt with the protection being on though?

I tried to change throught UEFI but i get error that write protection is on, how do i disable it. I have virtualisation off.

Mentrauz, 2024-02-19:
> @Dragofagnir
> 
> > hp bios is write protected and also flashing a modified bios will just brick the laptop
> 
> You don't have to modify the BIOS, just boot into EFI shell and edit the settings from there: https://github.com/GeographicCone/UefiVarTool

It deosnt let me change anything, it says "write protected" and nothing changes

lilkidsuave, 2024-06-26:
Same on 17t-ck1000
12800hx with 3060

MalfreCryvertia, 2024-08-12:
so... there is still no way to use virtualization for bluestacks/ldplayer/emulator and undervolting at the same time?

FadelHel, 2024-12-06:
> I've also been interested in this topic, so to sum up what I know so far:
> 
> For the **i7-12700H**, you can lower the [AC loadline from the default 2.3](https://edc.intel.com/content/www/us/en/design/ipla/software-development-platforms/client/platforms/alder-lake-desktop/12th-generation-intel-core-processors-datasheet-volume-1-of-2/010/vcccore-dc-specifications/#GUID-72B6320F-EC78-47E8-802A-D4E16E520955) to 1.4 [mΩ] for a maybe 5% reduction in power under load. This can be done with my [UEFI Variable Tool](https://github.com/GeographicCone/UefiVarTool) from EFI shell: `uvt CpuSetup:0x0132(2)=140` or any other utility for editing EFI variables. There's a [long thread](https://old.reddit.com/r/XMG_gg/comments/11f8n0z/launch_undervolting_via_ac_loadline_in_xmg_and/) on Reddit about this setting.
> 
> On TechPowerUp, there's [another long thread](https://www.techpowerup.com/forums/threads/cpu-i7-12700h-ring-edp-other-red-dell-g15-5520-motherboard-05j7dy.312420/) what to do about ring EDP throttling (with [ThrottleStop](https://www.techpowerup.com/download/techpowerup-throttlestop/)): lowering down _FIVR_ → _Cache Ratio_ to 36 from the default 40 makes it go away at a performance cost. Other ideas include setting _Speed Shift_ to 4 minimum and 52 maximum, at a bigger performance cost, or increasing IccMAX from the default 160 [A].
> 
> As for unlocking, these settings enable CPU configuration on the earlier firmware versions for my model (updates until September 2022, with ME version 16.**0**.x.x):
> 
> ```
> CpuSetup:0x0043=0  # Cfg Lock: No
> CpuSetup:0x01D9=1  # Overclocking: Yes
> CpuSetup:0x010E=0  # Overclocking Lock: No
> ```
> 
> Starting from October 2022, the firmware with a new ME version 16.**1**.x.x added the following two:
> 
> ```
> CpuSetup:0x0378=0  # Undervolting Lock: No
> CpuSetup:0x0379=0  # (Missing description, should be "Undervolt Protection"): No
> ```
> 
> The second option is missing title and description even in the hidden settings but I'm fairly sure that it's supposed to be the _Dynamic Undervolt Protection_ newly-added by Intel. Unfortunately even changing that value to `0` from `1` (which it is by default) doesn't seem to disable it: [HwInfo](https://www.hwinfo.com/) still reports _Dynamic Overclocking Undervolt Protection: Enabled_.
> 
> For the **i7-12700H** it's largely a moot point since it can't be undervolted anyway but if someone has an -H**X** CPU (I think at least some _Omen 17_ models came with these), seems it might take a BIOS downgrade to fully unlock the undervolting settings.
> 
> Note: I _think_ the offsets within the `CpuSetup` UEFI variable are the same for all _Alder Lake_ CPUs, regardless of the model or even laptop manufacturer but it's never a bad idea to re-check yourself.

Hello, I have an i7-14700HX I am willing to go to the furthest lengths to disable CFG Lock, Overclocking Lock, Undervolt protection. I am certain that it is possible to change these write-protected UEFI variables through WMI

FadelHel, 2024-12-06:
> > > I've also been interested in this topic, so to sum up what I know so far:
> > > For the **i7-12700H**, you can lower the [AC loadline from the default 2.3](https://edc.intel.com/content/www/us/en/design/ipla/software-development-platforms/client/platforms/alder-lake-desktop/12th-generation-intel-core-processors-datasheet-volume-1-of-2/010/vcccore-dc-specifications/#GUID-72B6320F-EC78-47E8-802A-D4E16E520955) to 1.4 [mΩ] for a maybe 5% reduction in power under load. This can be done with my [UEFI Variable Tool](https://github.com/GeographicCone/UefiVarTool) from EFI shell: `uvt CpuSetup:0x0132(2)=140` or any other utility for editing EFI variables. There's a [long thread](https://old.reddit.com/r/XMG_gg/comments/11f8n0z/launch_undervolting_via_ac_loadline_in_xmg_and/) on Reddit about this setting.
> > > On TechPowerUp, there's [another long thread](https://www.techpowerup.com/forums/threads/cpu-i7-12700h-ring-edp-other-red-dell-g15-5520-motherboard-05j7dy.312420/) what to do about ring EDP throttling (with [ThrottleStop](https://www.techpowerup.com/download/techpowerup-throttlestop/)): lowering down _FIVR_ → _Cache Ratio_ to 36 from the default 40 makes it go away at a performance cost. Other ideas include setting _Speed Shift_ to 4 minimum and 52 maximum, at a bigger performance cost, or increasing IccMAX from the default 160 [A].
> > > As for unlocking, these settings enable CPU configuration on the earlier firmware versions for my model (updates until September 2022, with ME version 16.**0**.x.x):
> > > ```
> > > CpuSetup:0x0043=0  # Cfg Lock: No
> > > CpuSetup:0x01D9=1  # Overclocking: Yes
> > > CpuSetup:0x010E=0  # Overclocking Lock: No
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Starting from October 2022, the firmware with a new ME version 16.**1**.x.x added the following two:
> > > ```
> > > CpuSetup:0x0378=0  # Undervolting Lock: No
> > > CpuSetup:0x0379=0  # (Missing description, should be "Undervolt Protection"): No
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > The second option is missing title and description even in the hidden settings but I'm fairly sure that it's supposed to be the _Dynamic Undervolt Protection_ newly-added by Intel. Unfortunately even changing that value to `0` from `1` (which it is by default) doesn't seem to disable it: [HwInfo](https://www.hwinfo.com/) still reports _Dynamic Overclocking Undervolt Protection: Enabled_.
> > > For the **i7-12700H** it's largely a moot point since it can't be undervolted anyway but if someone has an -H**X** CPU (I think at least some _Omen 17_ models came with these), seems it might take a BIOS downgrade to fully unlock the undervolting settings.
> > > Note: I _think_ the offsets within the `CpuSetup` UEFI variable are the same for all _Alder Lake_ CPUs, regardless of the model or even laptop manufacturer but it's never a bad idea to re-check yourself.
> > 
> > 
> > Hello, I have an i7-14700HX I am willing to go to the furthest lengths to disable CFG Lock, Overclocking Lock, Undervolt protection. I am certain that it is possible to change these write-protected UEFI variables through WMI
> 
> Please reply if you were able to unlock it....

Hey Mentrauz, I will certainly do. I believe I am getting close. If you want check out this thread on the HP website: https://h30434.www3.hp.com/t5/Notebook-Operating-System-and-Recovery/BIOS-Issues-unable-to-modify/m-p/9220631#M667899

The best suggestion to unlock it, is to "modify" or "spoof" the actions of the Omen Gaming Hub (since it is able to change write-protected UEFI varibales) and that is EXACTLY what OmenMon is doing. Once I understand how to deal with the WMIExecute commands, it will be done :) or when the developer of OmenMon replies to my email.

FadelHel, 2024-12-06:
> I have tried changing some values thry UVT on my Victus 16 r-1xxx with bios F-14 but it loos like all cpu realted settings are write protected... based on your coments ill be looking into some bios modification tools to posibly make a cutom version for it if someone is interested with helping with it id be glad. I'll be sharing any progres if possible here or on reddit. Might be best to implement some OGH spoofing into OmenMoon but it doesent look like somthing i have enought knowlage to try, but if someone would like to hop on some discort to discuse it might be good idea
> 
> Mainly im trying to get undervolting working outisde of OGH as its the only feature that i'd miss when unisnstalling it as my cpus is capable of -0.15V
> 
> Im also trying to fix gpu not turing off by uninstalling OGH as its the only solution i can fix about after all of debuging on i have done by that point, as it looks like OGH in some update has brokern deactivating dGPU

Hello pieotr, I am willing to hop on discord for a discussion and to help. Hopefully i could be useful.

FadelHel, 2024-12-06:
The final function which sets the undervolt in BIOS is called SetBasicTunings() present in the HP.Omen.AutoOverclockingModule inside of HPIntelOCHelperBIOS.cs
It takes CONTROL_ID and a decimal value as inputs. CONTROL_ID acts as a key with the value of "34" and the decimal value is the voltageValue.
Then the last command which executes the BIOS values change is:
UtilitySingleton.Instance.ExcuteWMICommand(131080, 55, 128, setBytes, 4)) 
where setBytes = {0x??, 0x??, 0x01, 0x00, 0x9B}
Bit #1: didnt figure it out yet
Bit #2: register of PROCESSOR_VOLTAGE_OFFSET in WMI, its value might be 0x03 in WMI
Bit #3: Sets the prefix to a negative sign . This corresponds to the UEFI variable address 0x1E0 in CpuSetup
Bit #4: didnt figure it out yet as well, maybe something that corresponds to address 0x1E1??
Bit #5: The voltValue in hexadecimal, in my case my UVT is -0.155 mV. This corresponds to the UEFI variable address 0x1E2 in CpuSetup

We already know that "131080" is the WMI default command, 55 is the "write" commandType, (54 is for read), 128 is the size of the input and 4 is the size of the output

pieotr, 2024-12-06:
@FadelHel i have just created a sevrer, might be usefull to leave link here if it would turn out more pople will be interested https://discord.gg/g3psn9t3