OmenMon/OmenMon #7: Intermittent GPU Frequency Drops to 200Mhz During Gaming while using OmenMon

memberRE, 2023-11-14:
Hello,

I am experiencing an issue with my OMEN notebook related to GPU performance while gaming. After uninstalling the OMEN Gaming Hub and only using OMENMON, post a system restart, the GPU frequency intermittently drops significantly during gameplay.

Here is a summary of the issue:

- **Environment**: OMEN by HP Laptop 16-c0006AX (484K9PA) **(5800H, RTX3070)**, playing Call of Duty MW3 and Escape from Tarkov
- **Symptoms**: While playing games, the GPU frequency frequently drops to around 200Mhz. This is accompanied by a power consumption of around 30W. The frequency stays at this reduced level for about a minute before recovering to the normal frequency of 1710Mhz.
- **Temperature readings**: During these episodes, GPU temperature readings seem normal at approximately 70°C, and CPU temperatures are also normal at around 90°C.

This issue occurs quite frequently and impacts the gaming experience severely. The fact that the temperatures are within normal ranges suggests that this should not be a thermal throttling issue. It seems like a software or a firmware-related issue where the GPU is not maintaining its frequency under load as expected.

Any insights or assistance in resolving this would be greatly appreciated. Has anyone else experienced similar issues? Any troubleshooting steps or fixes that can be recommended would be helpful.

Thank you for your time and help.

GeographicCone, 2023-11-19:
Thank you for trying **OmenMon** and reporting your experience!

You closed this issue right after opening it. I guess it must have been by accident?

> CPU temperatures are also normal at around 90°C

At first glance, I think this might be the problem here. I'm not sure if this is the normal range: there seems to be a hard-coded thermal threshold of 87°C and throttling starts when it's reached. Ideally you'd want to stay below that.

Can you try running the fans at maximum speed (so that the temperature stays below 87°C) while running these intensive tasks and see if that solves the problem.

It might also be something else, especially considering you have an AMD variant, and I could only test **OmenMon** on the Intel version I have myself. But this is what I'd suggest trying first.

GeographicCone, 2023-11-25:
I will close this issue as there has been no more information.
Please feel free to reopen.

closed 2023-11-25 by GeographicCone