OmenMon/OmenMon #8: Configuration setting to persist Automatic Fan Mode

n-elia, 2023-11-16:
Hi, I would like to point out that there is a very useful feature that is available in GUI mode but it is not configurable to be applied at startup.

According to [docs](https://omenmon.github.io/config#fan) we have the possibility to set a `FanProgramDefault` to be applied at startup.
However, we miss the possibility to automatically persist the fan program. As a consequence, the fan program will be loaded at startup, but it will get reset after the first timer timeout.

We can do that GUI-side by selecting _Const_ anfter having selected an _Auto_ program. It would be very useful to have a configuration file setting for that, e.g. `FanProgramDefaultPersist` (`True`|`False`).

GeographicCone, 2023-11-19:
That's strange. It should already work as you describe. And for me, it does. The fan program updates happen here:

https://github.com/OmenMon/OmenMon/blob/682475de136ede841ea1e4d00073b42f11d9b062/Hardware/FanProgram.cs#L193

Maybe the wording is unclear: a _fan program_ are the custom settings defined in the XML file. Something that might as well be called a "custom fan curve." It's the _Prog_ mode in the UI. That's what you set with `<FanProgramDefault>`.

> We can do that GUI-side by selecting Const anfter having selected an Auto program. 

On the other hand, _Auto_ is not a fan program: it's just leaving fans at their automatic, BIOS-controlled defaults, while optionally also selecting a different mode rather than the _Default_: for example, _Performance_.

Currently, for this mode to persist (not to revert to _Default_ while in _Auto_ mode), you have to move the selection to the _Const_ button. Then the timer will be kept from running out, as long as the main window is open. This is mostly to make sure nobody accidentally sets some weird settings, forgets about them, yet they are kept persistent and wreak havoc later.

* If an actual fan program doesn't persist, then it's an issue that should be fixed directly: perhaps it's related to the fans being controlled by the EC in your model because I can't reproduce it
* An enhancement to hold the fan mode without the main window open could be considered: for example, having a _Hold_ item in the _Fan_ submenu of the context menu
* Or is it that you'd want to set the _Performance_ mode on startup but without running a custom fan program?

I'm not really sure which one it is that would be a solution here. On a general note, this is why there's the CLI mode as well: you can make all those fine-grained changes one-by-one. At the end of the day, the GUI might not be able to accommodate every use case.

n-elia, 2023-11-19:
I am so sorry for these misunderstandings. It's been a while since we talk and I still make confusion with the wording.

> * An enhancement to hold the fan mode without the main window open could be considered: for example, having a _Hold_ item in the _Fan_ submenu of the context menu

This could be great!

> * Or is it that you'd want to set the _Performance_ mode on startup but without running a custom fan program?

And this is my expechet behavior, that in fact happens if you modify the line shown in my previous post.

As you suggest, the cli mode can be used, but I still can't find a "hold" command. However I used to do it by putting 0 in the timer register of the EC (as per your suggestion on Reddit).

I understand that the GUI should be easy to use and should also avoid misbehaviors. However there are those awesome features of starting at Windows login and applying settings at startup that would be so useful to persist Auto modes!

Please correct me if I'm wrong: in this case, "persist" means "reapply at omenmon startup", but the actual register would be reset with a laptop reboot.

GeographicCone, 2023-11-19:
@n-elia I went with your suggestion and added [FanCountdownExtendAlways](https://omenmon.github.io/config#fancountdownextendalways) in the [latest build](https://github.com/OmenMon/OmenMon/releases/latest) (0.57.0).

n-elia, 2023-11-19:
That's awesome! It means a lot to me, as it's part of a personal war against planned obsolescence.

Thank you so much for realizing that, I really appreciate your effort!

closed 2023-11-19 by n-elia