OmenMon/OmenMon #9: Independent fan control based on respective temperatures CPU/GPU

dd871, 2023-11-17:
Feature suggestion for something that in my opinion would be an awesome improvement in fan control (at least in my head it makes sense):

Make fan 0 (CPU) and fan 1 (GPU) rpm dependent on their relevant temperature readings. E.g. if we have a custom fan table like this:

```
<...>
<Level Temperature="66"><Cpu>36</Cpu><Gpu>40</Gpu></Level>
<Level Temperature="69"><Cpu>38</Cpu><Gpu>43</Gpu></Level>
<Level Temperature="72"><Cpu>41</Cpu><Gpu>46</Gpu></Level>
<Level Temperature="75"><Cpu>44</Cpu><Gpu>49</Gpu></Level>
<...>
```

And the temperature reading is CPU=75C and GPU=66C, then fan 0=44krpm and fan 1=40rpm.

GeographicCone, 2023-11-19:
That's how the _Omen Gaming Hub_ does it (CPU temp → CPU fan, GPU temp → GPU fan) but I'm not sure it was an informed design choice on their part: to me it looks like they started with the code for just one fan first, and that code was later cloned to accommodate one more.

Anyway, I considered doing it like that but in the end did not, mainly for two reasons: the first is that it complicates defining the settings (or, to be precise, coming up with the right settings).

More importantly, the second reason is that since these laptops have a single heat sink connecting everything together, the heat is supposed to be spreading evenly. It's embedded in the concept of `BD_PROCHOT` that this is how it's supposed to work. For this reason, it seems best to have both fans at higher speeds cooling everything down together, _even_ if it's only the GPU or only the CPU that's under high load.

Maybe I'm missing something but to me, defining these temperature thresholds separately on a laptop would leave users worse off, likely ending up with the situation where, for example, someone's only getting half of the cooling even when the GPU is already thermal throttling because as long as the CPU remains relatively cool, the CPU fan, controlled according to the CPU temperature only, is still running at low speed.

It'd make most sense if we assume the typical use case is very balanced between CPU and GPU usage but that's almost unheard of. In fact, it's usually the opposite: anything is either a CPU-bound or (more likely) a GPU-bound scenario.

(I'm talking about laptops here. Of course it makes sense to separate these on a desktop where the CPU is in one place and the GPU is far away. But on a laptop, it's just two chips sitting next to each other on the system board that are supposed to be cooled down jointly using the same mechanism.)

That's my current thinking but I'm open to counter-arguments. The question is interesting.

----

In fact, I was thinking the other way round: that there could actually be separate _temperature_ levels defined for the CPU and GPU, each leading to a defined level for _both_ fans. But I'm not sure what the best way to implement that would be, especially without complicating the configuration even further: right now one way I can think of would be defining "cooling levels", each consisting of a CPU and GPU fan speed level, and then matching both the CPU and the GPU temperatures to these levels separately. That's convoluted. And it'd work best with some sort of hysteresis, so that the fans don't end up changing their speeds on a whim.

Finally, on a general note, the fan programs need some rethinking, with the following two issues:

1. Right now the temperature level is taken as the maximum across 9 sensors. It is my working hypothesis that in fact only one of two of these temperatures is ever the highest (namely the CPU and GPU temperatures), so the rest of the sensors are not really necessary. But I need more data to verify this hypothesis, and I haven't been putting the laptop under enough load lately to observe it.

2. There is the case of the IR sensor (labeled "BIOS" since that's where its readings are coming from). This sensor is pointless the way things are now because the temperature it reports is always much lower than anything else. But it isn't really pointless in general, it'd just have to be considered on a different scale than the rest. I've been thinking it could be rescaled to work with the same settings: it seems the reading of 55°C is roughly equivalent to 85°C from the other sensors (near the maximum before throttling kicks in). So multiply the readings by 1.5 maybe? But that's based on the information from my laptop only, and likely might be model-specific. The alternative would be to allow separate temperature levels for this sensor too. In any case, I think the question what to do about this particular sensor (which _is_ used by the _Omen Gaming Hub_) should be addressed first before making any changes to fan programs.

Like I said, the topic is interesting. Let's continue the discussion. Meanwhile let me also say that even with the current version, it's possible to more or less have a makeshift solution by having two fan programs: one for CPU and GPU-bound usage each, and switching between them as the scenario changes.

dd871, 2023-11-22:
Very interesting! I haven't thought about it from that perspective but I agree that your point makes sense and could contribute to better overall cooling. Apparently my model has two shared heat pipes and one additional only used by the GPU - hence there would be a slight factor coming into play with the non-shared one, but this might or might not be negligible in the real world.

I was not sure how the Omen Hub is actually doing it, as their tooltip and wording is a bit ambiguous and seems to suggest it is not fully independent (showing only one fan speed "system fan" and also the tooltip for the dynamic fan curve says that the CPU/GPU curve is only a reference temperature and that the highest of both will take precedence (or something along these lines).

GeographicCone, 2023-11-22:
@dd871

> I was not sure how the Omen Hub is actually doing it

[_Omen Hub_ example fan-control data](https://github.com/GeographicCone/OmenHwCtl/blob/b3eda0b93269fedf329b9d55409d7ff020e956ac/Reference/Data%20Platform.cs#L75-L80): ([more](https://github.com/GeographicCone/OmenHwCtl/tree/master/Reference/Downloaded))

````cs
public SwFanControlCustom SwFanControlCustomPerformance {
    public FanTable FanTable {
        public List<int> Fan_Table_CPU_Fan_Speed_List = [22, 24, 28, 29, 32, 37, 40, 45];
        public List<int> Fan_Table_CPU_Temperature_List = [45, 60, 72, 75, 78, 81, 83, 86];
        public List<int> Fan_Table_GPU_Fan_Speed_List = [22, 24, 28, 29, 32, 37, 40, 45, 50, 55];
        public List<int> Fan_Table_GPU_Temperature_List = [45, 49, 52, 56, 59, 61, 64, 72, 75, 76];
        public List<int> Fan_Table_IR_Fan_Speed_List = [22, 24, 28, 29, 32, 37, 40, 45, 50, 55];
        public List<int> Fan_Table_IR_Temperature_List = [33, 35, 38, 40, 42, 43, 44, 47, 49, 51];
        get; set;
    }
    public double Lamda_Decrease { get; set; } = 0.07;
    public double Lamda_Increase { get; set; } = 0.05;
    get; set;
}
````

There are separate list pairs for (CPU temperature, CPU fan level); (GPU temperature, GPU fan level); and (IR temperature, IR fan level). These also confirm that the IR sensor readings being on a completely different scale is not a malfunction.

dd871, 2023-11-22:
Thanks @GeographicCone 

I am not surprised at all that they are doing something different to what they describe in the GUI :-)