OmenMon/OmenMon #10: Battery threshold

ArmynC, 2023-11-19:
HP Business models have a Battery threshold option. OMEN models got it _at least_ hidden. Since you have [stated](https://www.reddit.com/r/HPOmen/comments/17oet58/comment/k8y892s/) that

> There is a hidden UEFI Setup option called Battery Care which lets you stop charging at 50% or 80%. I've never checked if it works but it'd fit your description of the symptoms:
> 
> 72 * .8 = 57.6 [Wh]

my question would be: Is there any way (in future) to alter HP's Adaptive Battery Optimizer and use a forced threshold 60/80 via OmenNom calls? Do Omen models have this feature hidden in BIOS or is it removed altogether compared to Business ones?

Posted Lenovo threshold version: https://www.youtube.com/watch?v=1bm2Zbr74Qs

GeographicCone, 2023-11-19:
In my model, there's a setting for this:

````
Setup:0x0CA9 # Configuration: Battery Care Function [0x00 - 100%, 0x01 - 80%, 0x02 - 50%]
````

It's of course hidden, like nearly everything else. Never checked if it works.

If you want to check on your model, you'd have to look at the BIOS image yourself.

As a matter of fact, these past couple of days I was working on another program called **UEFI Variable Tool (UVT)**, which does just that: view and change the hidden settings. Yesterday I wrote the documentation and also published it.

So I can recommend this freshly-written guide on how to do it step-by-step:

https://github.com/GeographicCone/UefiVarTool#variable-information

_Note to anyone reading this: don't reuse the address I posted as an example, it's going to be different on every model, and might even change between BIOS versions._

ArmynC, 2023-11-19:
Got it, for my own model (**HP OMEN 17-ck0011nq / 88F7 /  2021**) some battery parameters are:

```
Setup:0x0A83  # Configuration: Battery Remaining Time [0x01 - Enabled, 0x00 - Disabled]
Setup:0x0A84  # Configuration: Battery Care Function [0x00 - 100%, 0x01 - 80%, 0x02 - 50%]
Setup:0x0A85  # Configuration: Adaptive Battery Optimizer [0x00 - Disabled, 0x01 - Enabled]
```

Full dump for generic purpose:

[omen_17_2021_uefi_f.45_rev_a.txt](https://github.com/OmenMon/OmenMon/files/13404353/omen_17_2021_uefi_f.45_rev_a.txt)

GeographicCone, 2023-11-19:
Great! So now you could actually use [UEFI Variable Tool](https://github.com/GeographicCone/UefiVarTool) aka `uvt` to see what the value is (most likely `0x00`):

`uvt Setup:0xA84`

(I think the other two settings actually _are_ visible in the _Setup_.)

And if you want to try and change it:

`uvt Setup:0xA84=0x01`

It runs from _EFI Shell_. Instructions how to set it up: https://github.com/GeographicCone/UefiVarTool#setup

And it's always good to remember you can hold <kbd>Win</kbd>-<kbd>V</kbd> when powering on to restore the default settings.

ArmynC, 2023-11-20:
So there is no realistic way of approach in order to modify the EC or to identify the cause?

> under heavy load, the battery can be losing charge despite being on AC power.

Off-topic: That is funny, I "feel" this. Before this Omen, I used to have a Lenovo laptop with a charger (135w?) that under-powered compared to what the components were using and it was always draining in not so intensive gaming conditions, even when it was charging.

This laptop has a 330w brick. I'm not sure how... would it be the problem, even on the smaller counterparts. I really hope that this is not the reason. The other brands got it manually only HP has it "adaptive" which is not bad as a second option but still.

GeographicCone, 2023-11-20:
> So there is no realistic way of approach in order to modify the EC or to identify the cause?

It's possible that it's as simple as writing to some EC register. I only know there are EC registers that _report_ battery status (current, voltage, etc.) but I didn't look into it extensively. The summary of what I know is here:
https://github.com/GeographicCone/OmenHwCtl/blob/master/Reference/EC_8A14.txt

 You can decompile the ACPI DSDT (Differentiated System Description Table) using [Intel ACPI Component Architecture Tools](https://www.intel.com/content/www/us/en/download/774881/acpi-component-architecture-downloads-windows-binary-tools.html):

````cmd
acpidump -n DSDT -o DSDT.hex
acpixtract dsdt.hex
iasl -d dsdt.dat
````

Or alternatively, just using [RwEverything](http://rweverything.com/). If you look at the logic of what's going on there, there might be some pointers.

Another approach: the Framework laptop has an open-source EC. Although it's completely different (based on Chromebooks), it gives some insight into how things are done ([source](https://github.com/FrameworkComputer/EmbeddedController/blob/6e38e82b9553240820c241c80a7d94fdc3ae5914/board/hx20/host_command_customization.h#L40-L53)):

````cpp
/* Configure the behavior of the charge limit control */
#define EC_CMD_CHARGE_LIMIT_CONTROL 0x3E03
#define NEED_RESTORE 0x7F

enum ec_chg_limit_control_modes {
	/* Disable all setting, charge control by charge_manage */
	CHG_LIMIT_DISABLE	= BIT(0),
	/* Set maximum and minimum percentage */
	CHG_LIMIT_SET_LIMIT	= BIT(1),
	/* Host read current setting */
	CHG_LIMIT_GET_LIMIT	= BIT(3),
	/* Enable override mode, allow charge to full this time */
	CHG_LIMIT_OVERRIDE	= BIT(7),
};
````

From [another description]( https://www.howett.net/posts/2021-12-framework-ec/) of the Framework laptop's EC:

> User-toggleable features such as the battery charge limit and the power button brightness are copied from NVRAM to the EC by `OEMBIOSSyncToECDxe` on every boot.

Assuming the implementation is similar, this would mean you could:

* Find an HP laptop model where this setting works
* Analyze the UEFI image to find a similar DXE that updates the EC from the UEFI variable setting
* Reverse engineer the code to see how it manages to do so

And maybe you'd then find the EC register to write to with `OmenMon -Ec` to change this setting. And then it's also possible you'd find out it doesn't work anyway because it's been disabled or not implemented in the EC as well.

ArmynC, 2023-11-21:
> You can decompile the ACPI DSDT (Differentiated System Description Table)

From a quick look inside, **I think** it reports the same thing on my side, just the battery status.
Dump:  [dsdt.dsl.txt](https://github.com/OmenMon/OmenMon/files/13422597/dsdt.dsl.txt) (Omen 17 2021 88F7)

> If you look at the logic of what's going on there, there might be some pointers.

Strangely, It won't load the driver.

> Reverse engineer the code to see how it manages to do so

Sadly I think that it is above my usual knowledge so I'll keep this Issue as a reference. Thanks!