OmenMon/OmenMon #12: [878D] HP Envy 15 15-ep0003nl (2020)

n-elia, 2023-11-20:
# HP Envy 15 15-ep0003nl (2020)

## Model and OmenMon release

This report refers to OmenMon version [0.57.0-Release](https://github.com/OmenMon/OmenMon/releases/tag/0.57.0).

The laptop model is HP Envy 15 15-ep0003nl, 2020.

## Issues

**Major issues**: None

**Minor issues**:
- Wattage (permanently shows 0W)
- CPUT is not shown

## Screenshot

[image]

## OmenMon configuration

**For future users**: leave the current default configuration file. [FanLevelNeedManual](https://omenmon.github.io/config#fanlevelneedmanual) should be set to false, and [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec) also false.
Please note that the keyboard section is greyed out because our model does not support keyboard backlight customisation.

GeographicCone, 2023-11-20:
Thank you for the report!
I'm really delighted to hear **OmenMon** works not just on some _Omen_ models but even on totally different series.

n-elia, 2023-11-24:
Update: everything working with the 0.58 version, too. The "0W" wattage has gone as expected 👍.
I have a question: in versions < 0.57 putting `<FanProgramDefault>Performance</FanProgramDefault>` and `<AutoConfig>true</AutoConfig>` resulted in _Auto_ mode being set up at startup, with _Performance_ settings selected. Newer versions seem to work only with Fan Programs. Is this okay, or is this a bug with my model?

GeographicCone, 2023-11-25:
What goes inside `<FanProgramDefault>` (and now also `<FanProgramDefaultAlt>` since the previous version) should be the name of a fan program defined in `<FanPrograms><Program Name="...">`

If the fan program with that name isn't defined, nothing should be happening:
https://github.com/OmenMon/OmenMon/blob/c0ece9496da7cc09a1c3f083bd603f78a7a9f188/Hardware/FanProgram.cs#L279-L281

There are two programs in the example config: `Power` and `Silent` and unless your custom version of `OmenMon.xml` specifically defines a fan program that happens to be named `Performance` (just as the _Performance_ [fan mode](https://omenmon.github.io/cli#fanmode) by coincidence, which by the way would be an acceptable program name, there is no collision possible), then it should have never worked the way you describe to begin with. So no, it's not a bug.

And it's not an omission either: as I can't really be sure, especially with all the different hardware models around, that "just" setting the _Performance_ mode will take care of fan levels appropriately when the temperatures rise (and that they don't, for example, remain fixed at whatever they were at the moment the _Performance_ mode was being switched on), I didn't think it'd be a good idea to have just the mode being set automatically.

The mode is set and maintained as part of running a fan program, where **OmenMon** monitors and responds to temperature changes, and if the program is ended, it's restored back to what it was before. It can also be set manually and then maintained by extending the countdown indefinitely but it has to be a conscious choice by the user, either by selecting the _Const_ radio button in the UI, or setting the [FanCountdownExtendAlways](https://omenmon.github.io/config#fancountdownextendalways) to **true**.

n-elia, 2023-11-27:
> I can't really be sure, especially with all the different hardware models around, that "just" setting the _Performance_ mode will take care of fan levels appropriately when the temperatures rise [...]

I got it now, and therefore I confirm that it works as expected. Thanks for explaining the details so clearly! I definitely agree with your decision.
For what concerns the laptop in the subject, I can confirm for future visitors that setting an Auto mode works exactly as the genuine HP Command Center did :tada:

I'm leaving this issue open for future feedbacks on newer versions, or discussions for this Envy laptop, if it's okay for you!

GeographicCone, 2023-11-28:
> I'm leaving this issue open [...] if it's okay for you!

Of course!