OmenMon/OmenMon #13: [88EC] HP Victus 16-e0008nt (2021) (4H0S0EA)

Rupurudu, 2023-11-21:
# [88EC] HP Victus 16-e0008nt (2021) (4H0S0EA)
## OmenMon release

Tested with OmenMon version [0.57.0-Release](https://github.com/OmenMon/OmenMon/releases/tag/0.57.0).

## Device Status

**Working**: 
* Fan Monitoring (RPM & Percent)
* Fan Control (Prog/Auto/Max/Const/Off)
* Fan Off mode
* GPU Power mode 
* Intercept Omen key
* Temp sensors (Mostly)
* Power source (AC Power Present)

**Not Working**:
* Sometimes some sensors does not shown.
* AC Power usage always shows as "0W"
* Commandline fails unless "BiosErrorReporting" is set to false

**Not Tested / Available**:
* Keyboard backlight (No RGB keyboard on this model)
* Discrete Exclusive / Optimus (No MUX switch on this model)
* Standard / High refresh rate (No idea what this does?)

## Fan Configuration

This model's fan configuration is different from the developers model, so I would highly recommend changing fan programs.

**Warning**: Since Omen Control Center does not have manual fan control for this model (Other than normal/max), the values below are my own observations.

Neither of the fans have minimum RPM limit enforced by EC. However, they stop (and get automatically restarted) when they set below 700RPM. I would not recommend setting them below 800rpm, and I'm using 1000rpm as minimum. For the Maximum fan settings, CPU can go up to 4700rpm, while GPU can go 5100rpm.

You'll need to set minimum/maximum fan levels like this:
```
        <FanLevelMax>51</FanLevelMax>
        <FanLevelMin>4</FanLevelMin>
```

Here is the fan program I'm using: 
```
        <FanPrograms>
            <Program Name="Power">
                <FanMode>Performance</FanMode>
                <GpuPower>Maximum</GpuPower>
                <Level Temperature="00"><Cpu>00</Cpu><Gpu>00</Gpu></Level>
                <Level Temperature="36"><Cpu>12</Cpu><Gpu>12</Gpu></Level>
                <Level Temperature="39"><Cpu>13</Cpu><Gpu>13</Gpu></Level>
                <Level Temperature="42"><Cpu>14</Cpu><Gpu>14</Gpu></Level>
                <Level Temperature="45"><Cpu>15</Cpu><Gpu>15</Gpu></Level>
                <Level Temperature="48"><Cpu>16</Cpu><Gpu>16</Gpu></Level>
                <Level Temperature="51"><Cpu>18</Cpu><Gpu>18</Gpu></Level>
                <Level Temperature="54"><Cpu>20</Cpu><Gpu>20</Gpu></Level>
                <Level Temperature="57"><Cpu>22</Cpu><Gpu>22</Gpu></Level>
                <Level Temperature="60"><Cpu>24</Cpu><Gpu>24</Gpu></Level>
                <Level Temperature="63"><Cpu>26</Cpu><Gpu>26</Gpu></Level>
                <Level Temperature="66"><Cpu>29</Cpu><Gpu>29</Gpu></Level>
                <Level Temperature="69"><Cpu>32</Cpu><Gpu>32</Gpu></Level>
                <Level Temperature="62"><Cpu>35</Cpu><Gpu>35</Gpu></Level>
                <Level Temperature="75"><Cpu>38</Cpu><Gpu>38</Gpu></Level>
                <Level Temperature="78"><Cpu>41</Cpu><Gpu>41</Gpu></Level>
                <Level Temperature="81"><Cpu>44</Cpu><Gpu>46</Gpu></Level>
                <Level Temperature="84"><Cpu>47</Cpu><Gpu>51</Gpu></Level>
            </Program>
            <Program Name="Silent">
                <FanMode>Default</FanMode>
                <GpuPower>Minimum</GpuPower>
                <Level Temperature="00"><Cpu>00</Cpu><Gpu>00</Gpu></Level>
                <Level Temperature="36"><Cpu>10</Cpu><Gpu>00</Gpu></Level>
[... 140 lines cut ...]
- Thermal Throttling Status: 0x00 = 0b00000000 = 0 [Unknown]
```

EC:
```
OmenMon Hardware Monitoring & Control Utility Version 0.57-Release
Embedded Controller (-Ec)
0x _0 _1 _2 _3 _4 _5 _6 _7 _8 _9 _A _B _C _D _E _F
0_ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1_ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2_ 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 39 00
3_ 01 0e 6f 01 1a 1a 33 00 00 00 00 00 00 00 00 00
4_ 0b 01 04 14 00 10 00 00 00 2b 29 00 00 00 00 00
5_ 80 00 00 0f 00 03 00 32 31 29 00 00 00 00 00 00
6_ 00 00 06 56 00 00 00 00 00 74 00 0a 00 0c 0c 00
7_ c6 11 38 0e 28 3c 80 33 31 33 2d 31 44 2d 34 42
8_ 2d 41 00 00 00 38 0e c2 41 00 00 00 01 e0 00 00
9_ d1 d1 38 0e 05 31 5f 00 00 00 00 00 00 d3 0b 00
a_ 03 80 00 88 01 00 00 00 00 33 00 00 00 01 00 01
b_ 22 0a 35 0a 00 c0 5e 2b 00 00 00 07 00 00 00 00
c_ 03 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00
d_ 5f 00 00 00 24 02 00 01 00 00 1d 03 00 00 00 00
e_ 00 00 00 00 00 00 00 00 00 00 08 00 00 68 00 00
f_ 08 00 20 00 00 00 00 00 31 00 38 30 37 32 30 00
```

GeographicCone, 2023-11-22:
@Rupurudu Thank you for your report! Very thorough.

> AC Power usage always shows as "0W"

[Will be hidden](https://github.com/OmenMon/OmenMon/commit/e33b7380f136540d3e6f37fcea598f1bf423981a) if 0 from the next build.

> Standard / High refresh rate (No idea what this does?)

Just a quicker way to change the display refresh rate between the default 60Hz and the display's highest, which can be 144Hz, 165Hz, or 240Hz depending on the model. Both values can be customized in `OmenMon.xml` with [PresetRefreshRateHigh](https://omenmon.github.io/config#presetrefreshratehigh) and [PresetRefreshRateLow](https://omenmon.github.io/config#presetrefreshratelow)

> I use a custom Omen key action that'll press the home key instead: [...]

That's a great idea! Maybe we could have a separate issue just for custom actions. Would you like to start it and repost it there?

> 0x \_0 \_1 \_2 \_3 \_4 \_5 \_6 \_7 \_8 \_9 \_**A** [...] \_**F**
> **a**\_
> [...]
> **f**\_

I've only just realized the column headers have been upper-case, while the row headers are lower-case. Made them consistent now, for the record.

Rupurudu, 2023-11-24:
> @Rupurudu Thank you for your report! Very thorough.
> 
> > AC Power usage always shows as "0W"
> 
> [Will be hidden](https://github.com/OmenMon/OmenMon/commit/e33b7380f136540d3e6f37fcea598f1bf423981a) if 0 from the next build.
Thank you for the update. :)

> > I use a custom Omen key action that'll press the home key instead: [...]
> 
> That's a great idea! Maybe we could have a separate issue just for custom actions. Would you like to start it and repost it there?

I would love to do that, but before that let me ask a question, currently we can only detect keypreses, (KeyDown, KeyUp) Can we detect when the Omen key is held down? I would love to update the key command to support that.

> > 0x _0 _1 _2 _3 _4 _5 _6 _7 _8 _9 _**A** [...] _**F**
> > **a**_
> > [...]
> > **f**_
> 
> I've only just realized the column headers have been upper-case, while the row headers are lower-case. Made them consistent now, for the record.

Good catch, I haven't noticed it

GeographicCone, 2023-11-25:
> Can we detect when the Omen key is held down?

I'm afraid not because the way it works is not really like any regular keypress. The moment it's pressed, a WMI event is generated, and everything that happens later is in response to the event filter query:

`SELECT * FROM hpqBEvnt WHERE eventData = 8613 AND eventId = 29`

No further events are generated if the key is held down, so there's no way to tell that's what's going on.

**OmenHwCtl** [has a script implementation](https://github.com/GeographicCone/OmenHwCtl/blob/master/OmenHwCtl.ps1#L378-L395), if you want to experiment with how it works:

````pwsh
$OmenKeyConsumer = New-CimInstance -CimSession $Session -ClassName 'CommandLineEventConsumer' `
    -Namespace 'root\subscription' -Property  @{`
        CommandLineTemplate = 'C:\Windows\System32\schtasks.exe /run /tn "Omen Key"';
        ExecutablePath = 'C:\Windows\System32\schtasks.exe';
        Name = 'OmenKeyConsumer';
    }
$OmenKeyFilter = New-CimInstance -CimSession $Session -ClassName '__EventFilter' `
    -Namespace 'root\subscription' -Property @{`
        EventNameSpace = 'root\wmi';
        Name = 'OmenKeyFilter';
        Query = 'SELECT * FROM hpqBEvnt WHERE eventData = 8613 AND eventId = 29';
        QueryLanguage = 'WQL';
    }
$OmenKeyBinding = New-CimInstance -CimSession $Session -ClassName '__FilterToConsumerBinding' `
        -Namespace 'root\subscription' -Property @{`
            Consumer = [Ref] $OmenKeyConsumer
            Filter = [Ref] $OmenKeyFilter;
    }
````

But unless another way is discovered to query key state, all that can be done for now is detecting individual keypresses.

Rupurudu, 2024-01-10:
Update:
* Marked "AC Power usage always shows as "0W"" as fixed.
* Moved Keyboard backlight from Not Available to Not Working. (Omen Hub does not allow switching keyboard backlight at all. I can only switch with <kbd>Fn</kbd> + <kbd>F4</kbd>)
* Updated Fan Configuration
    * Power program now uses LegacyPerformance fan mode. With the default Performance fan mode BIOS sometimes disables GPU power boost.
    * Power program fans are reduced by ~200rpm. This reduces overall noise while still being reasonably cool & without sacrificing performance.
    * Silent program now uses LegacyQuiet mode. 
    * Silent program now uses much more slower fan speeds and a different curve. I use it mostly for internet browsing on battery power.