OmenMon/OmenMon #14: fan profile resets to default on lockscreen

ourslfs, 2023-11-24:
After 3 minutes and something on lockscreen the programmable fan profile resets to auto default. HP 88D1 (hp omen 2021 15-en1xxx). Windows 10 21h2. Set fancountdownextendalways to true, maybe should set faneveluseec to true?

ourslfs, 2023-11-24:
with fanleveluseec enabled doesn't seem to reset it

GeographicCone, 2023-11-24:
Are you using a custom fan program (such as _Power_ or _Silent_ in the example configuration file), or do you just want to keep the _Performance_ mode permanently on without a fan program active?

In the latter case, you need to either set [FanCountdownExtendAlways](https://omenmon.github.io/config#fancountdownextendalways) to **true** or keep the main window open and the _Const_ button selected there. Note that the configuration file entries are case-sensitive.

As for [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec) (and [FanLevelNeedManual](https://omenmon.github.io/config#fanlevelneedmanual)), the best way to decide if you need them is to try to set the fans manually using the [command-line mode](https://omenmon.github.io/cli#fanlevel): if it doesn't work via the BIOS call, you can then try with the EC.

Separately, since you mentioned lockscreen: **OmenMon** only runs on user logon, so in such situations it's possible the process hasn't been started yet or got terminated, and that's why it cannot extend the countdown.

It seems the issue is resolved but if not, please try to reproduce without the lockscreen: in such case, you can also check the countdown timer in the main window or from the command line by running `OmenMon -Ec 0x63`. Unless the problem is lockscreen-specific, in which case please provide more details how to reproduce.

GeographicCone, 2024-01-07:
I will close this issue assuming it's been resolved.
Please feel free to reopen if necessary.

closed 2024-01-07 by GeographicCone