OmenMon/OmenMon #15: Getting rid of the Omen services

Rupurudu, 2023-11-24:
Even through the Omen application is uninstalled, the Omen services remain installed on the computer. Since we don't support running with the Omen Control Center concurrently and, we don't use Omen services to run. It's a good idea to remove those services.

However, these services are a pain to uninstall, since they got installed by Windows Update automatically and there is no perfect way to remove those services completely.

Here is the little research I did:

There are two dummy ACPI devices exposed by BIOS, those are device IDs `ACPI\HPIC0003`, `ACPI\HPIC000C` and `ACPI\HPIC0004`. I believe `ACPI\HPIC0003` is for consumer computers,`ACPI\HPIC000C` is for business computers, and `ACPI\HPIC0004` for Omen and Victus devices. I have devices `ACPI\HPIC0003` and `ACPI\HPIC0004`.

They install two drivers, hpcustomcapdriver.inf for `ACPI\HPIC0003`and `ACPI\HPIC000C` and hpomencustomcapdriver.inf for `ACPI\HPIC0004`. They are named HP Application Driver and HP Omen Driver respectively.

They are composed of two components: a system driver (`hpcustomcapdriver.sys` and `hpomencustomcapdriver.sys`) and a software device (`SWC\HPIC0003`, `SWC\HPIC000C` and, `SWC\HPIC0004`)

These software devices are the ones which download a lot of services, including analytics.

I tried to remove those drivers, and so far I have only found one method that partially works, using [Group Policy to create Device Installation Restrictions](https://www.tenforums.com/tutorials/146562-prevent-windows-update-updating-specific-device-driver.html). This method works in a way that while Windows Update automatically downloads the drivers, it fails to install them and causes an Windows Update error. Thankfully, this error only happens once, unless the driver is removed from `%windir%\System32\DriverStore\FileRepository\` using pnputil.

I wonder if it's possible to replace those driver with our own, either by replacing them with a [null driver](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/installing-a-null-driver) or replacing them with a functional driver to do the background work, like setting fan profiles on login screen (fixes #14), setting fan profiles without a user logged in, and possibly running GUI without Admin privledges/UAC. Creating a kernel mode driver (like HP did) is out of the question since it requires certification from Microsoft, but creating a user mode service running as `NT AUTHORITY\SYSTEM` should be possible.

Rupurudu, 2023-11-24:
Here is the registry file for creating Device Installation Restrictions, if you are running Home version which don't have Group Policy Editor:

```
Windows Registry Editor Version 5.00

[HKEY_LOCAL_MACHINE\SOFTWARE\Policies\Microsoft\Windows\DeviceInstall\Restrictions]
"DenyDeviceIDs"=dword:00000001
"DenyDeviceIDsRetroactive"=dword:00000001

[HKEY_LOCAL_MACHINE\SOFTWARE\Policies\Microsoft\Windows\DeviceInstall\Restrictions\DenyDeviceIDs]
"1"="ACPI\\HPIC0003"
"2"="ACPI\\HPIC0004"
"3"="ACPI\\HPIC000C"
```

GeographicCone, 2023-11-25:
That's an excellent guide! Thank you for sharing this.

It also mirrors what I know from a slightly different perspective. I'll summarize.

As far as I know there are four driver `.inf` files in total. The **first two** can be safely installed:

* `HpCustomCapDriver.inf` for `ACPI\HPIC0003` and `ACPI\HPIC000C` version `1.37.2536.0` of `2021-09-09`
* `HpOmenCustomCapDriver.inf` for `ACPI\HPIC0004` version `1.0.220.0` of `2018-05-17`

The **third** one doesn't seem necessary but should be harmless:

* `HpOmenCustomCapExt.inf` for `ACPI\HPIC0004` version `1.0.220.0` of `2018-05-17`

These are the latest versions as of now. As you can see, they are not really updated often, if at all.

Now, all of the above come with the following **fourth** driver, which you most definitely _don't_ want to install:

* `HpOmenCustomCapComp.inf` for `SWC\HPIC0004`

That's the software component, which includes the bloated background service with dependencies such as `HPAnalyticsNativeClientLib.dll`  that also instructs Windows to install the HSA UWP application, i.e. the _Omen (Gaming) Hub/Control Center_.

As long as you only install the first two drivers (possibly the first three), there will be no background service running and Windows will never know to look for the UWP app.

I can't give the exact steps and verify for sure because my Windows installation is heavily modified to the point nothing happens automatically anyway and I slipstream all the drivers into the image before Windows is even installed, so I never do this manually, but with this caveat in mind:

What has to be done is to manually override Windows' driver choice and only point it to the first two or three of the `.inf` files listed above. Then the drivers are installed, and since they are generally never updated (no newer versions are released by HP), it should be the end of story.

The driver binaries can be easily located in the Windows Catalog, and downloaded from there:

* https://www.catalog.update.microsoft.com/Search.aspx?q=ACPI\HPIC0003
* https://www.catalog.update.microsoft.com/Search.aspx?q=ACPI\HPIC0004

Now if you already happen to have the fourth, problematic driver installed, it should be possible to remove it with `pnputil /delete-driver` or `devcon dp_delete` like you say, although I think it's also doable from the _Device Manager_ GUI with <kbd>Win</kbd>-<kbd>X</kbd>, <kbd>M</kbd>, select and right-click on the device, choose _Uninstall_, check the box next to _Delete the driver software for this device_ and click the _Uninstall_ button, and maybe that's the easiest way for the casual user.

----

Note: as for #14, I think that issue was resolved. I've just checked again that **OmenMon** works behind the lockscreen if the user is already logged in. That it doesn't run with no-one logged in is a known design limitation. The only situation I can think of when this would realistically matter is if someone were running a server from their laptop with no interactive use. I don't think it's a very likely scenario but for such edge cases there's the [command-line mode](https://omenmon.github.io/cli), and you can also [run a fan program](https://omenmon.github.io/cli#prog) from it: so I don't think anything needs to be fixed here as things are.

Drivers are a means to an end. Once new ways of controlling the hardware are discovered, the rest is just an implementation detail. That the _Omen Hub_ has a custom driver doesn't necessarily demonstrate how advanced it is: it's primarily to work-around the limitations of the UWP model. Although it does say a lot that the `HpOmenCustomCapComp.inf` installation is 50 MB in total, of which 30 MB is the `p95bench.exe` and then `linpack_xeon64.exe` is another 10MB. So basically the whole layer for interfacing with all the hardware (as well as "analytics," etc.) is only 10 MB (already too big if you ask me). But then the 400+ MB UWP app is essentially a front-end that only provides the user interface for it.

Rupurudu, 2024-01-09:
Apologies for the late reply, I just noticed I didn't reply to this thread after being pined on #23 .

 
> That's an excellent guide! Thank you for sharing this.
>
> It also mirrors what I know from a slightly different perspective. I'll summarize.
>
> As far as I know there are four driver `.inf` files in [...]

The sole reason I did the Device Installation Restrictions to deny the installation of `ACPI\HPIC0003` and `ACPI\HPIC0004` devices is to prevent the creation of `SWC\HPIC0003` and `SWC\HPIC000C` devices. As you mentioned, `HpOmenCustomCapComp.inf` for `SWC\HPIC0004` is the one I don't want to install, but if the device exists, Windows Update downloads and installs it anyway, regardless of the device is disabled or not. 

The only ways to disable this behavior is to disable driver updates from WU entirely, (Which I believe what you did on your heavily modified Windows installation.), or hide a specific update with "wushowhide.diagcab" (I have to do it again every time when `HpOmenCustomCapComp.inf` updates, which is [fairly often](https://www.catalog.update.microsoft.com/Search.aspx?q=SWC\HPIC0004).

Since both of the fixes to stop WU does not suit my needs, I found the solution by not having the software devices, since WU does not install the driver when the device does not exist.

> It should be possible to remove it with `pnputil /delete-driver` or `devcon dp_delete` like you say, although I think it's also doable from the _Device Manager_ GUI with Win-X, M, select and right-click on the device, choose _Uninstall_, check the box next to _Delete the driver software for this device_ and click the _Uninstall_ button, and maybe that's the easiest way for the casual user.

I agree, doing it on the _Device Manager_ GUI also works. 

> Note: as for #14, I think that issue was resolved. I've just checked again that **OmenMon** works behind the lockscreen if the user is already logged in. That it doesn't start automatically with no-one logged in is a known design limitation. The only situation I can think of when this would realistically matter is if someone were running a server from their laptop with no interactive use. I don't think it's a very likely scenario but for such edge cases there's the [command-line mode](https://omenmon.github.io/cli), and you can also [run a fan program](https://omenmon.github.io/cli#prog) from it: so I don't think anything needs to be fixed here as things are.
> 
> Drivers are a means to an end. Once new ways of controlling the hardware are discovered, the rest is just an implementation detail. That the _Omen Hub_ has a custom driver doesn't necessarily demonstrate how advanced it is: it's primarily to work-around the limitations of the UWP model. Although it does say a lot that the `HpOmenCustomCapComp.inf` installation is 50 MB in total, of which 30 MB is the `p95bench.exe` and then `linpack_xeon64.exe` is another 10MB. So basically the whole layer for interfacing with all the hardware (as well as "analytics," etc.) is only 10 MB (already too big if you ask me). But then the 400+ MB UWP app is essentially a front-end that only provides the user interface for it.

Yeah, as for the reason I asked for a driver is not the limitation of the OmenMon, it's the limitation of the BIOS. Since I have to not install drivers for `ACPI\HPIC0003` and `ACPI\HPIC0004` to not have devices `SWC\HPIC0003` and `SWC\HPIC0004`. (These drivers are the ones creating the software devices.) 

However, doing this with Device Installation Restrictions, has two problems.
1. Windows Update automatically downloads the drivers (for `ACPI\HPIC0003` and `ACPI\HPIC0004`), it fails to install them because of the restrictions and causes an Windows Update error. This only happens one time. Driver is downloaded and in the driver store, but not installed. No more error happens if the driver is not removed with `pnputil`.
2. Device manager shows 2 Unknown devices. This is just visual, but annoying.

The suggestion of the [null driver](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/installing-a-null-driver) is to fix this issue.

Also I agree that the _Omen Hub_ is extremely bloated.

GeographicCone, 2024-01-09:
Thank you @Rupurudu! This is all very helpful to all the people with a more standard setup that want to use **OmenMon** without _Omen Hub_ getting in the way. (As for me, yeah I do block all the updates entirely.)