OmenMon/OmenMon #19: Any chance of a version for desktop Omens?

zejjnt, 2023-11-29:


GeographicCone, 2023-11-30:
Thank you for your interest in **OmenMon**!

What kind of functionality would be needed for _Omen_ desktops? And which part of it is missing?

I implemented all the WMI BIOS calls I'm aware of in the [command-line mode](https://omenmon.github.io/cli). I _guess_ some of them must be desktop-related because they don't work on the laptops. For example the [LED animation table](https://omenmon.github.io/cli#anim) or the [fan table](https://omenmon.github.io/cli#fantable). Would it be just about a GUI for some of these then? Because if it's not something that can be done from the CLI mode already, then it's likely because I didn't have the information needed to implement it.

On a general note, I'm not really familiar with how much an _Omen_ desktop differs from any other desktop PC so excuse my naiveté, but is there really a use case for an _Omen_-specific utility for the desktop? As much as I enjoyed writing **OmenMon** and the earlier [OmenHwCtl](https://github.com/GeographicCone/OmenHwCtl), I wouldn't have bothered if the same could have been accomplished with pre-existing software. I'm under the impression that on a desktop, you can just use the CPU and GPU-specific utilities directly, and maybe all you need is to unlock some of the hidden BIOS settings, for example with [UEFI Variable Tool](https://github.com/GeographicCone/UefiVarTool).

zejjnt, 2023-12-17:
Hi, and thank you for replying.
HP basically does not follow any standard practices for reporting data such as fan speed (and fan control), temperatures and controlling the case lights.
Very recently OpenRGB launched a new version which can finally control the case lights, but neither read nor alter the fan speeds.

I have still not found anything at all that can show any of the fan speeds, and neither anything that can control the speeds/profiles of them
Most motherboards have one or a few temperature sensor(s) on other parts of the board (this has AMD's chipset B450 which certainly allow for it), but I have not been able to find or read that either.

Both OmenMon, OmenHwCtl and OmenHubLight are either not working at all, or not working to the point that they are of use. The WMI calls either fail, has no effect, or the softwares crash in one way or another.
I have tweaked around with them quite a bit, changing the xml parameters in OmenMon and trying out things in Visual Studio before compiling OpenMonLight but nothing has had any effect at all; literally none whatsoever.
I have tried _literally_ every command line parameter for OmenMon and OmenHwCtl respectively along with the above changes in the configuration files.
The LEDs are not applicable since the arrangement is different, but none of the fan speed/type/count/profile... commands read, write or alter anything at all.
The GUI crashes when I try to open it, and the fan options "Power" and "Silent" do nothing.

I have not worked with WMI/WMIC previously but my guess would be that the WMI calls are using different namespaces, paths and commands. Not an entirely new set, since some calls are able to execute without warning but also without effect. Changes have clearly been made.

* I would like to monitor, and hopefully change, fan speeds.
* Access as many sensor readouts as possible.
GUI or CLI doesn't matter to me.

Your last suggestion is the only one I haven't gotten to yet; I was hoping to flash a modified BIOS but was stonewalled by the fact that HP apparently requires some sort of signature for their BIOS binaries in order to be able to reprogram.
While digging through the BIOS it is clear that there are lots of options simply disabled in there, so it should be possible.
However I was unable to find any option for changing anything cooling related anywhere in the BIOS, but it is possible to enabled 0 RPM warnings for three of the fans separately, and the stock BIOS shows current RPM for the two fans I have plugged in ATM so again it is possible in **some** way.
It's just obfuscated, and the only software able to access these variables in Windows so far is the standard Omen suite.

I will check if I can get something out of the UEFI variable changing methods there.
I have overclocked CPU & RAM, changed voltages and timings respectively without a single problem using Ryzen Master just like with any other motherboard, so at least the most basic of functions are available in there somewhere.

If I can assist in some way just let me know.

GeographicCone, 2024-01-07:
> If I can assist in some way just let me know

If _Omen Hub_ can control these settings, then you could use a tool like [dnSpy(Ex)](https://github.com/dnSpyEx/dnSpy) to see how it manages to do it, so that the same method can then be reproduced.

zejjnt, 2024-01-08:
> > If I can assist in some way just let me know
> 
> If _Omen Hub_ can control these settings, then you could use a tool like [dnSpy(Ex)](https://github.com/dnSpyEx/dnSpy) to see how it manages to do it, so that the same method can then be reproduced.

Thanks, I'll try to get something out of it. I have never really used any debugging tool so it's extremely overwhelming and there's not really any documentation but I hope I can find something to help me.