OmenMon/OmenMon #20: Bypass Dynamic Boost

Dragofagnir, 2023-12-01:
It would be great it there were a way to bypass the dynamic boost feature with wmi calls as we can unlock the cpu tdp with omenmon, I know that msi laptops have that feature to run a max wattage no matter what the cpu wattage is at

GeographicCone, 2023-12-01:
Have you tried setting [Cpu:PL4](https://omenmon.github.io/cli#cpupl4) and [Cpu:PLGpu](https://omenmon.github.io/cli#cpuplgpu) from the CLI? (For the record, the first one works for me, the second one doesn't seem to do anything on my model.)

You can also try [unlocking what you can](https://github.com/GeographicCone/UefiVarTool/blob/master/extra/8A14_Set_CL34.uvt) in the hidden BIOS settings (note: just an example, offsets might be different on every model and even between BIOS versions).

Beyond that, if you'd like to do away with the PPAB and increase the custom TGP accordingly instead, I think that's at the nVidia VBIOS level. You can succesfully flash a VBIOS from another variant: I was able to change the 115W TGP to 150W doing just that on my model. However, part of it is still PPAB: to get rid of it completely would entail making a custom BIOS. Last I checked, it was not possible but there were some interesting developments [here](https://www.techpowerup.com/forums/threads/nvflashk-flash-any-bios-to-nvidia-gpus-safe-board-id-bypass-up-to-4xxx-series.312608/) and [here](https://www.techpowerup.com/forums/threads/omg-vflash-fully-patched-nvflash-from-x-to-ada-lovelace-v5-780.312601/). I haven't had the time to follow up on this lately.

Dragofagnir, 2023-12-03:
the PLCpu also doesnt work for me, i dont know if the pl4 commands works tho but i managed to make make cpu sips 150w with throttlestop, and yes disabling the platform controller will end up with a decrease in power comsuption,