OmenMon/OmenMon #21: **Feature Request** Discrete GPU disable mode

islaka, 2023-12-05:
I currently own a Victus 16, and I've noticed that the battery life significantly decreases to 2 hours when the NVIDIA is enabled, compared to 4 hours when it's disabled. I came across a control app that offers this feature for the ROG Zephyrus, and I'm curious if a similar implementation could be made available for Omen and Victus as well.
The link to the control app i mentioned: https://github.com/seerge/g-helper
Thank you guys so much for making this great app \('o')/

islaka, 2023-12-07:
Oh I haven't known this. Goona try it right now. Thank you so much 🥳

closed 2023-12-07 by islaka

GeographicCone, 2024-01-07:
@islaka There is no HP-specific way for disabling the dGPU that I know of but you can use one of the general methods like @Rupurudu suggested.

I just wanted to add that you should probably figure out which process is waking up the dGPU and deal with it specifically. It's possible you have an [Electron](https://www.electronjs.org/) app (essentially a Chromium browser instance) running in the background using the dGPU, which you could switch to iGPU instead. Or it could be one of the temperature-monitoring tools, in which case there should be a setting somewhere for it not to wake up the dGPU.

You shouldn't have to disable the dGPU on a system-wide level, just prevent a specific application from using it.