OmenMon/OmenMon #22: Fan monitoring and control not working

PearlNotHere, 2023-12-14:
i have hp omen 16-wf0061tx and the fans control is not working for me , keyboard lights work just the fans not working

PearlNotHere, 2023-12-14:
i dont know anything about coding so plz give me detailed information thnx

originalgod11, 2024-01-06:
i am also running into this problem, everything works but fan control. leaves my fans on medium speed no matter if put it on OFF or MAX. (OMEN 16 2023 R9 4070)

GeographicCone, 2024-01-07:
On a general note: setting arbitrary fan levels, setting fans to maximum speed, and turning them off entirely are distinct cases, handled differently. Depending on the model, one functionality might work while another won't.

For setting arbitrary fan levels, you can try enabling the [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec) setting, and possibly also [FanLevelNeedManual](https://omenmon.github.io/config#fanlevelneedmanual) in the `OmenMon.xml` [configuration file](https://omenmon.github.io/config).

The models in question are the newest gaming models though so all of these should be possible with WMI, in principle. I think someone already reported **OmenMon** to be working with a 2023 model. Could it be that the _Omen Hub_ is also setting the fan levels at the same time and interfering?

originalgod11, 2024-01-11:
i have omenhub uninstalled and all other omen bloat disabled in Taskmanager, after a fresh install of Omenmon i still can control everything but the fans, it sits them in a medium state 24/7.

GeographicCone, 2024-01-11:
> Try enabling the [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec) setting, and possibly also [FanLevelNeedManual](https://omenmon.github.io/config#fanlevelneedmanual) in the OmenMon.xml [configuration file](https://omenmon.github.io/config).

Have you tried this?

GeographicCone, 2024-02-07:
All the fan control methods I'm aware of are already implemented in **OmenMon** and can be selected with the different configuration settings listed above.

I'm open to further expanding compatibility with more models but in this case it'd have to depend on someone with the hardware in question contributing a working method to control the fans.

closed 2024-02-07 by GeographicCone