OmenMon/OmenMon #23: OmenMon won't open. OmenMon in terminal doesn't work too apparently

umarhamzy, 2024-01-02:
[image]
Above is what I'm getting when I tried to troubleshoot with the guide provided.
* OmenMon Version: **0.59.0-Release**
* Model & Version: **HP 88ED 80.73** 
* Working (what I've tried):
  * Uninstalled OGH
  * Restarted laptop
  * Ran OmenMon as administrator
* Not Working:
  * GUI won't open and OmenMon -Bios shows above error

Note: 88ED is the Victus 16 (2021) laptop. Tested on Windows 11 23H2

n-elia, 2024-01-02:
Hi!
There is an error in your prompt. You should move in the directory where _OmenMon.exe_ is, and then you have to execute it using its full filename.
Let me give you an example: suppose that you put _OmenMon.exe_ in the `omenmon` folder on your Desktop.
Then, you will need to move into that folder:
`C:\Users\umarh> cd .\Desktop\omenmon\`
And then you can invoke OmenMon.exe using the following syntax:
`C:\Users\umarh\Desktop\omenmon> .\OmenMon.exe -Bios`

Please notice that you were missing the `.\` at the beginning and the `.exe` at the end. Also, if you write "omen" and press TAB, the autocompletion will suggest you the correct name.

umarhamzy, 2024-01-02:
> Hi! There is an error in your prompt. You should move in the directory where _OmenMon.exe_ is, and then you have to execute it using its full filename. Let me give you an example: suppose that you [...]
Thank you so much!
> Please notice that you were missing the `.\` at the beginning and the `.exe` at the end. Also, if you write "omen" and press TAB, the autocompletion will suggest you the correct name.
Appreciate it!

[image]
I ran into this upon running it.

n-elia, 2024-01-02:
> Thank you so much!

You're very welcome!

> I ran into this upon running it.

You are running OmenMon correctly.

The issue here is that you're specifying the Operation Context (`-Bios`), but you are not issuing any command. Please take a look carefully to the [CLI documentation](https://omenmon.github.io/cli)!

Example: `-Bios FanMode=Performance` tells OmenMon to set the Performance mode as Fan Mode in the context of Bios-related operations.

umarhamzy, 2024-01-02:
Your help is very much appreciated! I'm familiarizing myself with the commands now. I hope the GUI is supported soon! Thank you, kindly

GeographicCone, 2024-01-07:
@n-elia wrote:
> Please notice that you were missing the .\ at the beginning and the .exe at the end.

Thank you @n-elia for helping out!

@umarhamzy Regarding the _first_ report, as @n-elia pointed out, the issue is that when you're using _PowerShell_ (as opposed to _Command Prompt_), the current directory is not searched for the executable command unless you explicitly specify `.\` (or the full path) in front of the executable name.

If you foresee using **OmenMon**'s command-line mode often, you could open a _Command Prompt_ window instead by running `start cmd` first, and then typing the commands there instead. Or you could put `OmenMon.exe` (and `OmenMon.xml`) in a directory that is included in your `PATH`, for example `%SystemRoot%\System32` (which resolves to `C:\Windows\System32`).

For the record, the `.exe` extension at the end should be redundant, it should work regardless.

Now, the _second_ issue is that when you run `OmenMon -Bios` it just runs all the available BIOS queries (that don't change any settings). Specifically, at the time of writing, these are:
https://github.com/OmenMon/OmenMon/blob/0998bbd1c2a697a53230ba5dd73c47905f7b7eb5/App/Cli/CliOp.cs#L43-L46

Among these, the `Anim` call to query the [LED animation table](https://omenmon.github.io/cli#anim) fails, which then interrupts the flow. So far this has not been encountered or reported by anyone (for example on my laptop the table is all zeroes and trying to change it does nothing but also doesn't fail) so for now this error is unhandled. It might be worth adding an error check here in a future version but for now please try the other `-Bios` context commands bypassing this one, which can also be done in a single run as follows:

`OmenMon -Bios BornDate System Gpu GpuMode Adapter HasOverclock HasMemoryOverclock HasUndervolt KbdType HasBacklight Backlight Color FanCount FanMax FanType FanLevel FanTable Temp Throttling`

Note the lack of `Anim` in the above list.

There are a lot of disparate models and not all operations work on all of them. Some of those I have already changed to fail silently instead of throwing an exception. But meanwhile, if any other operation errors out, you can also remove it from the list. 

Issue three:

> GUI won't open

What happens exactly?

umarhamzy, 2024-01-07:
> OmenMon -Bios BornDate System Gpu GpuMode Adapter HasOverclock HasMemoryOverclock HasUndervolt KbdType HasBacklight Backlight Color FanCount FanMax FanType FanLevel FanTable Temp Throttling

Thank you for this, I understand what's happening now. 

As for the GUI not opening, when I click on the .exe, it just wouldn't open. My cursor shows the loading icon, but nothing after that, doesn't fire up OmenMon- blank.

GeographicCone, 2024-01-07:
> As for the GUI not opening, when I launch the .exe, it just wouldn't open. My cursor shows the loading icon, but nothing after that, doesn't fire up OmenMon- blank.

Can you try to run it from the command line like before but with no arguments (just `.\OmenMon`)? It should start the GUI then but if it doesn't maybe it throws some error.

By default there is only a notification icon when **OmenMon** starts, no window. But if you run it again when it's already running, it should show the main window and a balloon tip.

umarhamzy, 2024-01-07:
> Can you try to run it from the command line like before but with no arguments (just `.\OmenMon`)? It should start the GUI then but if it doesn't maybe it throws some error.

Can I run it in parallel to OGH for the time being? Or will it not work at all?

umarhamzy, 2024-01-07:
@GeographicCone I just noticed OGH's performance mode only enables cTGP and doesn't enable PPAB (idk the significance in performance boost from the latter). Same with OmenMon's other Auto profiles. But the power profile in "Prog" enables both cTGP and PPAB. The fact that OGH doesn't enable PPAB when on performance mode has me thinking if my GPU (RTX 3050Ti mobile) doesn't support PPAB in the first place. I'd gladly appreciate it if you could clarify this to me. Is that technology not available with my gpu? or is the boost from PPAB not significant enough for OGH to bother enabling it

GeographicCone, 2024-01-07:
Glad you got it working!

You can have multiple [fan programs](https://omenmon.github.io/gui#fan-program), one of which can be launched automatically on startup and/or toggled on and off by pressing the _Omen_ key.

You can also have a different fan program activated whenever the AC power is disconnected, in which case the previous fan program will be re-enabled as soon as AC power is restored.

The two default programs: _Power_ and _Silent_ demonstrate how it works. There are many other possibilities. Have a look here: https://omenmon.github.io/gui

There is a separate page for the `OmenMon.xml` configuration file: https://omenmon.github.io/config

A word of warning though: running both **OmenMon** and _Omen Hub_ at the same time you are likely to run into issues, for example if both attempt to control fan levels.

As for what the cTGP and PPAB toggles do exactly on your model, I don't have an answer. I suggest you run some benchmark together with an overlay that shows you or a tool that records the TGP, and then compare the different settings. On my system, it was 80/100/115W TGP with the three different levels with the default VBIOS, and 80/115/150W after I flashed the BIOS from another model.

umarhamzy, 2024-01-08:
I'm familiarizing myself with the program and getting the hang of it.

> A word of warning though: running both OmenMon and Omen Hub at the same time you are likely to run into issues, for example if both attempt to control fan levels.

Understood, in that case- would a simple uninstall of OGH & restart do the trick?

GeographicCone, 2024-01-08:
> Understood, in that case- would a simple uninstall of OGH & restart do the trick?

I don't know the specifics because I don't install it in the first place. As far as I know, _Omen Hub_ is installed automatically as a hardware support application (HSA) by a driver extension. So if you just uninstall the app, it's possible it will get reinstalled automatically due to this. You may have to remove the driver as well and block it from being reinstalled. @Rupurudu helpfully posted a guide how he did this in #15, check that out.

umarhamzy, 2024-01-08:
Appreciate it, thanks a bunch!

closed 2024-01-08 by GeographicCone