OmenMon/OmenMon #24: a way to change what the dynamic icon in the taskbar is showing

viroholic, 2024-01-02:
I don't know where else to look and ask for this. where do I go so I can change the taskbar icon from showing the TNT5c to showing the CPU temperature? That is something I need to monitor as my CPU gets very hot and I need to keep an eye out. thank you in advance.

originalgod11, 2024-01-06:
hi viroholic, i also have this question, I'm not so well versed in coding and all the rest. i do need to keep track of my GPU temps, and having it in the taskbar would be great, instead, it tracks my TNT5c. i believe it states in the readme how to change which valve is shown, not working for me though.

GeographicCone, 2024-01-07:
The notification icon shows the highest temperature reading across all sensors set to be used, i.e. excluding those with the attribute `Use="false"`:

https://github.com/OmenMon/OmenMon/blob/0998bbd1c2a697a53230ba5dd73c47905f7b7eb5/App/Gui/GuiTray.cs#L311-L315

https://github.com/OmenMon/OmenMon/blob/0998bbd1c2a697a53230ba5dd73c47905f7b7eb5/Hardware/Platform.cs#L177-L181

If the `TNT5` reading is falsely high and obscures the other legitimate readings, you can either remove the sensor entirely in the `<Temperature>` section of the [configuration file](https://omenmon.github.io/config#sensors), or add a `Use="false"` attribute so that it's shown in the main window but otherwise ignored.

GeographicCone, 2024-01-11:
To recap, the icon shows the maximum across all the enabled sensors: https://omenmon.github.io/gui#icon

Which sensors are taken into account can be customized: https://omenmon.github.io/config#sensors

I don't really see a use case where you _wouldn't_ want to see the maximum temperature at any given time, so I believe this is a question of configuring the sensors on your specific laptop model to disable any that do not provide useful information. But if I'm missing something, please reopen and describe how you'd like it to work and I'll consider it as a feature request.

Thank you for using **OmenMon**!

closed 2024-01-11 by GeographicCone