OmenMon/OmenMon #25: Can't open "Show Monitor" GUI

rotgruengelb, 2024-01-04:
## Microsoft .NET Framework Exception when trying to access/click on "Show Monitor"

```
Informationen über das Aufrufen von JIT-Debuggen
anstelle dieses Dialogfelds finden Sie am Ende dieser Meldung.

************** Ausnahmetext **************
System.IndexOutOfRangeException: Der Index war außerhalb des Arraybereichs.
   bei OmenMon.AppGui.GuiKbd.SetColors(ColorTable colorTable)
   bei OmenMon.AppGui.GuiKbd.GetHw()
   bei OmenMon.AppGui.GuiKbd..ctor(GuiTray context)
   bei OmenMon.AppGui.GuiFormMain..ctor()
   bei OmenMon.AppGui.GuiTray.ShowFormMain()
   bei OmenMon.AppGui.GuiTray.EventIconMouseClick(Object sender, MouseEventArgs e)
   bei System.Windows.Forms.NotifyIcon.OnMouseClick(MouseEventArgs mea)
   bei System.Windows.Forms.NotifyIcon.WmMouseUp(Message& m, MouseButtons button)
   bei System.Windows.Forms.NotifyIcon.WndProc(Message& msg)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Geladene Assemblys **************
mscorlib
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9181.0 built by: NET481REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll.
----------------------------------------
OmenMon
    Assembly-Version: 0.59.0.0.
    Win32-Version: 0.59.0.5010.
    CodeBase: file:///C:/Users/Daniel/Downloads/OmenMon-0.59.0-Release/OmenMon.exe.
----------------------------------------
System.Drawing
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9032.0 built by: NET481REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll.
----------------------------------------
System
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9206.0 built by: NET481REL1LAST_B.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll.
----------------------------------------
System.Windows.Forms
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9181.0 built by: NET481REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll.
----------------------------------------
System.Xml
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9032.0 built by: NET481REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll.
----------------------------------------
mscorlib.resources
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9032.0 built by: NET481REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_de_b77a5c561934e089/mscorlib.resources.dll.
----------------------------------------
System.Configuration
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9032.0 built by: NET481REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll.
----------------------------------------
System.Core
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9200.0 built by: NET481REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll.
----------------------------------------
Microsoft.Management.Infrastructure
    Assembly-Version: 1.0.0.0.
    Win32-Version: 10.0.22621.1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Microsoft.Management.Infrastructure/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.dll.
----------------------------------------
Microsoft.Management.Infrastructure.Native
    Assembly-Version: 1.0.0.0.
    Win32-Version: 10.0.22621.1 (WinBuild.160101.0800).
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll.
----------------------------------------
Microsoft.Management.Infrastructure.UserFilteredExceptionHandling
    Assembly-Version: 0.0.0.0.
    Win32-Version: 10.0.22621.1 (WinBuild.160101.0800).
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll.
----------------------------------------
Accessibility
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9032.0 built by: NET481REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll.
----------------------------------------
System.Windows.Forms.resources
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.9032.0 built by: NET481REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_de_b77a5c561934e089/System.Windows.Forms.resources.dll.
----------------------------------------

************** JIT-Debuggen **************
Um das JIT-Debuggen (Just-In-Time) zu aktivieren, muss in der
Konfigurationsdatei der Anwendung oder des Computers
(machine.config) der jitDebugging-Wert im Abschnitt system.windows.forms festgelegt werden.
Die Anwendung muss mit aktiviertem Debuggen kompiliert werden.

Zum Beispiel:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Wenn das JIT-Debuggen aktiviert ist, werden alle nicht behandelten
Ausnahmen an den JIT-Debugger gesendet, der auf dem
Computer registriert ist, und nicht in diesem Dialogfeld behandelt.

```

GeographicCone, 2024-01-07:
Thank you for your interest in **OmenMon** and the report.

This is related to a possibly unsupported keyboard not detected as such.

Can you please say what kind of keyboard backlight type do you have and post your `OmenMon -Bios` output (or just `OmenMon -Bios KbdType HasBacklight Backlight Color` specifically)?

rotgruengelb, 2024-01-07:
This is the output of `OmenMon -Bios KbdType HasBacklight Backlight Color`:

```
OmenMon Hardware Monitoring & Control Utility Version 0.59-Release
BIOS (-Bios)
- Keyboard Type: 0x00 = 0b00000000 = 0 [Standard]
- Keyboard Backlight Support: Yes
- Keyboard Backlight: 0x64 = 0b01100100 = 100 [Off]
- Keyboard Backlight Color Table: 1 Zones - Right: 000000
```

GeographicCone, 2024-01-08:
Great, thank you!

So with your keyboard, are you able to set the backlight to an arbitrary RGB color?

Or is it a single, predetermined color (likely either red or white), which you can only toggle on and off?

Meanwhile, before we get this issue sorted out, you should be able to use the rest of **OmenMon**'s functionality by temporarily setting [BiosErrorReporting](https://omenmon.github.io/config#bioserrorreporting) to **false** in `OmenMon.xml`.

rotgruengelb, 2024-01-08:
> Or is it a single, predetermined color (likely either red or white), which you can only toggle on and off?

Yes Im only able toggle a white key backlight on or off

>Meanwhile, before we get this issue sorted out, you should be able to use the rest of **OmenMon**'s functionality by temporarily setting [BiosErrorReporting](https://omenmon.github.io/config#bioserrorreporting) to **false** in `OmenMon.xml`.

The bios reporting false was already done but did nothing

GeographicCone, 2024-01-08:
Could you please try the test build from this artifact and see if the issue is resolved for you:
https://github.com/OmenMon/OmenMon/actions/runs/7449156009

You should be able to toggle the backlight on and off from the main window or the context menu but the color-related part of the interface should be grayed out and there should be no errors of any kind of course.

rotgruengelb, 2024-01-09:
## Working
- Monitor Opening.

## Not Working
- "Keyboard Backlight & Color" Section Check Mark does nothing when clicked.
- Keyboard > Backlight Dose nothing (Keyboard Backlight doesn't change)

## Test Video

Uploading 2024-01-09 14-31-58.mp4…

GeographicCone, 2024-01-09:
Thank you!
One more thing:

Does the backlight toggle work when you run:
`OmenMon -Bios Backlight=On`
and
`OmenMon -Bios Backlight=Off`
from the command line?

rotgruengelb, 2024-01-09:
```
C:\Users\Daniel\Desktop\OmenMon\experimental>OmenMon -Bios Backlight=Off
OmenMon Hardware Monitoring & Control Utility Version 0.60-Snapshot
BIOS (-Bios)
+ Keyboard Backlight: No
C:\Users\Daniel\Desktop\OmenMon\experimental>OmenMon -Bios Backlight=On
OmenMon Hardware Monitoring & Control Utility Version 0.60-Snapshot
BIOS (-Bios)
+ Keyboard Backlight: Yes
C:\Users\Daniel\Desktop\OmenMon\experimental>OmenMon -Bios Backlight=Off
OmenMon Hardware Monitoring & Control Utility Version 0.60-Snapshot
BIOS (-Bios)
+ Keyboard Backlight: No
```
The backlight did not change for any command

DilutedOxygen1451, 2024-01-09:
> Could you please try the test build from this artifact and see if the issue is resolved for you: https://github.com/OmenMon/OmenMon/actions/runs/7449156009
>
> You should be able to toggle the [...]

i attempted to build the version  linked, however on running the -prepare, it failed, telling me this issue
[image]

DilutedOxygen1451, 2024-01-09:
> Could you please try the test build from this artifact and see if the issue is resolved for you: https://github.com/OmenMon/OmenMon/actions/runs/7449156009
>
> You should be able to toggle the [...]

running "make prepare" results in this error within the main folder results in this

[image]

 and attempting to run (make build) results in an error telling me "OmenMon.exe could not be found"

GeographicCone, 2024-01-09:
@rotgruengelb

> The backlight did not change for any command

Thank you for reporting back. So the backlight toggle WMI calls that seem to work on most *Omen*s are not compatible with that _Victus_ model.

On a general note, I'm really happy **OmenMon** has been found operational, at least partially, on such a variety of laptops, even including some _Envy_ (i.e. non-gaming) models and I always welcome any ideas how to extend the compatibility even further.

As a last resort, even if it doesn't work, **OmenMon** should always fail gracefully, and not with some ugly exception: I'm always willing to address issues like that. Beyond that though, in this case I don't know of any other way for toggling keyboard backlight programmatically, and considering it's likely not a priority issue since there must be some <key>Fn</key> key combination to do just that, I'm afraid I'll have to leave things the way they are for now. I appreciate your understanding.

@DilutedOxygen1451

> I attempted to build the version linked [...]

Apologies for being unclear. It's already been built by GitHub actions, just download the ZIP build artifact, unpack it, and run the `.exe` from there: https://github.com/OmenMon/OmenMon/actions/runs/7449156009/artifacts/1154269868

For the record, if you wanted to build **OmenMon** yourself, you'd have to install _Build Tools_ or _Visual Studio_ version 2022 (17). The details how to do that are available here: https://omenmon.github.io/build#instructions

But I'd never ask or expect anyone to run a build themselves just because they reported an issue. Sorry for the hassle I caused you due to this misunderstanding.

**Thank you both for for giving OmenMon a try!**

The linked preview will become the next release if no issues are found with the fix.

DilutedOxygen1451, 2024-01-09:
ah, I downloaded it and it works flawlessly so far, I haven't tested toggling the backlight because fn+f4 on my keyboard does exactly that so i have no need, but the GUI is working perfectly as far as i can tell, thanks for the help

also its not your fault i misunderstood, im still relatively new to github and from the original artifact link i only found the unbuilt version, as i dont exactly know how to navigate github as much as i should, but the link you subsequently sent works perfectly

rotgruengelb, 2024-01-09:
> As a last resort, even if it doesn't work, **OmenMon** should always fail gracefully, and not with some ugly exception: I'm always willing to address issues like that. Beyond that though, in this case I don't know of any other way for toggling keyboard backlight programmaticall,[...]

Thank you!
I never asked for backlight support and don't need it because there is indeed a <key>fn</key> key combination.

I just needed OmenMon for the fan control and it bothered me how it crashed every time I accedentaly clicked the taskbar icon that opens 
the monitor. 

Consider this issue resolved (at least for me).

Thank you @GeographicCone for your OpenSource contributions to this alternative to the "HP OmenHub garbage".

<3 -Daniel

DilutedOxygen1451, 2024-01-09:
the issue is resolved for me too, thanks for the help, and I'm glad i can finally get rid of HP's garbage software, at least as much as i can

closed 2024-01-09 by rotgruengelb

GeographicCone, 2024-01-10:
Thank you both!
The fix is included in **build 0.60.0**, which [has just been released](https://github.com/OmenMon/OmenMon/releases/latest).