OmenMon/OmenMon #26: Is there a buidl without the RGB GUI

DilutedOxygen1451, 2024-01-09:
attempting to open the GUI immediately crashes it, and from the errors I'm assuming the issue is that my laptop (HP Victus 15-fa0xxx) only has a one tone backlight (white across the whole keyboard), but the software attempts to get the RGB settings, which don't exist, is there a build or config file already made that does not attempt to control the backlight/RGB so i can use the GUI version?

everythign ive tested in the taskbar version seems to work just fine, but the monitor mode just breaks any time i try to open it

(Error Here)
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at OmenMon.AppGui.GuiKbd.SetColors(ColorTable colorTable)
   at OmenMon.AppGui.GuiKbd.GetHw()
   at OmenMon.AppGui.GuiKbd..ctor(GuiTray context)
   at OmenMon.AppGui.GuiFormMain..ctor()
   at OmenMon.AppGui.GuiTray.ShowFormMain()
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
    Assembly Version: 0.59.0.0
    Win32 Version: 0.59.0.5010
    CodeBase: file:///C:/Users/dilut/Downloads/OmenMon-0.59.0-Release/OmenMon.exe
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9206.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9200.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Management.Infrastructure
    Assembly Version: 1.0.0.0
    Win32 Version: 10.0.22621.1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Microsoft.Management.Infrastructure/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.dll
----------------------------------------
Microsoft.Management.Infrastructure.Native
    Assembly Version: 1.0.0.0
    Win32 Version: 10.0.22621.1 (WinBuild.160101.0800)
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
Microsoft.Management.Infrastructure.UserFilteredExceptionHandling
    Assembly Version: 0.0.0.0
    Win32 Version: 10.0.22621.1 (WinBuild.160101.0800)
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

GeographicCone, 2024-01-09:
Same issue has already been reported under #25 (immediately before this one). A fix is available but I've been waiting for feedback from the person that reported it before making a new release.

Please try the test build from the artifact linked in the following comment: https://github.com/OmenMon/OmenMon/issues/25#issuecomment-1881182335 and let's keep the discussion in one place.

closed 2024-01-09 by GeographicCone

GeographicCone, 2024-01-09:
Let me add that I appreciate your report and the fact that you pinpointed where the issue was exactly, which would have been very helpful in fixing it were it a new, separate problem.

The title of the other (prior) issue did not make it immediately apparent that the problem it was referring to was the same. I edited that title now to make it clear.

Thank you for trying **OmenMon**. I'm waiting to hear your feedback on the test build in that other (prior) issue.

GeographicCone, 2024-01-10:
Fixed in **build 0.60.0**, together with #25.