OmenMon/OmenMon #27: "Failed to initialize the embedded controller"

rm1337, 2024-01-09:
So this error always pops up on windows start up but if I just restart the application, it runs fine. Is there any way to fix this? 
[image]

GeographicCone, 2024-01-09:
Hi and thank you for your interest in **OmenMon**!

It'll be difficult to pinpoint the exact cause without more information but let's try:

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Library/LocaleData.cs#L336

That error is triggered when the Embedded Controller fails to initialize:

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Library/Hw.cs#L199-L205

which occurs when:

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Hardware/Ec.cs#L227-L236

the kernel-mode driver fails to load in `Ring0::Open()`

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Driver/Ring0.cs#L101-L147

There is a way to narrow it down further but it'd take compiling a special version that would output `Ring0::GetStatus()`

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Driver/Ring0.cs#L252-L256

As a first attempt at working around this issue, can you try moving `OmenMon.exe` to a different location and running it from there? **OmenMon** needs to be able to extract and load the kernel-mode driver: by default this is the file called `OmenMon.sys` in the same directory. Although it also tries extracting it to a temporary location as a last resort:

https://github.com/OmenMon/OmenMon/blob/6f20cd8fdc3de60f383bfdfe49cae5614b0113e4/Driver/Ring0.cs#L178-L188

There are two attempts and several locations the driver can be extracted to so if it fails, it must be in a very specific situation.

rm1337, 2024-02-05:
Hi

Sorry for delay in responding. I forgot to follow up on this, I tried moving the OmenMon.exe and .sys file as you suggested to another directory. The issue still persists. 

Again, I just want to re-iterate, this only happens whenever I start my laptop, if I just run it again, it works just fine.

GeographicCone, 2024-02-07:
> This only happens whenever I start my laptop

I assume you have _Settings_ → _Start with Windows_ enabled? Can you try disabling that and restarting? Does it also happen when you try to start manually?

----

In any case, it seems the the detailed error message as to why the driver couldn't be loaded should be reported to the user.

GeographicCone, 2024-02-09:
Following up on what I wrote myself:

> [...] the detailed error message as to why the driver couldn't be loaded should be reported to the user.

From now on, there will be an additional detailed error message if the driver fails to load, which should hopefully assist in getting to the bottom of the cause of this issue. Release 0.61.0 will be posted soon, meanwhile the [last-built snapshot](https://github.com/OmenMon/OmenMon/actions/runs/7838749654/artifacts/1232146937) already has this feature as well, if you want to try it.

rm1337, 2024-04-16:
[image]

Hi @GeographicCone 

This is the error on start up and as you mentioned I do have this option enabled: Settings → Start with Windows

I will now disable that option and try to manually start the program and see if I'm facing the same issue.

GeographicCone, 2024-05-06:
OK. So now we know, on the initial attempt it fails with error code `0x431`, which is [ERROR_SERVICE_EXISTS](https://learn.microsoft.com/en-us/windows/win32/debug/system-error-codes--1000-1299-#ERROR_SERVICE_EXISTS) _"The specified service already exists."_ And on the subsequent attempt, it's `0xB7` [ERROR_ALREADY_EXISTS](https://learn.microsoft.com/en-us/windows/win32/debug/system-error-codes--0-499-#ERROR_ALREADY_EXISTS) _"Cannot create a file when that file already exists."_

I think the culprit might be the unusual path it's under and the possibility that's somehow being interfered with by _OneDrive_. I'd suggest changing the path to something more regular:

* Close **OmenMon** if running
* Run **OmenMon** GUI from `C:\Users\hidaa\OneDrive\Documents\Rohit\OmenMon\OmenMon-0.61.1-Release`
* Under the context menu, uncheck _Start with Windows_, _Intercept Omen Key_ and _Advanced Optimus Fix_ under _Settings_ if any were checked
* Close **OmenMon** GUI
* Run **OmenMon** CLI: `C:\Users\hidaa\OneDrive\Documents\Rohit\OmenMon\OmenMon-0.61.1-Release\OmenMon.exe -Ec` a couple of times,
  * This makes sure the file `C:\Users\hidaa\OneDrive\Documents\Rohit\OmenMon\OmenMon-0.61.1-Release\OmenMon.sys` does **not** exist any longer (it should be removed when **OmenMon** is closed)
  * If the file exists, delete it. If it's in use, reboot first. As a matter of fact, reboot anyway, just to be sure.
* Now, move `OmenMon.exe` and `OmenMon.xml` to some "normal" path. If you create the directory `C:\OmenMon` for it, it should be fine. Or you can put it under `C:\Windows\System32` as I do, which has the added benefit that it'll always be in the `PATH` if you want to use the CLI mode.
* Once in the new location, run `OmenMon.exe` again, and you can set it to _Start with Windows_.
* Make sure you only have a single `OmenMon.exe` somewhere, and not accidentally running it from multiple locations.