OmenMon/OmenMon #28: Is there a virus in OmenMon? (Spoiler alert)

GeographicCone, 2024-01-10:
I've noticed there are people out there somewhere discussing if there is a virus in **OmenMon**.

Apparently, recent versions have been flagged by a single piece of some obscure anti-virus software as "AI-detected malware", whatever that is supposed to mean, while being cleared by all the other 67 scanners. You can see the full report on [VirusTotal](https://www.virustotal.com/gui/file/05550297dd98590c408a8e9bf0938d013ca985d6a79d615fc1e93c322f3529d1/detection).

**So is there a virus?**
**No, there is not.**

But don't take my word for it:

* Full source code is posted here on GitHub and you can [inspect it yourself](https://github.com/OmenMon/OmenMon)
* Releases are built with [automated GitHub workflows](https://github.com/OmenMon/OmenMon/tree/master/.github/workflows) so they correspond to the source in the repository
* **OmenMon** is an unobfuscated .NET application, which makes it trivial to inspect the executable file with a tool such as [dnSpy](https://github.com/dnSpyEx/dnSpy)
* It does not connect to the network at all so it has no means to exfiltrate any data from your computer

Why can the file be flagged then? There are a number of possibilities:

* It needs to run as an administrator
* It does things that are "suspicious" to virus scanners such as using a kernel-mode driver (even if it's a well-known, signed one), interacting with _Windows Management Instrumentation_, adding entries to the _Task Scheduler_, etc.
* It is not signed with a digital certificate — as that would only mean extra hassle for me, and something I'd likely have to shed money on, while all it would accomplish is my glorious name being plastered all over your screen when starting **OmenMon**, as opposed to there being a warning about an "unknown" publisher. I figure it's not worth it, especially as in most scenarios, **OmenMon** is rarely started manually anyway.

However, the publisher isn't really unknown: you can still find my name if you look for it. And neither is the source. There is nothing malicious going on, and evidence in support of this can be independently verified.

Please just make sure you only download **OmenMon** builds from the official source, which is:
https://github.com/OmenMon/OmenMon/releases/

Although the software is fairly niche, I can't rule out it'll eventually grow large enough for some malicious actor to take interest in impersonating it. Of course, anyone is allowed to fork the project and make their own build, in accordance with the [terms of the license](https://github.com/OmenMon/OmenMon/blob/master/LICENSE.md). But while all of the above statements apply to the original version in this repository, they won't necessarily be applicable to any of its forks as well.

Thank you for using **OmenMon**!

GeographicCone, 2024-01-15:
To keep everyone interested in the loop regarding the recent false positive report by a certain obscure anti-virus program, I have contacted that software vendor with a request for that automatic determination to be reassessed.

Separately, I have contacted VirusTotal as well, and posted a public comment in the _Community_ section. I also reached out to another obscure third-party website that apes the VirusTotal report and sees the application as "malicious" because of that single positive match in it.

I have then submitted the executable file from the latest release to Microsoft for manual verification as it was reported by @atmagyan in #29 to have been flagged by SmartScreen. Windows Defender currently reports "no malware detected" although "final determination" is still "pending":

[image]

I believe this is the extent of what I can do, short of spending a considerable amount of money on a signing certificate.

If you have questions or comments regarding any anti-virus reports, please feel free to discuss it here.

GeographicCone, 2024-01-17:
The latest **OmenMon** release [0.60.0](https://github.com/OmenMon/OmenMon/releases/tag/0.60.0) has been confirmed as not malware by Microsoft. Any _SmartScreen_ issues should be resolved as of Windows Defender definitions version 1.403.2266.0 from January 15, 2024:

[image]

GeographicCone, 2024-02-09:
New builds from now on include a self-signed certificate, which will hopefully help in establishing reputation in order to prevent false-positive reports. 

If anyone needs to workaround trust issues, it is now possible to download the certificate at https://omenmon.github.io/ca.cer and manually add it to the certificate store with `certutil -user -addstore Root ca.cer` or using the GUI, so that **OmenMon** can be considered trusted. This is an optional step that shouldn't be necessary.

----

In other news, the issue with the original positive report has been resolved but another virus scanner seems to be now flagging **OmenMon** as "malicious (score: 100)" whatever that is even supposed to mean. The same scanner apparently also [identifies part of Microsoft's Windows components as malicious](https://learn.microsoft.com/en-us/answers/questions/1506355/cynet-is-marking-many-of-microsoft-signed-files-as). Go figure.

IssamBouchahine, 2024-02-11:
Reference to issue #40 

I have no doubt that OmenMon is safe and virus free. It just happened to me so I wrote to shine light on it. I am definitely sure that it is a false positive, but bloody Kaspersky is weird haha. Oh and no, I am not a paying Kaspersky user, and will never be, but I will definitely try and report it as a false positive to them. Thanks for the detailed response on #40, and for making OmenMon. Also, I meant no disrespect in my original comment, so apologies if it sounded like that.

GeographicCone, 2024-02-11:
**I should apologize for going off on a rant in my reply to you. It's just that I can't see what I can do about this issue, which I acknowledge is of course real, and it irks me.**

_Some more details below:_

----

So far I've been manually submitting new versions to VirusTotal myself, and adding a comment linking to the repository. I also submit it to Windows Defender for review.

But the processes around it are completely flawed, for example now for the latest version (0.61), [VirusTotal](https://www.virustotal.com/gui/file/05550297dd98590c408a8e9bf0938d013ca985d6a79d615fc1e93c322f3529d1/detection) reports the file as "flagged by 1 sandbox" and the details say:

[image]

It's one thing if **OmenMon** is considered "suspicious" for the things it actually does (using a kernel driver, setting up a WMI event filter, restarting the nVidia Display Container or the Explorer shell process, etc.) but the stuff in that report is something **OmenMon** never attempts at all. It doesn't load `vsstrace.dll`, neither does it attempt to bypass PowerShell execution policy (which, incidentally, is trivially easy to bypass, to the point it hardly serves any purpose). None of this is anywhere in **OmenMon**'s source code. The only plausible explanation is that some other processes that do these things were running in parallel in that sandbox (which would mean it wasn't much of a sandbox to begin with).

In the same spirit, one of the prior versions of **OmenMon** was "accused" of deleting some files. What in fact was going on is that an Adobe Acrobat update was running in the background in the same sandbox, and the installer deleted some files.

VirusTotal's defense is that they _"only aggregate data from a variety of vendors, produce no verdicts of [their] own and as such, [...] are not intended to be an authoritative reputation engine, [...] users [...] can make the best decision. Each decision [is] ultimately the responsibility of users or the security vendors who use the data to improve their services."_ That's an actual quote from the response I received from them. Very convenient, while at the same time ignoring the fact that is completely not how they are marketing their services by claiming to offer _"an objective and unbiased service to [their] users."_ In fact what they're doing is the definition of subjective: apparently anybody can add their "anti-virus" scanner, no matter how poorly trained the "AI" model is, there are no entry criteria, the more the merrier. On top of that, the results can be biased by whatever else was also running at the same time in one or another poorly-set up sandbox, and there's just no accountability of any kind.

**Enough with the backstory. It is what it is. If anyone has an idea what else can be done, please share it. And I want to reiterate, I truly appreciate anyone's feedback regarding any issue in OmenMon. Thank you for your interest in the project, I'm really happy people are using it!**

kytpbs, 2025-09-18:
Reference to issue #105 

The Winring0 driver that OmenMon and many other gaming hardware control apps use has been flagged as malware due to a vulnerability in the driver.

Read more here:
https://support.microsoft.com/en-us/windows/microsoft-defender-antivirus-alert-vulnerabledriver-winnt-winring0-eb057830-d77b-41a2-9a34-015a5d203c42#id0ebd=windows_11