OmenMon/OmenMon #30: Add support for Omen Sequencer

slysherz, 2024-01-18:
I looked around but couldn't find anything about this. Has anyone considered adding support for the Omen Sequencer keyboard?

It seems to have the same features / issues:
- it's also configurable via Omen Light Studio and Gaming Hub (colors, macros, etc)
- an annoyingly hard to configure Omen key instead of Home

GeographicCone, 2024-01-20:
Hi, and thank you for the suggestion. I'm open to all ideas how to extend **OmenMon**, even if I can't implement them personally.

So the _Omen_ key on this keyboard does nothing when you press it, even when **OmenMon** is set to intercept it?

If that's the case, maybe it works like a standard key, just with a special scancode, and can be remapped using the features built into Windows? [SharpKeys](https://github.com/randyrants/sharpkeys) is a tool that makes the whole process easy.

For backlight settings (related to #3), maybe the best course of action would be to look into adding the support for it into [OpenRGB](https://openrgb.org/)?

slysherz, 2024-01-21:
I can't use OmenMon, I get this error when opening the program:
[image]

I guess it requires an HP board to launch?

Just tried SharpKeys, the key appears with code `Unknown: 0x0076 (00_76)`, which is the same for all the macro (P1-P5) keys in the left. It works, but then all macro keys also work as an home button. Interestingly, the macros I had setup in the Omen Gaming Hub continue to work, even with it closed. Maybe they're doing something weird, like storing the macro in the keyboard?

I'll look more into OpenRGB too. Thanks for the help!

GeographicCone, 2024-01-21:
Yes, the HP WMI BIOS interface is required for the GUI to work, since practically all of **OmenMon**'s functionality depends on it.

If that keyboard somehow generates WMI BIOS events, you could add an event filter and consumer in the same way [I did in **OmenHwCtl**](https://github.com/GeographicCone/OmenHwCtl/blob/b3eda0b93269fedf329b9d55409d7ff020e956ac/OmenHwCtl.ps1#L365-L396) (**OmenMon**'s predecessor, a _PowerShell_ script):
````pwsh
        '-OmenKeyOff' {
            Write-Information 'Set Omen Key Off'
            Set-Variable -Name OperationAttempted -Scope Global -Value $True
            Get-CimInstance -CimSession $Session -ClassName '__EventFilter' -Namespace 'root\subscription' `
                -Filter "Name='OmenKeyFilter'" | Remove-CimInstance
            Get-CimInstance -CimSession $Session -ClassName 'CommandLineEventConsumer' -Namespace 'root\subscription' `
                -Filter "Name='OmenKeyConsumer'" | Remove-CimInstance
            Get-CimInstance -CimSession $Session -ClassName '__FilterToConsumerBinding' -Namespace 'root\subscription' `
                -Filter "Filter = ""__EventFilter.Name='OmenKeyFilter'""" | Remove-CimInstance
        }
        '-OmenKeyOn' {
            Write-Information 'Set Omen Key On'
            Set-Variable -Name OperationAttempted -Scope Global -Value $True
            $OmenKeyConsumer = New-CimInstance -CimSession $Session -ClassName 'CommandLineEventConsumer' `
                -Namespace 'root\subscription' -Property  @{`
                    CommandLineTemplate = 'C:\Windows\System32\schtasks.exe /run /tn "Omen Key"';
                    ExecutablePath = 'C:\Windows\System32\schtasks.exe';
                    Name = 'OmenKeyConsumer';
                }
            $OmenKeyFilter = New-CimInstance -CimSession $Session -ClassName '__EventFilter' `
                -Namespace 'root\subscription' -Property @{`
                    EventNameSpace = 'root\wmi';
                    Name = 'OmenKeyFilter';
                    Query = 'SELECT * FROM hpqBEvnt WHERE eventData = 8613 AND eventId = 29';
                    QueryLanguage = 'WQL';
                }
            $OmenKeyBinding = New-CimInstance -CimSession $Session -ClassName '__FilterToConsumerBinding' `
                    -Namespace 'root\subscription' -Property @{`
                        Consumer = [Ref] $OmenKeyConsumer
                        Filter = [Ref] $OmenKeyFilter;
                }
        }
````

This would trigger a scheduled task when the key is pressed. In that repository, there is also [an example task](https://github.com/GeographicCone/OmenHwCtl/blob/master/Omen%20Key%20Task.xml) and [a batch script](https://github.com/GeographicCone/OmenHwCtl/blob/master/Omen%20Key%20Task.cmd) that adds it.

From what you're saying though, these keys do generate regular scancodes, just longer than the regular 2 bytes. This is the case for some standard key combinations as well, such as the _Pause_ key (see for example [this StackOverflow question(https://stackoverflow.com/a/38850436) and the [Microsoft scancode translation table (PDF)](https://download.microsoft.com/download/1/6/1/161ba512-40e2-4cc9-843a-923143f3456c/translate.pdf).

_SharpKeys_ is a great and very useful tool but it's just a GUI frontend for editing the Registry key `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Keyboard Layout\Scancode Map` which is very cumbersome to edit manually. Since that method itself only supports 2-byte scancodes, there is no way to get past that using _SharpKeys_.

What I'd suggest is trying to find out what the full scancodes are for these keys: _maybe_ it's enough to go the address https://keycode.info/ in your browser and press those keys while the window is active. Otherwise, Nir Sofer's [KeyboardStateView](https://www.nirsoft.net/utils/keyboard_state_view.html) utility should be able to get them.

Once you know the scancodes, there are multiple ways to capture them. I'd consider checking out AutoHotkey. See: https://www.autohotkey.com/docs/v2/KeyList.htm#SpecialKeys

slysherz, 2024-01-22:
I didn't know anything about the WMI BIOS interface, I'm sorry if I'm way off track here ahah.

I tried running these scripts with the `-OmenKeyOn` but it doesn't seem to change anything. I thought maybe the className is different so I tried to look for it with `Get-CimClass | Select-Object -ExpandProperty CimClassName` but there's nothing mentioning `hp`, `omen` or `sequencer`. The only ones that could be related are `Win32_Keyboard` and `CIM_Keyboard`, but those seem like general purpose keyboard stuff.

So either the WMI BIOS interface doesn't exist or I don't know how to find it.

I tried both keycode.info and KeyboardStateView:
- both report just the F24 key being pressed both for the Omen key and the P keys
- if I remap F24 to home with PowerToys, only Home appears
- if I bind the Home key to P2 (using the Omen program), then uninstall it, it reports the F24 key + Home being pressed

PowerToys can remap keys like pause just fine, so I don't think long scan codes are the issue. Also it seems the keyboard can store the keybinds by itself (but the omen program doesn't support configuring the Omen key, only the P keys).

I think I will need to dig through the USB communication to understand more -.-'

GeographicCone, 2024-01-23:
> I tried running these scripts with the -OmenKeyOn but it doesn't seem to change anything. 

It won't work as is but you can adapt this approach to your situation.

> I thought maybe the className is different

In this particular case, you could modify the numerical values on the following line to capture other, different events:

````pwsh
Query = 'SELECT * FROM hpqBEvnt WHERE eventData = 8613 AND eventId = 29';
````

Here's yet another approach for other  _HP Envy_ laptop keys: https://github.com/dcartermath/hp-envy-special-keys

But that's all for built-in laptop keyboards, which are a different thing entirely. I don't know how that Sequencer keyboard works. If the mappings are persistently stored in the firmware, you could probably map the extra keys to some unused scancodes using the original utility first, and then map these scancodes to whatever keys you want them to be by editing the Registry, possibly using _SharpKeys_. This way the latency would be minimal and you wouldn't have to have anything extra running in the background.

I'm afraid that's all I can help you with. On a general note, if you want a truly customizable keyboard, consider getting something with the QMK firmware: https://qmk.fm/

slysherz, 2024-01-26:
That's alright, you've helped a ton already. Thank you!

GeographicCone, 2024-02-09:
I'll close the issue as I don't foresee any new activity but please feel free to reopen anytime.

closed 2024-02-09 by GeographicCone

slysherz, 2024-02-13:
Alright. If someone is looking into this, I found a way to control the lights through USB:

https://github.com/slysherz/lights-for-omen-sequencer

GeographicCone, 2024-02-13:
@slysherz That's great news! Congratulations!

Let's keep the issue open then. I've also added a mention of your project to the `README.md` in **OmenMon** repository.