OmenMon/OmenMon #31: More options for Omen Key button intercept

afterthawed, 2024-01-25:
Hi, firstly just want to say thanks for your hard work on this fantastic program! It does pretty much everything you could need it to.

I'd like to make a couple of feature requests if possible, specifically in regards to the Omen Key button and KeyToggleFanProgram config option.

1. Currently, the KeyToggleFanProgram option toggles the default profile on and off when you press the Omen Key. I'd love an option to instead cycle between all availble Fan Programs instead. I had a look to see if it was possible with the CLI and a batch script or something, but I couldn't figure it out with my limited skills

2. Is it possible to get an config option to not have the OmenMon main window open when pressing the Omen Key? If using the key to toggle fan programs, I find it's not that useful also having the main window open, maybe it's just me.

3. An option to send a windows notification when Fan Programs change would also be amazing.

Completely understand if none of this is feasible, just thought I'd try my luck. Thanks!

GeographicCone, 2024-02-07:
These are actually very good suggestions. I'll look into implementing them. Thank you!

----

> I had a look to see if it was possible with the CLI and a batch script or something

It's doable and scriptable from the command-line mode: https://omenmon.github.io/cli#prog with some hacking but I agree that's not very convenient.

> Is it possible to get an config option to not have the OmenMon main window open when pressing the Omen Key?

The reason for doing it this way was to avoid inadvertently disabling the fan program after accidentally pressing the _Omen_ key and receiving no feedback, and then wondering why the performance has tanked. But I agree this is not the best course of action to address that scenario.

> An option to send a windows notification when Fan Programs change

I've avoided this because any standard balloon-tip notification (assuming that's what you have in mind) leaves a message in the action center that has to be discarded manually, which I find annoying. I'll look into how to implement such a notification in a transient way that would disappear on its own.

GeographicCone, 2024-02-09:
> I'll look into implementing them.

I've added the features you asked for, and with them some new configuration options: 

* [KeyToggleFanProgramCycleAll](https://omenmon.github.io/config#keytogglefanprogramcycleall) to cycle through all fan programs with the _Omen_ key instead of toggling the default fan program on and off
* [KeyToggleFanProgramShowGuiFirst](https://omenmon.github.io/config#keytogglefanprogramshowguifirst) to stop showing the application window on first _Omen_ key press if it's hidden
* Showing a balloon tip notification whenever the fan program is changed by pressing of the _Omen_ key and the main window is hidden, which can optionally be disabled with [KeyToggleFanProgramSilent](https://omenmon.github.io/config#keytogglefanprogramsilent)

Release 0.61.0 will be posted soon, meanwhile the [last-built snapshot](https://github.com/OmenMon/OmenMon/actions/runs/7838749654/artifacts/1232146937) already has this feature as well, if you want to try it.

GeographicCone, 2024-02-13:
Seems to be working fine. Closing as implemented.

closed 2024-02-13 by GeographicCone

afterthawed, 2024-02-19:
Using the update now and it works amazingly. The notifications are perfect! Thanks so much.