OmenMon/OmenMon #32: HP Victus 16-d1120TX Laptop I5-12500H issue with standby mode

mikor-85, 2024-01-26:
Thanks for great program!
Everything works well, but there is one problem. When I close the lid of the laptop (it goes into standby mode), after some time (one minute or two) the GPU fan turns on at full speed and runs for about 10...15 seconds. Then it turns off. And so everything repeats itself. I have to turn off my customized program and return the setting to default.
I am attaching my configuration file. I use silent mode.
Is it possible to do something in this case?
[OmenMon.zip](https://github.com/OmenMon/OmenMon/files/14063225/OmenMon.zip)

Rupurudu, 2024-01-26:
I am suspecting that your computer does not enter sleep mode properly.

Type these commands to the command prompt and paste the result here, 

```
powercfg /devicequery wake_armed
```
```
powercfg /a
```

mikor-85, 2024-01-26:
Here are the results.

```
C:\Windows\System32>powercfg /requests
DISPLAY:
No.

SYSTEM:
No.

AWAYMODE:
No.

EXECUTION:
No.

PERFBOOST:
No.

ACTIVELOCKSCREEN:
No.
```

```
C:\Windows\System32>powercfg /devicequery wake_armed
Realtek Gaming GbE Family Controller
```

```
C:\Windows\System32>powercfg /waketimers
There are no active wake timers on the system.
```

```
C:\Windows\System32>powercfg /lastwake
Wakeup log countdown - 1
Awakening Journal [0]
  Countdown of awakening sources - 0
```

```
C:\Windows\System32>powercfg /a
The following sleep states are available on this system:
    Standby (S0 low power idle) Network connected
    Hibernate
    fast Startup

The following sleep states are not available on this system:
    Standby (S1)
       The system firmware does not support this standby mode.
       This standby mode is disabled when S0 low power idle is supported.

    Standby (S2)
        The system firmware does not support this standby mode.
        This standby mode is disabled when S0 low power idle is supported.

    Standby (S3)
       The system firmware does not support this standby mode.
       This standby mode is disabled when S0 low power idle is supported.

    Hybrid Sleep
        Standby (S3) is not not available.        
        The hypervisor does not support this standby mode.
```

Rupurudu, 2024-01-26:
> Here are the results.

> ```
> C:\Windows\System32>powercfg /devicequery wake_armed
> Realtek Gaming GbE Family Controller
> ```
Do you use Wake-on-LAN function on your Ethernet adapter? If not you can disable the adapter from waking up your computer.

1. Open the Device Manager
2. Find your Ethernet adapter from the list, right-click and go to properties
3. Tap Power-Management and uncheck “Allow this device to wake the computer”.

> ```
> C:\Windows\System32>powercfg /a
> The following sleep states are available on this system:
>     Standby (S0 low power idle) Network connected
>     Hibernate
>     Fast Startup
>
> The following sleep [...]

This explains it. Your laptop does not support proper sleep mode (S3), only low power idle mode, which allows all of your applications (Including OmenMon) to run in the background. 

There is an [endless](https://www.reddit.com/r/windows/comments/10ioxzg/will_microsoft_ever_fix_the_abomination_called/) [list](https://www.reddit.com/r/Dell/comments/x61qtb/i_am_tired_of_modern_standby_so_i_am_letting/) [of](https://answers.microsoft.com/en-us/insider/forum/all/xps-15-7590-overheating-when-idle-or-asleep-in-s0/8a521fc8-4892-4c47-a243-9e50e6e714d6) [Reddit](https://www.reddit.com/r/Windows11/comments/109ma0v/windows_modern_standby_is_physically_degrading_my/) [and](https://www.reddit.com/r/gadgets/comments/pv20ry/some_windows_10_laptops_will_overheat_in/) [forum](https://notebooktalk.net/topic/854-windows-11-modern-standby-problems/) [posts](https://forums.tomshardware.com/threads/laptop-getting-super-hot-during-transport.3749276/) [about](https://community.frame.work/t/extreme-burning-overheating-with-the-lid-closed/13983/5) [how](https://www.reddit.com/r/Windows10/comments/11haft8/you_cannot_disable_modern_standby_on_windows_10/) [Modern Standby](https://answers.microsoft.com/en-us/surface/forum/all/surface-laptop-3-overheats-in-sleep-mode-windows/49694ff3-8e41-4ffb-9cd5-27ea5fd054a2) [has](https://www.reddit.com/r/ASUS/comments/puw6j9/finally_a_fix_for_modern_standby_s0_sleep/) [caused](https://answers.microsoft.com/en-us/windows/forum/all/how-to-stop-laptop-from-overheating-in-hibernate/ae116ef4-528b-4ee1-b49c-71acce1e62cc) [problems](https://www.elevenforum.com/t/modern-standby-what-a-pain-it-is.8548/).

It’s even been covered by [news outlets](https://www.laptopmag.com/news/nasty-windows-flaw-causing-dell-lenovo-and-asus-laptops-to-overheat) and [tech youtubers](https://www.youtube.com/watch?v=OHKKcd3sx2c)

Check your BIOS settings to see if there is an toggle for disabling Modern Standby and enabling S3 sleep. My Victus has one. If there is no BIOS option for that then your options are limited.

You can try [disabling modern standby](https://www.elevenforum.com/t/disable-modern-standby-in-windows-10-and-windows-11.3929/) or [disabling network connectivity on modern standby](https://www.elevenforum.com/t/enable-or-disable-modern-standby-network-connectivity-in-windows-11.3286/) if there is not an option in the BIOS, however disabling modern standby can break sleep feature entirely. (disabling just the network connectivity is safe)

You can also ask @GeographicCone to detect and pause the fan program while device is in modern standby, but this will only fix the fans ramping up, not the other issues.

mikor-85, 2024-01-27:
> Check your BIOS settings to see if there is an toggle for disabling Modern Standby and enabling S3 sleep.
Unfortunately the BIOS does not support S3. Now I have completely abandoned S0 mode in favor of hibernation. Taking into account the SSD, turning off and waking up takes approximately the same amount of time. In this case, there are no problems with the OmenMon.
Thanks for the information and useful links!

GeographicCone, 2024-02-07:
Thank you (yet again) @Rupurudu for the amazingly detailed troubleshooting, explanation and suggestions.

During proper sleep or standby, **OmenMon**, or any other user-space application, should not have any control over what is going on. 

A workaround as described by @Rupurudu should be possible by reacting to a [WM_POWERBROADCAST](https://learn.microsoft.com/en-us/windows/win32/power/wm-powerbroadcast) message and disabling the fan program accordingly, to be re-enabled upon resume.

Personally I'd recommend disabling the so-called "modern" standby for all the issues it can cause but if there's demand for this fix, it can be implemented. It'd be an opt-in configuration setting, in order to avoid any extra operations causing a potential slowdown in situations where it's not necessary.

GeographicCone, 2024-02-09:
I've now added the feature to pause fan program on sleep, standby or suspend, and to re-enable it on resume. This feature is enabled by default, if you want to disable it, set [FanProgramSuspend](https://omenmon.github.io/config#fanprogramsuspend) to **false** in `OmenMon.xml`.

Release 0.61.0 that includes this change will be posted soon, meanwhile the [last-built snapshot](https://github.com/OmenMon/OmenMon/actions/runs/7838749654/artifacts/1232146937) already has this feature as well, if you want to try it.

Since I'm not using "modern" standby myself, I'm not certain if it is being detected correctly. If anyone's using it, please let me know.

mikor-85, 2024-02-10:
> Since I'm not using "modern" standby myself, I'm not certain if it is being detected correctly. If anyone's using it, please let me know.

Unfortunately, version 0.61 did not fix the situation in my case. I copied the settings file with the addition of the _FanProgramSuspend_ parameter from version 0.6 to the folder with the new version. I don't think that's the problem.

GeographicCone, 2024-02-10:
OK. Thank you for the report. Just to be sure what's really going on here, can you please try one more time with the special version attached, which keeps a console window open to show some extra information: [OmenMon-Suspend-Resume-Detection-Test.zip](https://github.com/OmenMon/OmenMon/files/14229078/OmenMon-Suspend-Resume-Detection-Test.zip)

* Make sure a fan program is running
* Suspend the system
* When you resume, if the detection is working, it should show something like this:

[image]

Please let me know what it shows, or if it doesn't show anything.

mikor-85, 2024-02-10:
Everything seems to be working as it should now, the fan won't turn on.
```
21:38:10 Into callback, type 0x0004
21:38:10 On suspend
21:38:11 Fan program paused
21:39:24 Into callback, type 0x0012
21:39:24 On resume
21:39:25 Fan program re-enabled
21:39:25 Into callback, type 0x0007
21:41:38 Into callback, type 0x0004
21:41:38 On suspend
21:41:39 Fan program paused
21:42:55 Into callback, type 0x0012
21:42:56 On resume
21:42:57 Fan program re-enabled
21:42:57 Into callback, type 0x0007
```

The only point is that the log was recorded only when running on battery power (I tried it 2 times). If mains power was connected, no entries appeared.

GeographicCone, 2024-02-10:
**Thank you! Good to know it's somewhat working, in principle.**

For anyone interested and also as a note to future self, I'll summarize what's going on here:
_(scroll down for TL,DR below)_

----

In a truly suspended state, no user-mode application should be able to do anything anyway, which is why disabling the fan program for the most part shouldn't matter in these scenarios. However, Microsoft's "modern" standby, the renamed "connected standby" back from the Windows 8 days when Microsoft was trying to turn every PC into a Windows Phone, an approach which largely failed… Well, "modern" standby is nothing like the real standby: applications are allowed to, sort of, run, while they're being, sort of, throttled — but also kept in the dark about when that happens.

Since in **OmenMon**'s case such circumstances lead to erratic fan levels according to the report, and further considering that a custom fan program shouldn't be necessary when the device is in a low-power state, the obvious solution is to disable the fan program when the system is going into standby ("modern" or otherwise), and re-enable it when it resumes.

For that to happen though, **OmenMon** needs to be notified by the operating system whenever standby and resume power events are triggered. From Microsoft's documentation for the [Desktop Activity Moderator](https://learn.microsoft.com/en-us/windows/win32/w8cookbook/desktop-activity-moderator) (DAM), the service in charge of "modern" standby:

> If your desktop app is impacted by the DAM, you can request notifications before suspension is engaged [...] using one of these methods:

> If your app has a window (`HWND`) and you want to handle these notifications through your window procedure, call `RegisterSuspendResumeNotification` to register for these messages (or `UnregisterSuspendResumeNotification` to unregister). You can use `DEVICE_NOTIFY_WINDOW_HANDLE` in the _Flags_ parameter, and pass your window’s `HWND` in as the _Recipient_ parameter. The message received is the `WM_POWERBROADCAST` message.

> If your app does not have a window (`HWND`) or you want a direct callback, call `PowerRegisterSuspendResumeNotification` to register for these messages (or `PowerUnregisterSuspendResumeNotification` to unregister). You must use `DEVICE_NOTIFY_CALLBACK` in the _Flags_ parameter and pass a value of type `PDEVICE_NOTIFY_SUBSCRIBE_PARAMETERS` in the _Recipient_ parameter.

The former is not directly feasible for **OmenMon**, since the application lives in the system tray and is not necessarily guaranteed to hold a window handle at any given point. Thus, I implemented the latter: [PowerRegisterSuspendResume(DEVICE_NOTIFY_CALLBACK, ...)](https://learn.microsoft.com/en-us/windows/win32/api/powerbase/nf-powerbase-powerregistersuspendresumenotification) from `powrbase.h` (the Power Profile API).

It's then up to the operating system to provide the notifications, and this is precisely what it fails to do with regard to the `PBT_APMSUSPEND` and `PBT_APMRESUMESUSPEND` (or any other, for that matter) notifications when entering or leaving "modern" standby while on AC power, as it would seem.

Reading on from the DAM documentation:

> When DAM suspension is engaged or disengaged, DAM triggers delivery of a `WM_POWERBROADCAST` message to those processes subject to suspension that have opted-in to message delivery (via API call or compatibility shim, described later). After a few seconds delay, DAM suspends the process.

So far so good. However:

> There are no notifications when DAM throttling is engaged or disengaged. Processes should not need modification; processes continue to function, albeit at a slower rate.

So, Windows is not _suspending_ **OmenMon**, only _throttling_ it, and it's been decided the process doesn't need to know (never mind it registered for the notification). Whether this is to be considered a bug or a feature is in the eye of the beholder.

So what can be done about it then?

First of all, it would be worth it to check if "modern" standby notifications arrive when registered with [RegisterSuspendResumeNotification(..., DEVICE_NOTIFY_CALLBACK)](https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-registersuspendresumenotification) from `winuser.h` (the Win32 User API). Even though these methods are supposed to be equivalent, one can never be sure. Next in line would then be _RegisterSuspendResumeNotification(..., DEVICE_NOTIFY_WINDOW_HANDLE)_.

I said before that **OmenMon** doesn't necessarily _have_ a window handle, unless the main window form has already been created, which is an obstacle. However, this is not strictly true: there is in fact a handle available but it's not supposed to be accessible from the high-level "managed code."

Mostly for the sake of it, I tried subscribing the top secret `TimerNativeWindow`, which _does_ in fact exist and has a handle at any point the **OmenMon** GUI is running (but nobody is supposed to know about it, so shhh…), to `WM_POWERBROADCAST` notifications. Alas, the notification filter won't receive these messages, even though it does get some others for that window, and `TimerNativeWindow`'s `WndProc` can't be overriden. It's one of the situations when a problem that would be simple to solve in C/C++ becomes insurmountable when C# gets in the way.

[... cut ...] f, not "still on a bit, but not really." It's an [extremely convoluted and contrived solution](https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/prepare-software-for-modern-standby) in search of a problem. You'd never knowingly enable it yourself, it's only enabled because that's the default.

----

**Bonus:** A simplified diagram from Microsoft that "depicts [the] phases of [the] Modern Standby session flow:"

[image]

So simple and straightforward… What can go wrong?

GeographicCone, 2024-02-13:
@n-elia I noticed your [Reddit comment](https://old.reddit.com/r/Hewlett_Packard/comments/17qxf87/to_envy_15_owners_i_suggest_you_to_avoid_updates/kj4z7sz/):

> The issue that I have at the moment is that the fan mode gets reset when the laptop locks (e.g. when going standby).

This feature I've just added in version 0.61 could be adapted to also restore the performance mode on resume from standby (if no fan program running).

However, if you're using Modern Standby, then perhaps it won't detect going into standby or resuming at all, as @mikor-85 reported when on AC power (at which point an additional workaround is still possible, as outlined above).

Anyway, just a heads-up, thought this might be of interest to you.

n-elia, 2024-02-13:
@GeographicCone thank you very much, I really appreciate it! I will definitely take a look into it!