OmenMon/OmenMon #33: There is a new temperature sensor with a constantly high value

mikor-85, 2024-01-28:
One day ago, an unknown temperature sensor appeared on my HP Victus 16-d1120TX laptop with a constant value of 67 degrees Celsius. Sometimes it shows my storage temperature, but actually my SSD temperature is less than 50 degrees Celsius. Accordingly, the turnover of fans has increased. I tried a reboot as well as a system restore before I installed OmenMon for the first time but nothing helped.
I have attached a gif animation of how this happens.
Please, help.
[image]

mikor-85, 2024-01-28:
I learned the https://omenmon.github.io/config#sensors section and disabled these two problematic sensors. Everything began to work as before. But the question remains, why did this happen?

GeographicCone, 2024-02-07:
Thank you for using **OmenMon**!

Glad you got it sorted out.

Yeah, it's likely that the configuration settings, including the sensors in particular, have to be adjusted, unless you happen to have exactly the same hardware as mine — and even then, since in the example configuration file all of the sensors are enabled by default, while using only `CPUT` and `GPTM` should be enough in most situations, and can be beneficial in that it limits EC use.

I believe it's all covered in the documentation. If anything is missing or unclear please let me know.

> But the question remains, why did this happen?

None of these low-level ways to monitor and control the hardware are documented by HP. The way **OmenMon** is able to do all it does is based on reverse engineering how things work on the specific laptop model I have, in particular, as far as the Embedded Controller is concerned, from decompiling and studying the ACPI DSDT (Advanced Configuration and Power Interface Differentiated System Description Table).

HP churns out hundreds of laptop models each year. Or, more accurately, orders them with several different ODMs, which means that even though they all have the HP logo plastered on them, they might have very little in common in terms of actual underlying hardware design. Considering all that, I'd say it's pretty much amazing that with only the feedback from other users, it's been possible to make **OmenMon** work, for the most part, on a wide spectrum of not just _Omen_ but also _Victus_ and _Envy_ models. 

But that some things will work differently or not work at all on different models is to be expected. In particular, some sensors might not report valid temperature data on other models, which cannot be detected automatically. This is what the configuration is for, and I encourage everyone to post the settings that work best on their model.

I believe this specific issue is resolved but feel free to reopen if need be.

closed 2024-02-07 by GeographicCone