OmenMon/OmenMon #34: Error idk

peter9811, 2024-02-03:
I think i have NET 4, 5 and 8 

`Consulte el final de este mensaje para obtener más detalles sobre cómo invocar a la depuración 
Just-In-Time (JIT) en lugar de a este cuadro de diálogo.

************** Texto de la excepción **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Failed to make the BIOS call
   en OmenMon.Hardware.Bios.Bios.Check(Int32 code, Boolean force)
   en OmenMon.Hardware.Bios.BiosCtl.GetGpuPower()
   en OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   en OmenMon.Hardware.Platform.Settings.GetGpuPower(Boolean forceUpdate)
   en OmenMon.Hardware.Platform.Settings.GetGpuDState(Boolean forceUpdate)
   en OmenMon.AppGui.GuiFormMain.UpdateSys()
   en OmenMon.AppGui.GuiFormMain.UpdateAll()
   en OmenMon.AppGui.GuiFormMain..ctor()
   en OmenMon.AppGui.GuiTray.ShowFormMain()
   en OmenMon.AppGui.GuiTray.EventIconMouseClick(Object sender, MouseEventArgs e)
   en System.Windows.Forms.NotifyIcon.OnMouseClick(MouseEventArgs mea)
   en System.Windows.Forms.NotifyIcon.WmMouseUp(Message& m, MouseButtons button)
   en System.Windows.Forms.NotifyIcon.WndProc(Message& msg)
   en System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Ensamblados cargados **************
mscorlib
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    Código base: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
    Versión del ensamblado: 0.60.0.0
    Versión Win32: 0.60.0.5012
    Código base: file:///C:/Users/Peter/Downloads/OmenMon-0.60.0-Release/OmenMon.exe
----------------------------------------
System.Drawing
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9032.0 built by: NET481REL1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9032.0 built by: NET481REL1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
mscorlib.resources
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9032.0 built by: NET481REL1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_es_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
System.Configuration
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9032.0 built by: NET481REL1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
Accessibility
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9032.0 built by: NET481REL1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
Microsoft.Management.Infrastructure
    Versión del ensamblado: 1.0.0.0
    Versión Win32: 10.0.22621.1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Microsoft.Management.Infrastructure/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.dll
----------------------------------------
Microsoft.Management.Infrastructure.Native
    Versión del ensamblado: 1.0.0.0
    Versión Win32: 10.0.22621.1 (WinBuild.160101.0800)
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
System.Core
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Management.Infrastructure.UserFilteredExceptionHandling
    Versión del ensamblado: 0.0.0.0
    Versión Win32: 10.0.22621.1 (WinBuild.160101.0800)
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_64/Microsoft.Management.Infrastructure.Native/v4.0_1.0.0.0__31bf3856ad364e35/Microsoft.Management.Infrastructure.Native.dll
----------------------------------------
System.Windows.Forms.resources
    Versión del ensamblado: 4.0.0.0
    Versión Win32: 4.8.9032.0 built by: NET481REL1
    Código base: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_es_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Depuración JIT **************
Para habilitar la depuración Just In Time (JIT), el archivo de configuración de esta
aplicación o equipo (machine.config) debe tener el
valor jitDebugging establecido en la sección system.windows.forms.
La aplicación también se debe compilar con la depuración
habilitada

Por ejemplo:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Cuando esté habilitada la depuración JIT, cualquier excepción no controlada
se enviará al depurador JIT registrado en el equipo
en lugar de controlarlo mediante el cuadro de diálogo.

`

Rupurudu, 2024-02-05:
Open `OmenMon.xml` and change `BiosErrorReporting` to `false`.

GeographicCone, 2024-02-07:
Thank you for trying **OmenMon**!

The WMI BIOS query to check GPU power settings is not supported on your model, either because the TGP is not adjustable, or perhaps it's controlled in a different way:

> OmenMon.Hardware.Bios.BiosException: BIOS call failed: Failed to make the BIOS call
> OmenMon.Hardware.Bios.BiosCtl.GetGpuPower()

As @Rupurudu said, you can enable the configuration option to disable BIOS error reporting, and still use **OmenMon**'s remaining functionality, to the extent it works for you.

Details: https://omenmon.github.io/config#bioserrorreporting