OmenMon/OmenMon #35: OmenMon Freezing/BSOD

memirsoydal, 2024-02-06:
I don't know why but OmenMon caused a BSOD, I have tried without it and nothing happened. This issue, and freezing, started to occur after I upgraded after 0.56 version. First with 0.59, now with 0.60. 

There were 20 Error Logs in the Event Viewer with the description as follows:

**: The embedded controller (EC) did not respond within the specified timeout period. This may indicate that there is an error in the EC hardware or firmware or that the BIOS is accessing the EC incorrectly. You should check with your computer manufacturer for an upgraded BIOS. In some situations, this error may cause the computer to function incorrectly.**

Is this a compability issue with OmenMon? I'll try to update my BIOS and share an update.

GeographicCone, 2024-02-07:
**OmenMon** as such should not cause a BSOD. What was the stop code?

That being said, together with the log entries you are mentioning, it could be related to Embedded Controller workload. A couple of users have reported this before, which is why in the recent versions I've been progressively trying to add more configuration settings that allow limiting EC usage. As a matter of fact, it's now possible to run **OmenMon** without any EC interactions happening in the background, although only in the limited scenario when `BIOS` is used as the single temperature sensor.

On the **OmenMon** side, I'd suggest going through the configuration settings, described here: https://omenmon.github.io/config and adjusting them in a way that limits EC usage: it's generally enough to leave only two temperature sensors: `CPUT` and `GPTM`, disable the dynamic icon, keep the main window hidden, possibly increase `UpdateProgramInterval`. Many new settings have been added in the last couple of versions, if your configuration file has not been updated, maybe it'd be best to start with the current sample as a template:  https://github.com/OmenMon/OmenMon/blob/master/OmenMon.xml

Separately, consider if there are any other programs running on your system that could cause increased EC workload. These include hardware monitors such as _AIDA64_ or _HwInfo_, especially if also configured to read data from the EC. (That being said, I have, among others, _MSI Afterburner_, _RTSS_ and _ThrottleStop_ running in the background all the time, and also occasionally use the mentioned hardware monitors with no issues.)

This is on the assumption the EC usage is even the culprit here: by deliberately overloading the EC, I've only been able to get the Event Log messages, and the system going to sleep suddenly but never a BSOD.

memirsoydal, 2024-02-07:
There wasn't a stop code if I remember correctly. I've updated my BIOS now and will configure the settings to minimize EC usage. And no, I wasn't using any other programs that could put work on the EC. My PC froze, but OmenMon (dynamic icon) was changing, then BSOD. So probably lightening the load on EC will prevent it in the future.