OmenMon/OmenMon #37: Omenmon runs abnormally on Omen 9 China version (same as #36)

Error-106, 2024-02-08:
Invalid attempt to modify fan speed and CPU power consumption limits
My BIOS version is the latest, and I noticed that the system automatically locks the fan to automatic and forces the CPU power consumption to be locked to 55 watts. After I completely exited omenlab for a few seconds (wtf this)
By the way, I detected SMDE and SME1 being used to control fan speed using EcMon

Error-106, 2024-02-08:
GPU Custom Total Graphics Power also seems to be unable to modify correctly
Fortunately, I successfully attempted to modify PL1 within a few seconds of exiting the Omen Hub

Error-106, 2024-02-10:
- Unfortunately, I didn't try OmenMon before the update, and my attempts to downgrade the bios were unsuccessful
- My omen hub was installed from the Microsoft Store
- I found out that SMDE and SME1 are used for my computer's fan control, 0 is automatic, 60 is full speed (full fan speed is 6000rpm) and the custom speed seems to be 42-60 (4200-6000rpm)

I can try to help, and if I can't do it, I can provide remote access to my computer.

GeographicCone, 2024-02-10:
You can most likely go to an earlier BIOS version if you use a hardware programmer like the CH341A. Only problem is if you don't have a dump from your own flash chip, you'd have to construct an image that merges the downloadable one with your system-specific data.

With the CPU you have, it'd be worth it to undervolt it, if you haven't already. _ThrottleStop_ is great for that.

WMI fan control would be better than using the EC, which has side effects. It's interesting why fan control via WMI wouldn't work, what they changed. Maybe it's the "secret" sign?
https://github.com/OmenMon/OmenMon/blob/db4b7f9360dd57ee2713da85244293243b86a79b/Hardware/BiosData.cs#L16-L19
But then, since it's been the same "`SECU`" for years, would they really change it now?

I take it some of the WMI calls _do_ work though? Can you post the output of your `OmenMon -Bios`?

If it can be done with _Omen Gaming Hub_, it must be there somewhere. Just a matter of sifting through the garbage:

[image]

The basic structure of the calls is like that:

`NativeRpcClient.dll`
↑
`Hp.Bridge.Client.NativeMethods::RpcExecuteCommand()`
↑
`Hp.Bridge.Client.RpcClient::ExecuteCommand()`
↑
`Hp.Bridge.Client.SDKs.WMISDK.WMIHelper::RunCustomCapabilityCommand<T>()`
↑
`Hp.Bridge.Client.SDKs.WMISDK.WMIHelper::ExecuteBiosWmiCommand()`
`Hp.Bridge.Client.SDKs.WMISDK.WMIHelper::ExecuteBiosWmiCommandAsync()`
↑ ← `Hp.Bridge.Client.SDKs.WMISDK.JsonObjects.ExecuteBiosWmiCommandRequest()`
`HP.Omen.Core.Common.OmenHsaClient::BiosWmiCmd_Get(System.Int32)`
`HP.Omen.Core.Common.OmenHsaClient::BiosWmiCmd_Get(System.Int32,System.Int32,System.Byte[],System.Int32,System.Int32)`
`HP.Omen.Core.Common.OmenHsaClient::BiosWmiCmd_GetSync(System.Int32,System.Int32,System.Byte[],System.Int32,System.Int32)`
`HP.Omen.Core.Common.OmenHsaClient::BiosWmiCmd_Set(System.Int32,System.Byte[])`
`HP.Omen.Core.Common.OmenHsaClient::BiosWmiCmd_Set(System.Int32,System.Int32,System.Byte[])`
↑
`HP.Omen.Core.Common.OmenHsaClient::SetFanModeAsync(HP.Omen.Core.Common.PowerControl.Enum.PerformanceMode,System.Boolean,System.Boolean)`
`HP.Omen.Core.Common.OmenHsaClient::SetMaxFanAsync(HP.Omen.Core.Common.PowerControl.Enum.MaxFanMode)`
`HP.Omen.Core.Common.OmenHsaClient::SetSwFanControlLevel(System.Int32,System.Int32)`

It's amazing how suboptimal all this is. Layers of abstraction of top of one another. And the WMI request is at some point actually encoded as JSON.

Error-106, 2024-02-11:
I do own a CH341A programmer, but taking my computer apart risks voiding the warranty.

I have set the voltage offset to -0.2 volts (before bios update allowed max -0.5 volts, after update only allowed max -0.2 volts)

OmenMon Hardware Monitoring & Control Utility Version 0.60-Release
BIOS (-Bios)
- Born-On Date: 20240102 [YYYYMMDD]
- System Design Data: 18 01 32 01 01 c8 01 0c 37 00 00 00 00 00 00 00 [↓]
- Status Flags: 0x0118 = 0b0000000100011000 = 280 [BiosPerfPpab3, Ppab4, Ppab8]
- Unknown Byte: 0x32 = 0b00110010 = 50 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x01 = 0b00000001 = 1 [V1]
- Support Flags: 0x01 = 0b00000001 = 1 [SwFanCtl]
- Default CPU Power Limit 4: 0xc8 = 0b11001000 = 200 [W]
- BIOS-Defined Overclocking: 0x01 = 0b00000001 = 1 [Supported]
- Graphics Mode Switching Support: 0x0c = 0b00001100 = 12 [Supported4, Supported8]
- Default CPU Concurrent Power Limit w/GPU: 0x37 = 0b00110111 = 55 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x01 = 0b00000001 = 1 [On]
- GPU Processing Performance AI Boost (PPAB): 0x01 = 0b00000001 = 1 [On]
- GPU Device Power State (DState): 0x01 = 0b00000001 = 1 [D1]
- GPU Peak Temperature Sensor Threshold: 0x57 = 0b01010111 = 87 [°C]
- Graphics Mode (Legacy): 0x00 = 0b00000000 = 0 [Hybrid]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
- Overclocking Support: 0x01 = 0b00000001 = 1
- Memory Overclocking Support: 0x00 = 0b00000000 = 0
- BIOS Undervolt Support: 0x01 = 0b00000001 = 1
- Keyboard Type: 0x00 = 0b00000000 = 0 [Standard]
- Keyboard Backlight Support: Yes
- Keyboard Backlight: 0x64 = 0b01100100 = 100 [Off]
- Keyboard Backlight Color Table: 4 Zones - Right: ffffff, Middle: 3cffff, Left: ffffff, Wasd: ffffff
- LED Animation Table:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
- Fan Count: 0x02 = 0b00000010 = 2
- Maximum Fan Speed: No
- Fan Type: 0x21 = 0b00100001 = 33 [↓]
- Fan #1 Type: 0x01 = 0b00000001 = 1 [Cpu]
- Fan #2 Type: 0x02 = 0b00000010 = 2 [Gpu]
- Fan #1 Level: 0x2a = 0b00101010 = 42 [Cpu]
- Fan #2 Level: 0x2d = 0b00101101 = 45 [Gpu]
- Fan Speed Level Table: 2 Fans, 41 Levels
# 00: Cpu 2000 rpm Gpu 2300 rpm @ 25 °C
# 01: Cpu 2100 rpm Gpu 2400 rpm @ 26 °C
# 02: Cpu 2200 rpm Gpu 2500 rpm @ 27 °C
# 03: Cpu 2300 rpm Gpu 2600 rpm @ 28 °C
# 04: Cpu 2400 rpm Gpu 2700 rpm @ 29 °C
# 05: Cpu 2500 rpm Gpu 2800 rpm @ 30 °C
# 06: Cpu 2600 rpm Gpu 2900 rpm @ 31 °C
# 07: Cpu 2700 rpm Gpu 3000 rpm @ 32 °C
# 08: Cpu 2800 rpm Gpu 3100 rpm @ 33 °C
# 09: Cpu 2900 rpm Gpu 3200 rpm @ 34 °C
# 10: Cpu 3000 rpm Gpu 3300 rpm @ 35 °C
# 11: Cpu 3100 rpm Gpu 3400 rpm @ 36 °C
# 12: Cpu 3200 rpm Gpu 3500 rpm @ 36 °C
# 13: Cpu 3300 rpm Gpu 3600 rpm @ 37 °C
# 14: Cpu 3400 rpm Gpu 3700 rpm @ 38 °C
# 15: Cpu 3500 rpm Gpu 3800 rpm @ 39 °C
# 16: Cpu 3600 rpm Gpu 3900 rpm @ 39 °C
# 17: Cpu 3700 rpm Gpu 4000 rpm @ 40 °C
# 18: Cpu 3800 rpm Gpu 4100 rpm @ 41 °C
# 19: Cpu 3900 rpm Gpu 4200 rpm @ 41 °C
# 20: Cpu 4000 rpm Gpu 4300 rpm @ 42 °C
# 21: Cpu 4100 rpm Gpu 4400 rpm @ 42 °C
# 22: Cpu 4200 rpm Gpu 4500 rpm @ 43 °C
# 23: Cpu 4300 rpm Gpu 4600 rpm @ 43 °C
# 24: Cpu 4400 rpm Gpu 4700 rpm @ 44 °C
# 25: Cpu 4500 rpm Gpu 4800 rpm @ 45 °C
# 26: Cpu 4600 rpm Gpu 4900 rpm @ 46 °C
# 27: Cpu 4700 rpm Gpu 5000 rpm @ 46 °C
# 28: Cpu 4800 rpm Gpu 5100 rpm @ 46 °C
# 29: Cpu 4900 rpm Gpu 5200 rpm @ 47 °C
# 30: Cpu 5000 rpm Gpu 5300 rpm @ 47 °C
# 31: Cpu 5100 rpm Gpu 5400 rpm @ 47 °C
# 32: Cpu 5200 rpm Gpu 5500 rpm @ 47 °C
# 33: Cpu 5300 rpm Gpu 5600 rpm @ 48 °C
# 34: Cpu 5400 rpm Gpu 5700 rpm @ 48 °C
# 35: Cpu 5500 rpm Gpu 5800 rpm @ 49 °C
# 36: Cpu 5600 rpm Gpu 5900 rpm @ 50 °C
# 37: Cpu 5700 rpm Gpu 6000 rpm @ 50 °C
# 38: Cpu 5800 rpm Gpu 6100 rpm @ 51 °C
# 39: Cpu 5900 rpm Gpu 6200 rpm @ 51 °C
# 40: Cpu 6000 rpm Gpu 6300 rpm @ 52 °C
- Temperature: 0x1b = 0b00011011 = 27 [°C]
- Thermal Throttling Status: 0x00 = 0b00000000 = 0 [Unknown]

Error-106, 2024-02-11:
-bios fanmax seems to work properly (when omen hub is running)

The fan control options in the GUI have no response except prog.

GeographicCone, 2024-02-11:
Good, so the WMI calls are working! That means the "secret" code most likely hasn't changed.

I'm now thinking, could it be that even though you close the _Omen Hub_, there is still a process running in the background that interferes with the settings you apply with **OmenMon**? (This would mean they're being reverted.)

I'm not really that familiar with how the _Omen Hub_ could be getting in the way, since I never install it in the first place, but @Rupurudu in #15 posted a guide how to disable it so that it doesn't interfere. Perhaps something similar is needed in your situation:

* [Getting rid of the Omen services](https://github.com/OmenMon/OmenMon/issues/15#issue-2010283530)

----

> Taking my computer apart risks voiding the warranty

Just FYI, as I'm definitely not telling you what to do, there were no seals or stickers anywhere on my model, and I've taken it apart quite a lot (repasting, replacing the Wi-Fi module "protected" by the heatsink, solving the ridiculous touchpad issue, etc.)

The flash chip is easily accessible as soon as you remove the bottom panel:

[image]

And HP themselves post instructional disassembly videos on their YouTube channel:

* YouTube: [Removing & replacing parts for OMEN by HP 16](https://www.youtube.com/watch?v=ZebpxQZP4JI)

So I figure they're probably chill with it. This actually a good thing about these laptops.

Error-106, 2024-02-11:
omen hub does run in the background and it automatically adjusts the PL1 limit of the CPU

I could try, but I don't have my tools and second computer around so it might be a while

Rupurudu, 2024-02-11:
Before you modify the BIOS, I would suggest removing the OMEN Hub and it's services, since it looks like OMEN Hub interferes with the fan profile you set up.

Error-106, 2024-02-11:
> Before you modify the BIOS, I would suggest removing the OMEN Hub and it's services, since it looks like OMEN Hub interferes with the fan profile you set up.

But my bios will identify in some way whether the omen hub is running, which prevents me from removing the omen hub

Error-106, 2024-02-12:
I may not be able to get a programmer these days, if I do I will export my bios and upload it here
If you need to use my device to find out how the system recognizes that omen hub is running, I'll be happy to provide remote access

GeographicCone, 2024-02-12:
I doubt it's a BIOS issue. There's nothing in the BIOS that would affect fan control.

You could maybe unlock some, mostly CPU-related controls by changing the hidden BIOS settings. For that, you need to read the Setup IFR (Internal Forms Representation) dump to know where these settings are stored. But it doesn't have to be a BIOS dump specifically from your machine, just a clean image downloaded from the HP Support website, as long as you make sure it's the same model and version.

I said you could use a hardware programmer to (1) go back to an earlier BIOS version, and (2) to make a BIOS backup so that you have something to restore if there's any problem. Your BIOS dump will have some information specific to your machine, and you shouldn't post it publicly. Just keep it for yourself, in case you ever need it. A BIOS dump as such is not required to change the settings by editing UEFI variables. Only make sure you know how to restore the default settings if the computer doesn't POST: the key combination for that is to hold <kbd>Win</kbd>-<kbd>V</kbd> before you power on. That's one thing good to remember before you start messing with the BIOS settings.

Anyway, all that is something else altogether. A step-by-step guide is at: https://github.com/GeographicCone/UefiVarTool

----

Now back to fan control. It's likely an _Omen Gaming Hub_ issue. To get rid of it, see @Rupurudu's guide in #15.

If you don't want to uninstall OGH just yet, then maybe as a simple test, you can boot into _Safe Mode_ and try fan control in **OmenMon** from there? Chances are (although I haven't tried it) the _Omen Gaming Hub_ drivers will not loaded but **OmenMon** should still run normally. If fan control works in _Safe Mode_, it's highely likely it's some OGH component preventing it from working normally when it's running in the background after a normal startup.

Alternatively, maybe you could try installing a fresh copy of Windows on another SSD, without installing OGH, and then checking if fan control works in it. I don't know what to tell you but I think you're 99% there. You posted the result of `OmenMon -Bios` and it shows the WMI calls work normally. There's no reason why something like `OmenMon -Bios FanMode=Performance FanLevel=50,50` shouldn't work either.

Error-106, 2024-02-12:
About fan control
In safe mode, -bios fanmode=49 fanlevel=50,50 has no effect(Locked to 55 watt and automatic)
My hardware should have additional checks for omen hub

Error-106, 2024-02-12:
omenmon's GUI doesn't seem to work properly in safe mode
[image]

Error-106, 2024-02-12:
> About fan control
> In safe mode, -bios fanmode=49 fanlevel=50,50 has no effect(Locked to 55 watt and automatic)
> My hardware should have additional checks for omen hub

My previous discussion probably shouldn't be called a BIOS, maybe I should call it an onboard subsystem

Error-106, 2024-02-12:
omenmon's interface does work, but my hardware won't use the interface's data without recognizing omenhub in some unknown way.

Error-106, 2024-02-12:
I may not read and modify the bios recently because the Chinese New Year make me busy

Error-106, 2024-02-13:
I have disassembled my laptop and found the same chip
[image]
But I failed when I tried to read the data. How did you read the bios?

GeographicCone, 2024-02-13:
> How did you read the bios?

I used a CH341A with a Pomona 5250 SOIC-8 clip and `flashrom --progress -r out.bin` (compiled under [MSys2](https://www.msys2.org/): I actually posted [build instructions](https://github.com/flashrom/flashrom/issues/242#issuecomment-1335918958) as an issue comment a while ago).

There are many things to pay attention to: unplug the power and battery, make sure the clip is well-seated and the pins match:

[image]

Pin 1 is the one marked with the dot on the chip.

Also connect to a USB port directly on the computer, don't use any hubs or extension cables.

When done make sure the result is not corrupt. I let it run a couple of times and compare with `fc /b` but before that also make sure the file is not full of `0xFF`s as that can happen if the connection is completely off.

Error-106, 2024-02-13:
I read out 3 times using different tools, but some pages are always missing, maybe the voltage is too high (the official specification states that the voltage range is 2.7v-3.6v)

GeographicCone, 2024-02-13:
Some versions of CH341A only supply 5V, which might work but probably isn't a good idea to use.

What's more likely though is that the chip is too new for those other tools you've used. Flashrom only [added support for the W25Q128JV](https://github.com/flashrom/flashrom/issues/220) recently.

You could try reading some older chip to narrow down the cause.

Error-106, 2024-02-14:
I spent some time successfully compiling it, but the program failed to read and prompted cb_in: error: LIBUSB-TRANSFER-TIMEd_OUT
[image]
我或许可以在Linux上重新尝试

GeographicCone, 2024-02-14:
Do you have the right driver? (Just an `.inf` file pointing to `WinUSB.inf` really.)

But if it found the chip somehow, then you probably do.

Maybe there is some regression. I'm attaching the version that works for me (and the "driver").

* [CH341A WinUSB Driver.zip](https://github.com/OmenMon/OmenMon/files/14277638/CH341A.WinUSB.Driver.zip)
* [FlashROM 1.4.0-devel (gitv1.2-1345-gd534100c).zip](https://github.com/OmenMon/OmenMon/files/14277640/FlashROM.1.4.0-devel.gitv1.2-1345-gd534100c.zip)

Error-106, 2024-02-14:
I successfully copied the binary file with flashrom under linux, but unfortunately, the copied binary file is still incomplete, I will repurchase a programmer with 3.3 volts

Error-106, 2024-02-14:
I bought a new CH341A programmer that supports 1.8/3.3/5 volts, and unfortunately, because it is Chinese New Year, the courier company is on vacation
I have no progress in my research on -ecmon, and I can't find any rules for the data of the remaining registers
Maybe this requires a reverse compiler, I have no experience, I can try research, it may take some time,Can you provide some assistance?

GeographicCone, 2024-02-15:
> Can you provide some assistance?

Like I said in [my comment](https://github.com/OmenMon/OmenMon/issues/37#issuecomment-1937347757) above: if it works with the _Omen Gaming Hub_, then the answer must be there. And it'll likely be in the form of WMI calls. Writing to the EC is not the official way of doing things. So look for the former, not the latter.

* Go to: https://store.rg-adguard.net/

* Paste the app link to the _Omen Gaming Hub_ from the `MICROS~1` store:
https://apps.microsoft.com/store/detail/omen-gaming-hub/9NQDW009T0T5, click the only button or press `Enter`.

* Download `AD2F1837.OMENCommandCenter_xxxx.xxxx.4.0_x64__v10z8vjag6ke6.msix` (the `xxxx` part is version number).

  * Note: Even though it's a third-party website, the actual download is from `*.dl.delivery.mp.microsoft.com`, so it's safe to run it. It's just that these links keep changing and become invalid every couple of hours, which is why the extra step is needed.

* Change the `.msix` extension to `.zip` and extract the contents, for example with [7-Zip](https://7-zip.org/download.html).

* You now have the _Omen Gaming Hub_ code to work with.

  * If you have it installed, you can skip the above steps and just find the `.msix` file somewhere under `%ProgramFiles%\WindowsApps` but the instructions cover how to get it even if it's not there.

Reverse engineering is a broad topic and not always easy. Fortunately for us, the _Omen Gaming Hub_ is mostly written in C# and unobfuscated. So it's relatively straightforward:

* Download either [dnSpyEx](https://github.com/dnSpyEx/dnSpy) or [JetBrains DotPeek](https://www.jetbrains.com/decompiler/download/download-thanks.html?platform=windowsWeb) (both are free, and the first is open-source)

* Open the main entry point is `OmenCommandCenterApp\HP.Omen.OmenCommandCenter.exe` and start exploring!

Broadly, there are 3 ways to go about it:

Option 1: Search the code for keywords and read it.

  * You can use my [OmenHwCtl](https://github.com/GeographicCone/OmenHwCtl) PowerShell script to run arbitrary WMI commands and see what they do.

  * The _Omen Gaming Hub_ depends on some files it downloads from the Internet for the configuration, and you can find a dump of them here: https://github.com/GeographicCone/OmenHwCtl/Reference alongside a batch script that downloads and saves them all in a readable form. 

Option 2: Step through the code in a debugger.

Option 3: Modify the code to save debug logging information (it's already being produced but discarded in their release version).

That's the gist of how to get to the missing part, as I see it.

Error-106, 2024-02-15:
Thank you for your help, I may have to take a little time to get familiar with C#

GeographicCone, 2024-02-15:
Just look at these methods I mentioned:

> `NativeRpcClient.dll`
> ↑
> `Hp.Bridge.Client.NativeMethods::RpcExecuteCommand()`
> ↑
> `Hp.Bridge.Client.RpcClient::ExecuteCommand()`
> ↑ [...]

The last 3 are used for fan control. Each runs a WMI command with certain parameters via `BiosWmiCmd_Set`. This is what **OmenMon** does as well:
https://github.com/OmenMon/OmenMon/blob/665f767b796401c21f8cc4c7d1615ed5e922dc6b/Hardware/BiosCtl.cs#L336-L341
https://github.com/OmenMon/OmenMon/blob/665f767b796401c21f8cc4c7d1615ed5e922dc6b/Hardware/BiosCtl.cs#L366-L369
https://github.com/OmenMon/OmenMon/blob/665f767b796401c21f8cc4c7d1615ed5e922dc6b/Hardware/BiosCtl.cs#L330-L334

I don't know what they changed that it doesn't work on your newer 2023 model but the answer will be somewhere around those calls and when you come up with a guess, you can check it immediately since you have the hardware.

Error-106, 2024-02-15:
Fan control is normal under some conditions. The behavior of my device is like HP has added a piece of code to detect the omen hub. When I completely turn off the omen hub, the fan speed and power can be modified until dozens of seconds later
We have absolutely no idea how this is accomplished, and the code to implement this functionality could be anywhere

Error-106, 2024-02-16:
> * The _Omen Gaming Hub_ depends on some files it downloads from the Internet for the configuration, and you can find a dump of them here: https://github.com/GeographicCone/OmenHwCtl/Reference alongside a batch script that downloads and saves them all in a readable form.

This link seems to have expired

Error-106, 2024-02-16:
It seems that the detected method is not in HP.Omen.OmenCommandCenter (because OmenCommandCenter is not running when running in the background)
I'll try some stupid way to try to find where(Kill them one by one and wait.)

Error-106, 2024-02-16:
Oh, I think I found it. It seems that it is in OmenCommandCenterBackground (the fan is locked dozens of seconds after ending this program, and this program also calls WMISDK) or HP.OMEN.VadRedirectHelper (the program will be started again when OmenCommandCenterBackground survives)

Error-106, 2024-02-16:
OmenCommandCenterBackground seems to exist as .resources.dll in the installation package
(OmenCommandCenterApp/en-us/OmenCommandCenterBackground.resources.dll)
But in the directory after installation it becomes .exe

Error-106, 2024-02-25:
I can't find the entry point, HP seems to put all the tasks that run in the background here, I need help

The CH341A programmer I bought is already in my hands. I will try to extract it again in a few days.

julienrobin28, 2024-12-16:
Something new here, with HP 16-s1034nf (board code 8C9C, Ryzen 5 8645HS, BIOS F.11, "HP Victus Laptop" Manufactured 2024/09)

**Workaround for not using Omen Gaming Hub**

When running `OmenMon.exe -Bios`, then during the following 120 seconds, the `OmenMon.exe -Bios FanMax=true` command works again. Manual Fan speeds are working too. It enables some "special mode" into the computer.

More precisely: the `OmenMon.exe -Bios FanCount` commands alone allows for 120 seconds of definable settings instead of factory default. The command should be played again to avoid the 120 second timeout to occur (when it does, FanMax is disabled). 

`OmenMon.exe -Bios FanMax=false` however has no effect (but after 120 seconds without FanCount command, it goes off anyway). 

**About thermal profiles:**

Also, using `RyzenAdj.exe --info` or `AMDRyzenMasterCLI.exe --api GetAllInfo`,  we can see that during this "120 seconds special mode", the CPU cTDP values aren't the same (from 45W / 45W / 45W  by default for STAPM / PPT Fast / PPT Slow, it goes to 65 / 65 / 55 or even 65 / 65 / 60 if "Performance" mode was the last mode enabled).

`OmenMon.exe -Bios FanMode=1` sets the Performance mode. I'm going to test the others

From `OmenMon-master/Hardware/BiosData.cs` :
```
        // Fan performance mode
        // Source: HP.Omen.Core.Common.PowerControl.PerformanceMode
        public enum FanMode : byte {
            LegacyDefault     =  0,  // 0x00 = 0b00000000
            LegacyPerformance =  1,  // 0x01 = 0b00000001
            LegacyCool        =  2,  // 0x02 = 0b00000010
            LegacyQuiet       =  3,  // 0x03 = 0b00000011
            LegacyExtreme     =  4,  // 0x04 = 0b00000100
            L8                =  4,  // 0x04 = 0b00000100
            L0                = 16,  // 0x10 = 0b00010000
            L5                = 17,  // 0x11 = 0b00010001
            L1                = 32,  // 0x20 = 0b00100000
            L6                = 33,  // 0x21 = 0b00100001
            Default           = 48,  // 0x30 = 0b00110000
            L2                = 48,  // 0x30 = 0b00110000
            Performance       = 49,  // 0x31 = 0b00110001 
            L7                = 49,  // 0x31 = 0b00110001 
            L3                = 64,  // 0x40 = 0b01000000
            Cool              = 80,  // 0x50 = 0b01010000
            L4                = 80   // 0x50 = 0b01010000
        }
```
I'm also taking a look into linux `hp-wmi.c` which has a lot in common with OmenMon software (as it's retro engineered from HP tools too); it has to be reworked to work with my laptop.

For now, didn't find the difference between Omen Gaming Hub's ECO and Balanced mode (same cTDP both on AC and Battery).
ECO mode doesn't allow setting the Fan speed, so it's possibly a "Quiet" mode rather than an "Eco" mode. 

Hope I hope this may help someone, 
Any way thanks both for the work and for the open source stuff!

julienrobin28, 2024-12-17:
Discovered tools like [dnSpyEx](https://github.com/dnSpyEx/dnSpy) and [dotPeek](https://www.jetbrains.com/decompiler/) and it allowed me to do the retro-engineering for my current laptop HP Victus 16s-1034nf

Here it is, attached as a file, I hope this may help anyone doing compatibility work!

[HP-WMI-retro-engineering-2024-12-17.txt](https://github.com/user-attachments/files/18159931/HP-WMI-retro-engineering-2024-12-17.txt)

Adding the automatic FanCount query as a "heartbeat" into OmenMon should be easy, as there is already a timer used to do others periodic queries. However I don't know if it still runs while the application is minimized as tray icon.

For CLI, automatic FanCount query call may be added, but after 120 seconds with no commands, laptop is still going to stop using our custom settings if OmenMon GUI isn't running as tray icon / on background.

Many thanks again for the work and the discovery of dnSpy and dotPeek anyway.

For Linux 6.12.4 at least this allowed me to quickly craft a dirty custom `hp-wmi.c` file that is working  with the following commands:

```
# Disable automatic fan speed: setting max speed now works!
echo 0 > /sys/devices/platform/hp-wmi/hwmon/hwmon6/pwm1_enable

# Now displaying correct fan speed values! And with the dirty FanCount queries additions to the driver, this also serve as a keepalive against 120 seconds timeout
watch -n 1 sensors

# Re-enable automatic fan speed
echo 2 > /sys/devices/platform/hp-wmi/hwmon/hwmon6/pwm1_enable
```
Having added the FanCount queries also enables cTDP values to go from 45/45/45 to 65/65/55 in my case (however, platform_profiles related stuff are completely outdated into this driver, and should be reworked). Also, the quick dirty changes I've made may break compatibility with older laptops, so this is just a test/draft which is useful for me, or for people to understand.

Anyway, for those interested by what are the changes I made, using Meld for example you can compare these

https://pix-server-sorel.luoss.fr/Manual/Linux/HP-WMI/Bug-Report-2024-12-12/hp-wmi-original-6.12.4.c
https://pix-server-sorel.luoss.fr/Manual/Linux/HP-WMI/Bug-Report-2024-12-12/hp-wmi-FAN-SET+GET-success.c

Good luck to everyone!