OmenMon/OmenMon #38: 欢迎光临！ Добро пожаловать! Dear International Users!

GeographicCone, 2024-02-09:
### 大家好
我最近发现 OmenMon 有蛮多中文为母语的用户，谢谢你们对 OmenMon 的兴趣和支持，我想让大家知道 OmenMon 的界面其实是可以翻译的，有兴趣的话请至 [Localization](https://github.com/OmenMon/Localization) 的存储库看详情。

创建议题时麻烦大家用英文来沟通，以便帮助来自世界各地的用户全体合作，但若真的无法的话，请在最前面加上一个英文句子可以当作问题所在焦点的摘要。

如果有任何关于 OmenMon 多语言化的意见可以使用此次议题留言（任重语言皆可），**谢谢！**

----

### Добро пожаловать!

Я только что заметил среди пользователей **OmenMon** есть довольно много русскоговорящих. Спасибо Вам всем за заинтересованность и поддержку проекта! Хотел бы чтобы все узнали в OmenMon встроена возможность перевода. Пройдите пожалуйста в репозиторий [Localization](https://github.com/OmenMon/Localization) чтобы узнать подробности.

Оставляя комментарии, пишите пожалуйста на английском, чтобы облегчить сотрудничество с пользователями из других регионов мира, но если действительно нельзя, добавьте пожалуйста сначала одно предложение на английском, которое объясняет суть оставляемого Вами комментария в сокращенном порядке.

Если по поводу русификации OmenMon (или перевода на другие языки) есть какие-нибудь вопросы, не стесняйтесь пользоваться этой веткой, чтобы высказать своё мнение (на любом языке). **Спасибо!**

----

### Dear **OmenMon** users throughout the world!

I really appreciate your interest in the project. I wanted to let you know **OmenMon** can be translated: see the [Localization](https://github.com/OmenMon/Localization) repository for details.

Please try to use English to communicate here if you can, so as to facilitate co-operation with users from other parts of the world. If writing in another language, please include a short English summary at the top.

Please feel free to let me know your feedback below (in any language). **Thank you!**

Initsnow, 2024-05-05:
I wrote a simple program to translate😎
[xmltranslator](https://github.com/Initsnow/xmltranslator)

GeographicCone, 2024-05-06:
@Initsnow Thank you for contributing your translation, and the tool! I've accepted your PR, and added a link to `README.md` in each respective repository.