OmenMon/OmenMon #39: OMEN X 17-ap0xx

OrtoDom, 2024-02-10:
Hi

As in title OMEN X 17-ap0xx - mobo 83A5 - BIOS F.28

`OmenMon.Hardware.Bios.BiosException: BIOS call failed: Command not available
   w OmenMon.Hardware.Bios.Bios.Check(Int32 code, Boolean force)
   w OmenMon.Hardware.Bios.BiosCtl.GetMaxFan()
   w OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   w OmenMon.AppGui.GuiFormMain.UpdateFanCtl()
   w OmenMon.AppGui.GuiFormMain.UpdateAll()
   w OmenMon.AppGui.GuiFormMain..ctor()
   w OmenMon.AppGui.GuiTray.ShowFormMain()
   w OmenMon.AppGui.GuiTray.EventIconMouseClick(Object sender, MouseEventArgs e)
   w System.Windows.Forms.NotifyIcon.OnMouseClick(MouseEventArgs mea)
   w System.Windows.Forms.NotifyIcon.WmMouseUp(Message& m, MouseButtons button)
   w System.Windows.Forms.NotifyIcon.WndProc(Message& msg)
   w System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

If i change BiosErrorReporting to false, program will run, but. Sensors are reading wrong values, temp changing form 1'C to 95'C
RPM readout is also showing wrong values. Changing anything has no effect, likewise cli commands.

Readout from -Bios command

`OmenMon Hardware Monitoring & Control Utility Version 0.61-Release
BIOS (-Bios)
- Born-On Date: 20080307 [YYYYMMDD]
- System Design Data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [↓]
- Status Flags: 0x0000 = 0b0000000000000000 = 0 [0]
- Unknown Byte: 0x00 = 0b00000000 = 0 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x00 = 0b00000000 = 0 [V0]
- Support Flags: 0x00 = 0b00000000 = 0 [0]
- Default CPU Power Limit 4: 0x00 = 0b00000000 = 0 [W]
- BIOS-Defined Overclocking: 0x00 = 0b00000000 = 0 [NotSupported]
- Graphics Mode Switching Support: 0x00 = 0b00000000 = 0 [0]
- Default CPU Concurrent Power Limit w/GPU: 0x00 = 0b00000000 = 0 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x00 = 0b00000000 = 0 [Off]
- GPU Processing Performance AI Boost (PPAB): 0x00 = 0b00000000 = 0 [Off]
- GPU Device Power State (DState): 0x00 = 0b00000000 = 0
- GPU Peak Temperature Sensor Threshold: 0x00 = 0b00000000 = 0 [°C]
- Graphics Mode (Legacy): 0x00 = 0b00000000 = 0 [Hybrid]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
- Overclocking Support: 0x00 = 0b00000000 = 0
- Memory Overclocking Support: 0x00 = 0b00000000 = 0
- BIOS Undervolt Support: 0x00 = 0b00000000 = 0
- Keyboard Type: 0x00 = 0b00000000 = 0 [Standard]
- Keyboard Backlight Support: Yes
- Keyboard Backlight: 0xe4 = 0b11100100 = 228 [On]
- Keyboard Backlight Color Table: 4 Zones - Right: 000000, Middle: 000000, Left: 000000, Wasd: 000000
- LED Animation Table:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
- Fan Count: 0x00 = 0b00000000 = 0
- Maximum Fan Speed: No
- Fan Type: 0x00 = 0b00000000 = 0 [↓]
- Fan #1 Type: 0x00 = 0b00000000 = 0 [Unsupported]
- Fan #2 Type: 0x00 = 0b00000000 = 0 [Unsupported]
- Fan #1 Level: 0x00 = 0b00000000 = 0 [Cpu]
- Fan #2 Level: 0x00 = 0b00000000 = 0 [Gpu]
- Fan Speed Level Table: 0 Fans, 0 Levels
- Temperature: 0x00 = 0b00000000 = 0 [°C]
- Thermal Throttling Status: 0x00 = 0b00000000 = 0 [Unknown]`

GeographicCone, 2024-02-10:
Thank you for your interest in **OmenMon** and the report!

I'm sorry but it seems there is zero support for any of the WMI BIOS calls that **OmenMon** depends on in the GUI mode. Since you say the EC readings are also incorrect, that would indicate the register map is different as well than the one I researched on my model.

While support for using arbitrary EC registers to control fans is envisioned in **OmenMon**:

https://github.com/OmenMon/OmenMon/blob/db4b7f9360dd57ee2713da85244293243b86a79b/Hardware/Platform.cs#L48-L115

It'd take a lot of work and, before that could even begin, a lot of research, to figure all this out starting from scratch.

You could still use **OmenMon**'s `-Ec` mode to read from and write to arbitrary registers but that's about it. To figure out which registers control what you could look at the ACPI DSDT table (which can be read with tools such as [RWEverything](http://rweverything.com/)).

Another way would be to look at a _HP_-provided utility that has this functionality if there is one. For the record, the _Omen Gaming Hub_ [model configuration file](https://github.com/GeographicCone/OmenHwCtl/blob/b3eda0b93269fedf329b9d55409d7ff020e956ac/Reference/Bundled/HP.Omen.Core.Model.Device.JSON.DeviceList.json#L14-L15) does not list `PerformanceControl` among the features for your model:

````json
  {
    "Name": "Dragons10",
    "ChipsetVendor": 1,
    "DisplayName": "OMEN X 17 LAPTOP",
    "ProductNum": [
      {
        "SSID": "83A5"
      },
      {
        "SSID": "83A6"
      }
    ],
    "Feature": [ "SystemInfo", "Overclocking", "DragonLighting", "NetworkBooster", "Macros" ],
    "BackgroundFeature": [ "NetworkBooster", "DragonKBMcu", "DragonChassisLighting", "Overclocking" ]
  },
````

I appreciate your posting the report. Having it here is very helpful in assessing compatibility.

I think there have been several reports of **OmenMon** generally not being useful on `85xx` and `83xx` series models, which I believe translates to the year 2019 or earlier. Perhaps it should be stated somewhere as a known limitation that **OmenMon** is unlikely to work on pre-2019 devices.

Sorry that's probably not the answer you were hoping to hear.

OrtoDom, 2024-02-15:
I don't know it'll help develop app to support more legacy Omen's. But i found something after installing "HPSupportAssistantSP54620" 
In dir. "C:\Program Files (x86)\Hewlett-Packard\HP Support Framework\Resources\HPThermalAssistant" is a HPThermalAssistant.exe (HP CoolSense [ver.1.0.1.4]) App with 3 options. Coolest Mode / Performance Optimized Mode (default) / Quietest Mode 

On HP site is also "sp60051" it installs in dir. "C:\Program Files (x86)\Hewlett-Packard\HP CoolSense" a CoolSense.exe (v2.1)
But its only have two options, on/off. 

With my little knowledge, I tried reverse engineering, but I think you can guess the effects.