OmenMon/OmenMon #40: Trojan Detection After Enabling "Advanced Optimus Fix"

IssamBouchahine, 2024-02-11:
I scanned OmenMon before using it and it was ok. Everything seems to be working just fine, until I toggle the "Advanced Optimus Fix" button, in which Kaspersky just completely terminates the process and deletes OmenMon. I know I can exclude it, but I believe fixing this issue might be better.

Thanks.

GeographicCone, 2024-02-11:
_Oh, snap! I guess I'll have to remove that Trojan then. I thought it'd go unnoticed, hidden under the option almost nobody uses, but now that Kaspersky caught up with my shady practices, seems I have no choice…_

Seriously though:

> I believe fixing this issue might be better.

I concur. It would be. But how can I fix it?

There is no Trojan, virus or any other malware in **OmenMon** of course. This has been addressed in detail in issue #28: 

* [Is there a virus in OmenMon? (Spoiler alert)](https://github.com/OmenMon/OmenMon/issues/28)

Everything **OmenMon** does is for a reason. There's nothing to change without removing functionality. And while I'm not sure anyone else in the world uses the _Advanced Optimus Fix_ (**OmenMon** does not collect any usage data or telemetry), I _do_ use it myself to workaround screen stutter, and that functionality is in fact one of the few original reasons I wrote **OmenMon** and the earlier [OmenHwCtl](https://github.com/GeographicCone/OmenHwCtl). So that option is not going anywhere, not that it would change anything: **OmenMon** does a lot of other stuff that can be considered "suspicious," pretty much always because that's the only way it can be done.

"Anti-malware" solutions these days largely fail to protect from actual malware, as long as it's sophisticated enough, or new enough. To compensate for that, they apply "AI"-based criteria and produce spurious positives to manufacture the illusion that they're offering actual protection from anything serious, which is usually not the case.

These days such software mostly exists for compliance reasons. And as there is no cost associated with generating falsely positive reports (to the authors of the said software: there are externalities that affect others, as can clearly be seen), and such pretend-protection might in fact encourage some customers to renew their subscriptions, no wonder things are the way they are. So, when evaluating any malware reports, I implore everyone to resort to their better judgment. One other anti-virus program that recently flagged **OmenMon** as "malware" on VirusTotal has also flagged numerous files by Microsoft that are part of the Windows operating system just the same, and with "100% confidence" (although ironically here, they might be onto something). But in case of **OmenMon**, at the end of the day, this is _open source_ software, for Pete's sake!

Of course, if you have any specific idea what more can be done to address this issue, I'd love to hear it. I've taken some steps already, as outlined in #28, but will have to stop short of anything that would require committing large amounts of money on an annual basis (such as for purchasing an EV certificate) for what is largely a niche hobby project.

Are you a paying customer of Kaspersky? Perhaps you could report this as a false positive to them, which it is.

Let's continue any further discussion in issue #28 so as to keep everything in one place.

GeographicCone, 2024-02-13:
Closing so as to keep all discussion related to malware reports in #28 .

closed 2024-02-13 by GeographicCone