OmenMon/OmenMon #41: BIOS call failed: Command not available

SamukaGamer, 2024-02-12:
Issue happens when launching GUI, setting setting to false bypasses the problem

OMEN by HP 15-dc1014np (https://support.hp.com/pt-pt/product/details/omen-by-hp-15-dc1000-laptop-pc-series/model/29171025?sku=6py04ea&serialnumber=5cd929079r)

Product
6PY04EA
Series
5CD929079R

Error without translation (its in portuguese)
----------------------------------------------------------------------------------------------------------------------------------------------------
Consulte o fim desta mensagem para obter detalhes sobre como invocar a depuração 
JIT (just-in-time) em vez desta caixa de diálogo.

************** Texto da excepção **************
OmenMon.Hardware.Bios.BiosException: BIOS call failed: Command not available
   em OmenMon.Hardware.Bios.Bios.Check(Int32 code, Boolean force)
   em OmenMon.Hardware.Bios.BiosCtl.GetMaxFan()
   em OmenMon.Library.Hw.<>c__DisplayClass13_0`1.<BiosGet>b__0(IBiosCtl bios)
   em OmenMon.AppGui.GuiFormMain.UpdateFanCtl()
   em OmenMon.AppGui.GuiFormMain.UpdateAll()
   em OmenMon.AppGui.GuiFormMain..ctor()
   em OmenMon.AppGui.GuiTray.ShowFormMain()
   em System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   em System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   em System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   em System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   em System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   em System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   em System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   em System.Windows.Forms.Control.WndProc(Message& m)
   em System.Windows.Forms.ToolStrip.WndProc(Message& m)
   em System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   em System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Assemblagens carregadas **************
mscorlib
    Versão da assemblagem: 4.0.0.0
    Versão Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
OmenMon
    Versão da assemblagem: 0.61.0.0
    Versão Win32: 0.61.0.5020
    CodeBase: file:///C:/Users/Samuel/Downloads/OmenMon-0.61.0-Release/OmenMon.exe
----------------------------------------
System.Drawing
    Versão da assemblagem: 4.0.0.0
    Versão Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Versão da assemblagem: 4.0.0.0
    Versão Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Versão da assemblagem: 4.0.0.0
    Versão Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Xml
    Versão da assemblagem: 4.0.0.0
    Versão Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
mscorlib.resources
    Versão da assemblagem: 4.0.0.0
    Versão Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_pt_b77a5c561934e089/mscorlib.resources.dll
[... 147 lines cut ...]
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_pt_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than being handled by this dialog box.
----------------------------------------------------------------------------------------------------------------------------------------------------
[image]

If you need something I will help gladly, just tell me what to do.
Thanks.

GeographicCone, 2024-02-13:
The relevant part is here:

> OmenMon.Hardware.Bios.BiosException: BIOS call failed: Command not available
> OmenMon.Hardware.Bios.Bios.Check(Int32 code, Boolean force)
> OmenMon.Hardware.Bios.BiosCtl.GetMaxFan()

The WMI call to get the maximum fan status is not supported by the hardware.

1. You might still be able to use maximum fan speed mode by writing to the EC:

* To enable: `OmenMon -Ec FFFF=0x0C`
* To disable: `OmenMon -Ec FFFF=0x00`
* Maybe also need to set a custom fan mode first: `OmenMon -Ec HPCM=0x31`

Does the maximum fan speed mode work for you if you try to enable it like that?

2. You might still be able to use the GUI by setting: [BiosErrorReporting](https://omenmon.github.io/config#bioserrorreporting) to **false** in `OmenMon.xml`

Other than the keyboard colors, does any of the other GUI functionality work for you?

What if you enable [FanLevelUseEc](https://omenmon.github.io/config#fanleveluseec), and maybe also [FanLevelUseManual](https://omenmon.github.io/config#fanlevelneedmanual)?

SamukaGamer, 2024-02-13:
All fan profiles either dont do anything or are set back to default after a few seconds
CPU temperature shows 1c
And keyboard rgb works good in the GUI, but thats it, no gpu boost or cpu tdp functions or anything, just the rgb works.
Im going to try this

> You might still be able to use maximum fan speed mode by writing to the EC:
To enable: OmenMon -Ec FFFF=0x0C
To disable: OmenMon -Ec FFFF=0x00
Maybe also need to set a custom fan mode first: OmenMon -Ec HPCM=0x31
Does the maximum fan speed mode work for you if you try to enable it like that?

SamukaGamer, 2024-02-13:
Tried using all combinations of this options from false to true

     
        <FanLevelNeedManual>true</FanLevelNeedManual>

        
        <FanLevelUseEc>true</FanLevelUseEc>

No change, fans still does not work.

[image]

Only the RGB works, the rest either isnt available to change or just dosent apply any changes.

[image]

This also does not make any changes to fan speed

[image]

Doing like this does not change the fans speed as well

Clicking on the taskbar icon the graphics tab does not apply any changes, only the monitor related options work, fans dont work, keyboard works all, and the rest I dont use/isnt supported by my laptop.

SamukaGamer, 2024-02-13:
Update, after I did all this I went over to gaming hub and trid switching the power modes from cool to default and the fans are now stuck at max speed, im going to try omenmon with gaming hub uninstalled to see if it was gaming hub making the problem

SamukaGamer, 2024-02-13:
Update 2 fans are now stuck at 100% and cant change it via any software, selecting the off option in omenmon turn them off for 2 seconds before they ramp up again to 100%

SamukaGamer, 2024-02-13:
update 3: fixed the issue, gaming hub, omen mon and notebook fan control were unable to change fan speeds after this "bug?" the solution I did was reflashing the BIOS with an spi programmer. everything works as normal now, something got perma written on the bios somehow.

GeographicCone, 2024-05-07:
Closing as resolved, please feel free to reopen if need be.

closed 2024-05-07 by GeographicCone