OmenMon/OmenMon #42: Doesnt seem to be wroking with my HP OMEN Transcend 16

Mentrauz, 2024-02-18:
Nothing works, no kb lighting or fan control

GeographicCone, 2024-05-07:
It might be not compatible. You could still try the individual commands in CLI mode:
https://omenmon.github.io/cli#bios

I take it's a new (2023) model? Which one? I'll update the topic to help others researching compatibility in the future.

Here's how to find out the model:
https://github.com/OmenMon/OmenMon/issues/1#user-content-model

GeographicCone, 2024-05-07:
Possibly related to #37 (some 2023 models might need extra steps) and keyboard backlight might be due to #3 (per-key RGB not supported).

Mentrauz, 2024-06-12:
> It might be not compatible. You could still try the individual commands in CLI mode: https://omenmon.github.io/cli#bios
> 
> I take it's a new (2023) model? Which one? I'll update the topic to help others researching compatibility in the future.
> 
> Here's how to find out the model: [#1 (comment)](https://github.com/OmenMon/OmenMon/issues/1#user-content-model)

[image]
[image]

Yes its 2023 model (i7 13700HX and RTX 4070 130w)
Yes it has per key rgb so maybe thats why lighting is not supported.

Mentrauz, 2024-06-12:
> It might be not compatible. You could still try the individual commands in CLI mode: https://omenmon.github.io/cli#bios
>
> I take it's a new (2023) model? Which one? I'll update the topic to help [...]

And im new to this so i would rather use omen gaming hub than mess up laptop with custom commands i dont know.

Mentrauz, 2024-06-12:
And this laptop supports 240Hz but this seems to be only supporting 165Hz at the moment.