OmenMon/OmenMon #66: [8C9C] HP Victus 16-s1023dx 2024

firefoot007, 2024-07-15:
* OmenMon Version: **0.61-Release**
* Model & Version: **HP 8C9C 79.23**  (Victus 16-s1023dx 2024)
* Working:
  * CPU/ GPU sliders seem to work. 
  * Fan Off, Max, and Const seem to work. I haven't tried the Fan program yet.
* Not Working:
  * Wrong EC temperature for CPU
  * GPU temp little off
  * Wrong EC CPU/GPU RPM and % for fan speeds
  * The backlight setting is greyed out and showing as unsupported but you can use -Bios Backlight 1 to turn on the backlight (It is only a white color backlight with two brightness levels)

I think in my particular version of Victus, RX2C looks to be the right temperature sensor. I am not sure about it but I used EceMon to find the changing register with almost right values.

I am attaching all the screenshots below to help you. 

[image]
[image]
[image]
[image]
[image]

I have read your guide to compiling the program myself but I am not good with C#.  Can you please look into this issue?

And finally thank you for your program. I just bought the laptop two days ago and I am surprised that HP let this Omen Gaming Hub bloatware pass as a control program. I was using G14 with G-Helper before so I am really glad that somebody is working in a similar direction. Please let me know if you need more information.

maxius4, 2024-08-29:
I have the same laptop and id love it if this fully worked on it

Sharwesh05, 2024-11-30:
@firefoot007 how did you find the registers??

firefoot007, 2024-11-30:
> @firefoot007 how did you find the registers??

I am not using it now. But I used the tool given with OmenMon and some other EC scanning tool to find the registers which varied according to the temp variation. But its not very easy.

Sharwesh05, 2024-11-30:
@firefoot007 would you try recollecting the tool name because I am trying to find the registers of my board. It don't care about the complexity