OmenMon/OmenMon #68: Performance Control require some heartbeat to work properly on some new devices

KawaiiNahida, 2024-08-01:
OmenMon Version: 0.61

KawaiiNahida, 2024-08-01:
The heartbeat part in the official app
```csharp
int num = ((this.PowerContextSettings != null) ? this.PowerContextSettings.IntervalHeartbeat : 30000);
	DateTime dateTime = DateTime.Now;
	for (;;)
	{
		Thread.Sleep(100);
		if ((DateTime.Now - dateTime).TotalMilliseconds >= (double)num)
		{
			dateTime = DateTime.Now;
			this.GetNumOfFan();
		}
	}
```

hadixcanimm, 2024-09-15:
Your application wasn't working for me. I couldn't control the fans, but the command 'OmenMon.exe -Bios FanCount' allowed me to control the fans the way I wanted for 2 minutes. How can I apply this as a permanent solution?

hadixcanimm, 2024-09-15:
Solution:

[fix.zip](https://github.com/user-attachments/files/17007245/fix.zip)