OmenMon/OmenMon #73: Installed OmenMon on a Omen desktop, now the fan are crazy.

Rewinday, 2024-08-26:
I installed OmenMon on a desktop by mistake. 

Now my fan are incontrollable and I don"t know how to pull back the original configuration... 
I reseted bios and windows and the fan curve is still messed up...

My mobo is a HP 89EB

Mejval5, 2025-08-11:
Omg this happened to me as well.

Here is the fix for my PC:

1. Install Omen Gaming Hub and setup your fan profiles. Technically you can still use omenmon, but I don't trust it anymore.

2. Download OmenMon.exe and change these settings (some of them might be optional) in OmenMon.xml:

<BiosErrorReporting>false</BiosErrorReporting>
<FanCountdownExtendAlways>true</FanCountdownExtendAlways>
<FanCountdownExtendInterval>5</FanCountdownExtendInterval>
<FanLevelMax>5</FanLevelMax>
<FanLevelMin>0</FanLevelMin>
<FanLevelNeedManual>true</FanLevelNeedManual>
<FanLevelUseEc>true</FanLevelUseEc>
<FanProgramModeCheckFirst>false</FanProgramModeCheckFirst>
3. Open command line/powershell in same folder. The commands are essentially same. I use powershell. Open it in same folder as the exe.

4. Run:

.\OmenMon.exe -Bios

-> This shows you how many fans you have, I have two CPU and GPU

5. Run:

.\OmenMon.exe -Bios FanLevel=2,2

-> The 2s are the speed of fans, this will hard set the default to 2. It is very low but during startup your PC will not overheat I think, so all good. You can tweak it. If you have more than two or less, adjust amout of 2s. They should be separated commas. e.g. 3 fans - .\OmenMon.exe -Bios FanLevel=2,2,2

6. Go to Task Scheduler and delete OmenMon from the list (right click delete)

-> this prevents it from running on startup, also disable that in config or in the GUI.

This sets the default values for the fans to be quite low, so you really need to also control the fans after you login. The omen gaming hub is okay for this, so I use it. You might be able to actually use the omenmon by injecting the fan table using the command line. But it is kind of hard to understand.

At least you don't have to wipe your PC and it won't make crazy noises next time you launch it.

Also delete OmenMon from PC if you don't need it. On the other hand, if it is not in the Task Scheduler, it won't run so you can keep it in some folder for later use.

Mejval5, 2025-08-12:
@GeographicCone

I found several people struggling with this issue on reddit and here as well.
If the instructions are well documented then where can we find them? I somehow doubt they cover the case I write here.

If you have a problem with me creating issue ticket that I made sure that others don't spend 5 hours fixing what OmenMon "broke" then at least write your reply to that ticket. Not to a a ticket in which I try to help someone else who fell in the same "hole".

Mejval5, 2025-08-12:
So, you deleted my other ticket.. Great, what a great approach. You don't reply to problems of people for a year after your software broke their system. And when someone finally tries to help, you delete their helpful issue ticket and write "you are spamming repository".... Cmon

I respect that you helped/made a great piece of open source software, but please understand that it can really fk up people's computer and you should write my guide (or similar) to your readme because many people will have disfunctional devices afterwards.