OmenMon/OmenMon #74: Support for HP Pavilion Gaming Laptop 15-dk0xxx

gmedina9144, 2024-09-01:
Hi, im so tired to search tools to control the fans, voltage, performance... I decided to download this tools on my older pc (it have 4 years old since i bought in the pandemic) to long it life due it runs very well. However, i replaced the termical paste and termal pads but it have problema with the battery and the temperatures.

When i download the program, i have this message from the program: BIOS called failed, and i can't modify the speed of the fans, for example. Here the information of the device.

Thank you so much for working on this tool to improve the uses of the devices uses. Answer ASAP.

[image]

[image]

Bubelelele, 2024-11-02:
HP Omen 17-cb1xxxx user here, having the same window pop up!
First run of the .exe creates an OmenMon.sys file and this window pops up trying to run it again.

Sharwesh05, 2024-11-30:
@gmedina9144 how did you find the registers