OmenMon/OmenMon #75: Paypal donation link? I want to assist to add support for amd modles with 144hz screens

maxius4, 2024-09-01:
I have a Victus by HP 16.1 inch Gaming Laptop PC 16-s1023dx. Let me say i absolutely love this program.  the simplicity controlling every thing needed with a few clicks is a godsend.

I would really like to assist you so full support can be added to my amd/nv model hopefully this could open up more amd users to this perfect program.

to be honest i was really concerned about how the hp omen app controls way to much on the laptop and the amount of bloat. what was really surprising on a 8845hs that has a integrated AMD Radeon 780M the chips full spec's are here https://www.amd.com/en/products/processors/laptop/ryzen/8000-series/amd-ryzen-7-8845hs.html with a 4070 how really lax the fan curves are set and with a 230w power adapter... seriously who at hp makes these decisions?  

my concerns were justified! with one long gaming session the laptop's performance tanked badly it was barely responsive and honestly the bios/windows should of shut the laptop down i immediately cut the power let everything cool down this is was on a appropriate incline 16" laptop cooling pad  I Checked everything out and no permanent damage was done and i immediately went looking for a higher watt power adapter. i managed to get an Official HP 330w Power adapter with the same output voltage of 19.5v as the 230w one

firefoot007, 2024-09-01:
Hi, 
I have the same laptop. @maxius4  Why do you think that 230W is not enough power for this laptop?

maxius4, 2024-09-01:
Simple i worked corp it my companies laptop brand of choice was hp. lower wattage bricks produce too much heat per watt. this configuration of laptop running at full power for a long gaming session made the original brick scalding hot when the performance issues occurred in the middle of the long gaming session. that is not something you want with a high performance laptop.

XiaoLin-monian, 2024-11-25:
it support 144hz display,<PresetRefreshRateHigh>144</PresetRefreshRateHigh>