OmenMon/OmenMon #79: Temporary fix to issue #58. Notebook is hibernating

seakyy, 2024-11-05:
OmenMon occasionally forces the notebook to enter hibernation mode and sometimes shows a “Low Battery” notification.

I haven’t had the chance to dive into the code and address this issue yet, but I’ve found a temporary solution for those who prefer using OmenMon over Gaming Hub and want to avoid unexpected hibernation.

**Temporary Fix:**

Search for “**Edit power plan**.”
Go to **Change advanced power** settings.
In the settings window, open the **Battery** section.
Set **Critical battery action** under _Plugged in_ to **Do nothing**.
_(Optional) If the low battery notification is bothersome, turn off Low battery notification under Plugged in._
Make sure to **Apply your settings**, then press **OK**.
This should help prevent OmenMon from forcing your laptop into hibernation.

**⚠️ Note: This is a temporary fix, and the root cause hasn’t been resolved yet.**

michelelorusso05, 2025-03-05:
Is there any updates on this? I also have this problem with my Omen 16 (i7-11800H and RTX 3070); not only that, sometimes the CPU temp reports as 11 °C or 98 °C causing the fans to go full blast, with the temperature reading back correctly only after 10 seconds or so

aleksander431, 2025-04-23:
Thank you a lot! Was wondering what was going on with my laptop) Didn't think OmenMon was the issue.

NotDarkn, 2026-05-10:
hey! i've been encountering this issue a lot lately ever since i installed OmenMon, and i'm really glad to see that there's an issue and possible fix for this :) i'll give this a shot and see if this fixes it.