OmenMon/OmenMon #88: Victus by HP Laptop 16-e1xxx

isachko, 2025-02-07:
Hi!

Downloaded application with hope to find help. But unfortunately application not working.
I changing profiles, setup constant RPM's but laptop just ignoring all those settings.
Please help with this.
Thanks!

h4rqq, 2025-02-10:
same situation with the 16-sxxx model

isachko, 2025-02-11:
OmenMon Hardware Monitoring & Control Utility Version 0.61-Release
BIOS (-Bios)
- Born-On Date: 20220219 [YYYYMMDD]
- System Design Data: c8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 []
- Status Flags: 0x00c8 = 0b0000000011001000 = 200 [BiosPerfPpab3, BiosPerf6, BiosPerf7]
- Unknown Byte: 0x00 = 0b00000000 = 0 [Observed Constant 0x35 = 53]
- Thermal Policy Version: 0x00 = 0b00000000 = 0 [V0]
- Support Flags: 0x00 = 0b00000000 = 0 [0]
- Default CPU Power Limit 4: 0x00 = 0b00000000 = 0 [W]
- BIOS-Defined Overclocking: 0x00 = 0b00000000 = 0 [NotSupported]
- Graphics Mode Switching Support: 0x00 = 0b00000000 = 0 [0]
- Default CPU Concurrent Power Limit w/GPU: 0x00 = 0b00000000 = 0 [W] [Cybug 23C1 Onwards]
- GPU Custom Total Graphics Power (cTGP): 0x01 = 0b00000001 = 1 [On]
- GPU Processing Performance AI Boost (PPAB): 0x01 = 0b00000001 = 1 [On]
- GPU Device Power State (DState): 0x01 = 0b00000001 = 1 [D1]
- GPU Peak Temperature Sensor Threshold: 0x00 = 0b00000000 = 0 [°C]
- Graphics Mode (Legacy): 0x00 = 0b00000000 = 0 [Hybrid]
- Smart Power Adapter Status: 0x01 = 0b00000001 = 1 [MeetsRequirement]
- Overclocking Support: 0x00 = 0b00000000 = 0
- Memory Overclocking Support: 0x00 = 0b00000000 = 0
- BIOS Undervolt Support: 0x00 = 0b00000000 = 0
- Keyboard Type: 0x00 = 0b00000000 = 0 [Standard]
- Keyboard Backlight Support: No
- Keyboard Backlight: 0x80 = 0b10000000 = 128
- Keyboard Backlight Color Table: 4 Zones - Right: 000000, Middle: 000000, Left: 000000, Wasd: 000000
Exception: BIOS call failed: Command not available

OmenMon: Void Check(Int32, Boolean)
   at OmenMon.Hardware.Bios.Bios.Check(Int32 code, Boolean force)
   at OmenMon.Hardware.Bios.BiosCtl.GetAnimTable()
   at OmenMon.AppCli.CliOp.BiosGetStruct[TResult](String message, Func`1 biosMethod, String extra)
   at OmenMon.AppCli.CliOp.Loop(String[] args)
   at OmenMon.AppCli.CliOp.Loop(String[] args)
   at OmenMon.App.Main(String[] args)