OmenMon/OmenMon #94: App not working properly on HP Victus 16 SxxxNT

ctnkyaumt, 2025-04-12:
Hello. This app is great compared to the OG Omen app. But some functions don't work for my laptop:

1. Keyboard backlight setting is passive. I actually tinkered with the code using cursor (I do not know much about coding) and changing the function of zones to act as one, that is changing a zone's color also changing other three zones' colors works.
2. The refresh rate of the panel wasn't recognized correctly. This panel is 144Hz, but it shows 165Hz.
3. Custom and manual fan profiles do work (max, const, off, prog) but when I try to switch back to auto, it doesn't do it.

Other than these, it is OK.

NorthernChicken, 2025-04-30:
I have the same laptop and the same issues. I am not able to adjust the fan speeds at all, I cannot change the color of the keyboard backlights, and I the app reads the GPU temperature but not the CPU temperature. Like you mentioned, it has the options for 165hz and 60hz, but the 165hz option doesn't work since the panel only supports 144hz.

[image]

ctnkyaumt, 2025-05-01:
hi. i forked this and modified the code some. you can download from https://github.com/ctnkyaumt/OmenMon/releases and try it yourself. my only problem with the fan modes is switching back to auto. my release still doesnt change anything about the fan modes. but maybe keyboard lighting, temperature readings and the fan rpm and % reading on the gui may work for you.

NorthernChicken, 2025-05-03:
This actually works perfectly for me... Thanks. You could try to open a PR, but the dev doesn't seem to be active anymore.