OmenMon/OmenMon #105: Windows defender found Vigorf in omenmon.sys

ckotomoto, 2025-09-06:
I've installed Omenmon few weeks ago and it worked just fine. Today Windows Defender deleted omenmon.sys because of Vigrof inside (screenshot atached). Same happens during fresh install. Version 0.61.1

[image]

ckotomoto, 2025-09-06:
I've checked Defender journal and found other messages:

[image]

[image]

Strange that these messages occured only once at first Defender's reaction and were not repeated during fresh installs.

GeographicCone, 2025-09-06:
This has already been reported and discussed.

closed 2025-09-06 by GeographicCone

kytpbs, 2025-09-18:
Extra info shared by Microsoft themselves:

https://support.microsoft.com/en-us/windows/microsoft-defender-antivirus-alert-vulnerabledriver-winnt-winring0-eb057830-d77b-41a2-9a34-015a5d203c42#id0ebd=windows_11